zwave-js/zwave-js #1618: Doc error at getValueMetadata()

hufftheweevil, 2021-02-06:
Unless I'm not understanding something, I think there is an error in the docs at [getValueMetadata()](https://zwave-js.github.io/node-zwave-js/#/api/node?id=getvaluemetadata). 

> Like `getValue` this takes a single argument of the type `ValueMetadata`.

should be

> Like `getValue` this takes a single argument of the type `ValueID`.

And the method returns a `ValueMetadata`.

I can submit a PR for correction, but I wanted to make sure I am not misunderstanding something first.

AlCalzone, 2021-02-06:
Uh, yes that is an error. If you could formulate the correction in a way that we can still link to the ValueMetadata interface, that would be perfect.

AlCalzone, 2021-02-07:
fixed by #1638

closed 2021-02-07 by AlCalzone