zwave-js/zwave-js #1625: Add compat flag to override values reported by a device

AlCalzone, 2021-02-06:
To work around this device firmware (!) issue on the zwave-js side, we need to add a compat flag that allows us to override specific values and avoid querying them from the device (because in this case the supported colors response is simply wrong).

__Originally posted by @AlCalzone in https://github.com/zwave-js/zwavejs2mqtt/issues/477#issuecomment-774475891__

scyto, 2021-02-06:
Is this fix required per device definition file or somewhere else?
Is this just new integration specific or would apply to the MQTT hass discovery code path too?

AlCalzone, 2021-02-06:
This needs to be done in the device config files where necessary to fix wrong device-specific capability reports.

In the mentioned issue, the device does not report that it supports the blue color channel, based on which the channel is not queried by zwave-js and RGB (hex) support is assumed to be false.

scyto, 2021-02-06:
thanks, could this be the root cause on the fibaro?  
https://github.com/Z-Wave-Me/z-way-issues/issues/163#issuecomment-774340754 

this person has indicate the device has a pretty bad bug on reporting channels....

AlCalzone, 2021-02-06:
That looks different

joselito11, 2021-03-15:
Update on version 7 for this rgb bug?

AlCalzone, 2021-03-16:
Sorry, that would require a larger rework of the interview process where I'd have to touch every CC implementation in several locations.

joselito11, 2021-04-01:
So on HA core-2021.4.0b0 and with zwaveJS addon 0.1.15 the light has color wheel, must click few times to get the color, but it works! Thanks!
[image]
[image]

AlCalzone, 2021-04-01:
I didn't do anything here 😳

joselito11, 2021-06-05:
Color picker is gone again

PR #5987 (2023-07-10): feat: add compat flag to override (almost) arbitrary CC API queries
A lot of device issues are caused by incorrect responses to certain queries. This PR attempts to solve this issue by declaratively overriding what a CC API call returns and which values need to be stored in the value DB. It is set up this way, because device interviews internally use the CC API methods.
By hooking into this process, we simply skip the original API call and simulate the result.

This allows us to avoid interview-specific implementations and provides a framework to solve this issue for most problematic devices.
 [... cut]
closed 2023-07-10 by zwave-js-bot