zwave-js/zwave-js #1628: feat(config): add support for Zooz ZEN22 Dimmer

Sudo-Rob, 2021-02-06:
**Which device is missing?**

Zoos ZEN22 Dimmer V2, Ver. 3.0 or later

**What are the IDs?**

Manufacturer ID: 0x015D
Product type: 0x0112
Product ID: 0x241C

**Did you check if we really don't have it?**

- [   ] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [✓] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [✓] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**

https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/1946/zooz-z-wave-plus-dimmer-switch-zen22-manual.pdf

https://cdn.shopify.com/s/files/1/0218/7704/files/zooz-zen22-VER4.0-manual.pdf?v=1601861689

Other relevant links:

https://products.z-wavealliance.org/products/1946?selectedFrequencyId=-1
https://www.support.getzooz.com/kb/section/5/
https://www.home-assistant.io/docs/z-wave/device-specific/

AlCalzone, 2021-02-06:
This one?
https://devices.zwave-js.io/?selectedPath=0x027a:0x1f1c:0xb112:20.15

Sudo-Rob, 2021-02-07:
Good question.

When I look at the entity attributes, it says "friendly_name: Zooz ZEN22 Dimmer V2", "application_version: 4.01."

My current integration shows 22 Node Configuration Options:
1: Paddle Control 
2: LED Indication Control
3: Enable Auto Turn-Off Timer
4: Auto Turn-Off Timer Duration
5: Enable Auto Turn-On Timer
6: Auto Turn-On Timer Duration
7: Association Reports
8: On Off Status After Power Failure
9: Ramp Rate Control
10: Minimum Brightness
11: Maximum Brightness
12: Double Tap Function
13: Enable/Disable Scene Control
14: Enable/Disable Double-tap
15: Smart Bulb Mode: Enable/Disable Paddle / Z-Wave Control
16: Physical Dimming Speed
17: Zwave Ramp Rate Control
18: Custom Brightness Level On
19: 3-Way Switch Type
20: Zwave tap and hold Control
21: Reporting behavior with disabled physical control
22: Night Light Mode 

The above list is accommodates Version 3 HA support.  There are a couple of additions for V 4 firmware.  These appear in the Change Log.  https://www.support.getzooz.com/kb/article/297-zen22-dimmer-change-log/

Is there any other info you might need?  I can perhaps do some legwork for you.

Sudo-Rob, 2021-02-07:
I edited https://devices.zwave-js.io/?selectedPath=0x027a:0x1f1c:0xb112:20.15# to include 13-22 based on the XML file.

https://github.com/Sudo-Rob/Zooz-ZEN22V4.1/tree/main

Please check to see if it conforms to the JSON conventions.  If there's a way to include Help text for each index, let me know so I can add it.

AlCalzone, 2021-02-07:
@blhoward2 something for the task force 😊

Sudo-Rob, 2021-02-07:
Don't take action just yet.  The file I started with, `zen22_3.0-19.0.json`, was missing index #7.  When it's added, I'll report here.

Sudo-Rob, 2021-02-07:
I've updated the file, but might've overachieved.  `Index #7` is pretty arcane.  Looking at the firmware change log, my guess is the configuration will work on V3 and above.  

Here's `Index #7` for the Configuration Parameters chart.

[Index_#7.xlsx](https://github.com/zwave-js/node-zwave-js/files/5939777/Index_.7.xlsx)

AlCalzone, 2021-02-07:
That parameter looks like what I reviewed here:
https://github.com/zwave-js/node-zwave-js/pull/1635#pullrequestreview-585003590

Sudo-Rob, 2021-02-07:
Looks like it.  I would expect a bunch of similarity between similar products by the same manufacturer.  Zooz products are made by Willis Electric Co.,LTD.

How are you mapping XLS to JSON?  Do you have an automated tool for that?

blhoward2, 2021-02-07:
We don't use XLS. We get a json of the device entry from the zwave alliance and use that. In a case like this where need to split them into partial parameters it is largely a manually process. Where did you get an XLS?

Sudo-Rob, 2021-02-07:
Oops.  I meant XML.

blhoward2, 2021-02-07:
Ah,you can download the xmls in json format through a hidden link so we do that.

Sudo-Rob, 2021-02-07:
Does the manufacturer provide them, or is it the Z-Wave Alliance?

blhoward2, 2021-02-07:
The manufacturer provides them to the zwave alliance as part of the certification process. Not all devices are certified, though.

Sudo-Rob, 2021-02-07:
I think I understand.  The XML files look like they have a richer set of entities.  Have you compared the two?

blhoward2, 2021-02-07:
The json we download are the same info in the individual xml, just in json format.

Sudo-Rob, 2021-02-07:
I think there's value in the help information.  I hope it's in the roadmap.

Let me know what I might be able do to advance this feature request.

blhoward2, 2021-02-07:
We've added the inclusion, exclusion, and reset info. And a link to the manual. These are used on the website being developed. We can run back through them and load anything we want fairly automatically but there isn't a plan to use the other info at this point.

If the metadata fields aren't yet in this file and the device exists in the database it means we haven't gotten to it yet.

Sudo-Rob, 2021-02-07:
Understand.  I still need to see if any of my other devices are in #1600. 

Just let me know if there's anything else I might be able to do for you.  I'm looking forward to a more capable integration, but OZW is working for me right now.  I had problems getting JS to work.

Sudo-Rob, 2021-02-08:
Is status "waiting for more info" still accurate?

blhoward2, 2021-02-08:
He's waiting for me to make the partial parameters, so you can add them to yours.

Sudo-Rob, 2021-02-09:
Sounds good.  I wasn't sure if I had any pending action.

PR #1667 (2021-02-10): fix: improved zen22 support
Closes #1628. Rebases zen22 config files to the official Zooz change log. Cleans up labels and descriptions.
closed 2021-02-10 by AlCalzone