zwave-js/zwave-js #1632: Lock fails to assemble partial CCs because the payload is invalid.

prwood80, 2021-02-06:
First - thanks for all the hard work.  I've spent some time digging into this, but I haven't found anyone else with this issue.  Hopefully I didn't miss something.

Issue: I have a Yale Lock (YRD226 - which identifies as YRD256 - which I believe is equivalent per Yale documentation).  When I enter a passcode at the deadbolt, my status is not reflected (doesn't make it to zwavejs2mqtt).  If I manually use the deadbolt (inside knob), it reports both lock and unlock status.  I believe the relevant error in the zwavejs log is: 

> 16:16:38.501 DRIVER   Could not assemble partial CCs because the payload is invalid. Dropping them.

The node in question is 11.  If you start at the bottom of the log, and search [Node 011], you'll see the last message received by the driver related to the unlock with a user code event (time: 16:16:38.444).  Other relevant events (stepping in reverse) would be a manual lock (time: 16:16:31.867) and a manual unlock (time: 16:16:26.386).

Attached are logs.  Sorry, I tried to set the environment variable with my docker container, and restarted, but it appears to have all my devices logged.

Any thoughts?
[yrd256-keypad-access.zip](https://github.com/zwave-js/node-zwave-js/files/5937817/yrd256-keypad-access.zip)

towerhand, 2021-02-06:
Probably related to #1602

prwood80, 2021-02-06:
Possibly, though, I was aware of that issue.  I also have a BE469 (both manual and keypad notifications are working, but I had to implement a work around for some of the status by using the zwave_js_event in home assistant).  The difference was that zwavejs driver was at least processing the zwave message.  In this case, it appears the the message doesn't make it beyond the zwavejs driver, so I'm unable to effectively work around the dropped message.

AlCalzone, 2021-02-06:
For the invalid payload I need your network key. You can send that to the email in my profile

prwood80, 2021-02-06:
@AlCalzone Key sent to your email.  Thanks

AlCalzone, 2021-02-07:
@prwood80 the device sends an invalid UserCode Report, which is being dropped by zwave-js. According to the specs, it must look like this:
[image]
The intent of that command is to report the status **and** code for a given user code ID.

The one sent by your device does not contain any bytes for the user code, so it is basically misusing the User Code CC for notifications.

prwood80, 2021-02-07:
Thanks for checking that. I guess I'm unsure how this works with my two prior controllers (smart things, then built in zwave with home assistant). I thought this was going to be a notification cc that was dropped. This device should report lock status and user code index using the notification alarm type and alarm level according to this document:
https://www.google.com/url?sa=t&source=web&rct=j&url=https://products.z-wavealliance.org/ProductManual/File%3Ffolder%3D%26filename%3DMarketCertificationFiles/3211/YaleRealLivingAssureSL_Z-WavePlus_SystemIntegratorsGuide.pdf&ved=2ahUKEwjAtY6M1dfuAhVWBc0KHcv5BhcQFjANegQIFRAB&usg=AOvVaw21ZMPsTnJ_oGD0ZYgbA4s5

Ikcelaks, 2021-02-07:
The HA legacy zwave-js integration inferred the lock status for the Yale YRD256 (and many other Yale locks) from the alarm_type.

prwood80, 2021-02-07:
@Ikcelaks This is probably a dumb question, but I don't seem to be getting the alarm type (as it seemed to come across in other integrations).  I've also monitored the status with MQTT explorer (states don't seem to update their either).  Am I overlooking something obvious or is the alarm_type not fully supported yet?  It was my basic understanding that the alarm type & code would come across in the Notification message upon a deadbolt action.

AlCalzone, 2021-02-07:
I took a look at the log again. It seems that we do get two Notification reports (one for lock, one for unlock) around the time where the user code report comes.
So basically, this is a duplicate of #1602. I don't think I can realistically do anything about the invalid user code report. So you won't know which user ID unlocked the door.

prwood80, 2021-02-07:
The sequence of events were:
* Manual unlock (time: 16:16:26.386, which was me unlocking from the inside of the house to open the door, notification message received and processed)
* Manual lock (time: 16:16:31.867, which was me locking the door while it was open so that I could then unlock with the keypad, notification message received and processed)
* At time=16:16:38.432, I had initiated an unlock using the keypad with a passcode entry.  This results in the CC payload issue, but no notification messages.  Because of this, nothing seems to change in the zwavejs web gui and no state change is reported over MQTT or to Home Assistant.

In other words, there should have been 3 notification reports, not 2.

Is it possible the user code report you noted above also has an unprocessed notification report CC in the security encapsulation CC?  It looks like multiple commands can be packed in the security message.  If the whole message is dropped, maybe the notification is dropped because of the malformed user code message?  I was trying to use Z-Wave controller to decode the security CC message, but I don't think I was using the right combination of Nonce internal & external (I see the nonce from zwavejs, the IV in the message from the deadbolt, and my local network key).   What tool do you use to decode the packed message?

AlCalzone, 2021-02-07:
> Is it possible the user code report you noted above also has an unprocessed notification report CC in the security encapsulation CC?

Let me check that

AlCalzone, 2021-02-07:
Okay - you were onto something... That IS a notification CC with an embedded (but shortened) User Code Report. I though I already had a workaround for that.

prwood80, 2021-02-07:
That's good to hear. I'd be glad to test if you want me to try something, but my use right now is limited to a docker container on a raspberry pi. 

Thanks so much for your time looking at this.

AlCalzone, 2021-02-07:
If you like, you can build a new container against the branch from #1640.
But since the manual test panned out, I'm probably going to merge this quickly.

towerhand, 2021-02-07:
> That's good to hear. I'd be glad to test if you want me to try something, but my use right now is limited to a docker container on a raspberry pi.
> 
> Thanks so much for your time looking at this.

Are you using zjs2m? if so, here you go: https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/docker_custom_builds

I'm using this command to build mine right now.
DOCKER_BUILDKIT=1 docker build --build-arg SRC=git-clone-src --build-arg Z2M_BRANCH=improved-mesh-graph --build-arg ZWJ_BRANCH=map-notification-lock --no-cache -f zwavejs2mqtt/docker/Dockerfile.contrib -t zwavejs2mqtt .

prwood80, 2021-02-07:
I am using zjs2m. Thanks for this info. It's very helpful!

AlCalzone, 2021-02-07:
2nd part (missing lock status update is done in #1641

closed 2021-02-07 by AlCalzone

prwood80, 2021-02-07:
Built and ran locally with the changes and all is working. Thanks

Quixote1, 2022-02-14:
Sorry for drudging up an old thread, but I seem to be having a similar issue with my Yale YRD256 and Zwavejs2mqtt. The lock operates as it should, and even toggles between the available commands of "Lock" and "Unlock" on its Lovelace card, but the status has shown "Open" since my install a couple of weeks ago. Everything is reasonably up-to-date, though I'm starting to wonder if there may be a firmware update for the lock.  Is there something that I may have failed to configure?  Thanks.

AlCalzone, 2022-02-14:
> in Home Assistant, but the status has shown "Open" since my install a couple of weeks ago

Please report this in https://github.com/home-assistant/core so the devs can figure out if this is an issue in HA or not.

Quixote1, 2022-02-14:
Will do. Thanks

PR #1640 (2021-02-07): fix(cc): ignore invalid evt params, add workaround for truncated User Code CC
fixes: #1632

We now decode a Notification Report `Keypad unlock operation` with the parameter `{userId: 1}` instead of dropping the notification.