zwave-js/zwave-js #1634: Kwikset SmartCode 916 configuration

mgriffin13, 2021-02-07:
**Which device is missing?**

Kwikset Smartcode 916

**What are the IDs?**

Manufacturer ID: 0x90
Product type: 0x0001 
Product ID: 0x0001

**Did you check if we really don't have it?**

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [X] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**
https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/2188/Install_Guide_916_01_23_2017.pdf

https://products.z-wavealliance.org/products/2188/

OZW configuration that works: https://github.com/OpenZWave/open-zwave/blob/master/config/kwikset/916.xml

If so, it helps in creating the config file.

marcus-j-davies, 2021-02-07:
We have this down as  910.
916 is there also but with codes : 0x0001:0x0642

910 : https://devices.zwave-js.io/?selectedPath=0x0090:0x0001:0x0001:0.0
916 : https://devices.zwave-js.io/?selectedPath=0x0090:0x0642:0x0001:0.0

@blhoward2, are these on your list, to cleanse?

mgriffin13, 2021-02-07:
Yeah, ozwdaemon had collision issues as well. Right now, my 916 lock is registering as the 910. I'm not sure exactly what was done to fix the collision for ozwdaemon, but they eventually got it to register correctly.

marcus-j-davies, 2021-02-07:
deleted my comment in error.

marcus-j-davies, 2021-02-07:
What in the name of mike - everything I write get's deleted.
see if tis one gets removed.

blhoward2, 2021-02-07:
They likely just renamed the file. The only way to distinguish devices is the manufacturer ID, product type, product ID, and firmware version. If those overlap between devices then we will need to make the label something like 910 / 916.

blhoward2, 2021-02-07:
@marcus-j-davies I have no idea man...I see your comments in my email.

toddos, 2021-02-07:
My 916 registers as a 914c

Mine has the following manfuacturer codes:

Manufacturer Id = 144, Product Type = 3, Product Id = 1602

blhoward2, 2021-02-07:
We'll get to this one but it is going to require some manual work. It appears that Kwikset failed to include the configuration parameters in its database entry. And if what you're both saying is correct about the identifier, they have the wrong ones listed as @mgriffin13 's only exist under 910.

mgriffin13, 2021-02-07:
Is there a way I can provide a data dump of what the lock shows when paired? Maybe that will help?

blhoward2, 2021-02-07:
Not really. The parameters exist but we'll have to make the file. And do detective work on the device identifiers. As I said we can only distinguish based on those three numbers and the firmware version.

AlCalzone, 2021-02-07:
We could if the device supports Configuration CC V3. In that case we'd have a somewhat decent info about the configuration params in the cache, even without a config file.

mgriffin13, 2021-02-07:
OK, as I said, the ozwdaemon config file i linked works fine. Right now, the lock registers as a 910, but when I try to actually attempt to lock and unlock via HA, nothing happens. It's also registering "current status of the latch" as open, but the door is definitely closed (and locked), and i've never seen that type of setting before (thinking it might be 910 specific).

previous experience with ozwdaemon seems to suggest that the 916 came in both zwave and zwave+ versions, and that they both registered just a little differently. (I believe i have the non-zwave+ version)

blhoward2, 2021-02-07:
> We could if the device supports Configuration CC V3. In that case we'd have a somewhat decent info about the configuration params in the cache, even without a config file.

Open to ideas in your end. I verified all of the zwa files for this set of type/id call this the 910. We're missing all of the parameters form the manual, though.

mgriffin13, 2021-02-07:
There's also this, if it helps? https://s7d2.scene7.com/is/content/Kwikset/SmartCode-916TRL-Z-Wave-Data-Sheet

blhoward2, 2021-02-07:
> There's also this, if it helps? https://s7d2.scene7.com/is/content/Kwikset/SmartCode-916TRL-Z-Wave-Data-Sheet

Unless it has the product type and product id numbers, it won't help anymore than the manual.

mgriffin13, 2021-02-07:
Understood. I actually opened a ticket with Kwikset back in July 2020 asking for that information, and unsurprisingly, they were totally unhelpful and basically said 
>The SmartCode 916 with Z-Wave Plus works with Amazon Alexa via SmartThings or Wink hub. Command Classes, Association Groups, Product Identification, etc is not provided. https://www.kwikset.com/products/detail/916-smartcode-traditional-electronic-deadbolt-with-z-wave-technology?variant=916trl-zw-1

This is the ozwdaemon cache from the lock node, which I think confirms the 0x0090-0x0001-0x0001 product code for the lock.
```xml
    <Node id="5" name="" location="" basic="4" generic="64" specific="3" type="Secure Keypad Door Lock" listening="false" frequentListening="true" beaming="true" routing="true" max_baud_rate="40000" version="4" secured="true" configrevision="15" query_stage="Complete">
        <Neighbors>1,208,94,50,3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0</Neighbors>
        <Manufacturer id="90" name="Kwikset (Spectrum Brands)">
            <Product type="1" id="1" name="Touchpad Electronic Deadbolt">
```

toddos, 2021-02-07:
Interesting.  I see the same thing looking at ozw code (kwikset/916.xml):

```xml
    <MetaDataItem name="OzwInfoPage">http://www.openzwave.com/device-database/0090:0001:0001</MetaDataItem>
    <MetaDataItem name="ProductPic">images/kwikset/916.jpg</MetaDataItem>
    <MetaDataItem id="0642" name="ZWProductPage" type="0003">https://products.z-wavealliance.org/products/2188/</MetaDataItem>
```

The data I'm seeing from zwavejs "Manufacturer specific section" matches up.  Manufacturer Id 144 in decimal = 90 hex, which would match the 0090:0001:0001.  Product id 1602 decimal = 0642 hex, which you can see in the third MetaDataItem.

blhoward2, 2021-02-07:
We can modify ours but if someone could put together a list in hex of product type/id for each model that would speed things up greatly. It's the difference of me changing it tonight versus sometime this week (maybe).

blhoward2, 2021-02-08:
@mgriffin13 How sure are you that your lock is a 916? I've looked in several places, including at OZW, and that is consistently listed as a 910. Type 0x0090, Type 0x0001, ID 0x0001.

blhoward2, 2021-02-08:
> My 916 registers as a 914c
> 
> Mine has the following manufacturer codes:
> 
> Manufacturer Id = 144, Product Type = 3, Product Id = 1602

This is fixed in the PR I just pushed.

mgriffin13, 2021-02-08:
> 
> 
> @mgriffin13 How sure are you that your lock is a 916? I've looked in several places, including at OZW, and that is consistently listed as a 910. Type 0x0090, Type 0x0001, ID 0x0001.
> 
> OZW:
> ` <Product config="kwikset/910.xml" id="0001" name="Touchpad Electronic Deadbolt" type="0001"/>`

Well, a 910 lock has 6 push buttons, whereas the 916 has a touch-screen. I definitely have the touch-screen. I guess it could be possible that my 916 lock has a 910 zwave core, but that would be highly unlikely. I do remember @firstof9 and I working through the quirks of this particular lock about 3-4 months ago with ozw to finally get it right.

blhoward2, 2021-02-08:
Are you sure you have the e correct type and id then? Unless I'm missing something as it stands in the latest version of ozw that type/id causes the 910 file to be loaded, which says 910.

blhoward2, 2021-02-08:
Here is the ozw PR from 7 months ago. But it isn't for that type/id. 

https://github.com/OpenZWave/open-zwave/pull/2319

mgriffin13, 2021-02-08:
When ozwdaemon starts up, i get the following from the log message

>[20200802 13:21:51.878 UTC] [ozw.library] [info]: Info - Node: 5 Node Identity Codes: 0090:0001:0001

I just re-launched my old ozwdaemon container, and it pulled down the latest manufacturer_specific.xml file, so now the ozwdaemon does register the lock as the 910. looking back at my history, i do remember now that i had to manually edit the manufacturer_specific.xml to register correctly, and i was never able to get it pushed up, mainly because it seems my lock is somewhat of an albatross in this regard :(

```
diff manufacturer_specific.xml manufacturer_specific.xml.2 | grep kwik
<     <Product config="kwikset/910.xml" id="0001" name="Touchpad Electronic Deadbolt" type="0001"/>
>     <Product config="kwikset/916.xml" id="0001" name="SmartCode 916" type="0001"/>
```

mgriffin13, 2021-02-08:
Is there some way I can do a similar hack? Update something in zwavejs to override the default discovery, similar to how i was able to manually update manufacturer_specific.xml?

blhoward2, 2021-02-08:
You can edit the device files in your local docker image but you'd need to do it every time. It looks like you probably have an early run of the lock in which they used up some old components. Unfortunately we can't mislabel all devices to accommodate that. The parameters are the same, though, so how much does it saying 910 bother you? It makes no functional difference. 

@AlCalzone ideas of a way to manually override the default device file permanently?

blhoward2, 2021-02-08:
Come to think of it, you could copy the 916 and 910 files outside of the image, modify the 916 to include your number. And modify the 910 to remove your number. Then just map those two files in docker compose and override just those two. It would work unless we renamed the files on our end which were unlikely to do.

mgriffin13, 2021-02-08:
> You can edit the device files in your local docker image but you'd need to do it every time. It looks like you probably have an early run of the lock in which they used up some old components. [...]

Yeah, I appreciate that you can't mislabel devices on my behalf. My concern was that my lock is not functioning at all right now, and it shows extra parameters that aren't applicable (`The current status of the latch`, for instance). Maybe the mislabel was a red-herring and it's more related to the other kwikset issues you guys are working on.

mgriffin13, 2021-02-08:
> Come to think of it, you could copy the 916 and 910 files outside of the image, modify the 916 to include your number. And modify the 910 to remove your number. Then just map those two files in [...]

ooh, that's an interesting idea. i would just need to be aware that i would need to back-port any updates to my mapped files.

blhoward2, 2021-02-08:
@AlCalzone and I discussed that on slack. If it's not working that's a separate issue. Let's close this issue and you can open a bug report. We'll need log files to diagnose the issue.

mgriffin13, 2021-02-08:
alright. which of these files would i override (or both)?

/usr/src/app/node_modules/@zwave-js/config/config/devices/0x0090/916.json
/usr/src/app/node_modules/@zwave-js/server/node_modules/@zwave-js/config/config/devices/0x0090/916.json

blhoward2, 2021-02-08:
That's a good question. It looks like a second set are being brought into the image at building by the server guys. I'd map both for now. You may wait a day though as my PR changes these. 

@AlCalzone Idea as to why they're adding a second set?

blhoward2, 2021-02-08:
We've raised a bug with the server as to why they're adding a second set, but for now the prudent course would be to map both.

mgriffin13, 2021-02-08:
sounds good. i'll wait to complete the mapping until after your PR has been merged and deployed in the zwavejs2mqtt image. i'll also open a new bug report regarding my non-functional lock when i have a little more time to properly debug and gather the logs.

closed 2021-02-08 by mgriffin13