zwave-js/zwave-js #1636: Aeon Labs Minimote DSA03202 v1

kan84, 2021-02-07:
**Which device is missing?**

Aeon Labs Minimote DSA03202 v1

**What are the IDs?**

zw:S type:0101 mfr:0086 prod:0001 model:0003 ver:1.19 zwv:2.78 lib:02 cc:86,72,70,9B ccOut:85,2B,26

**Did you check if we really don't have it?**

- [✓] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)  -
- Found one issue about not waking up but i was not able to add the node.  **_https://github.com/zwave-js/node-zwave-js/issues/1536_**
- [✓] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**
https://aeotec.freshdesk.com/support/solutions/folders/6000148175

If so, it helps in creating the config file.

marcus-j-davies, 2021-02-07:
Hi,

Is this a big report or config request?
This is already in the DB.

https://devices.zwave-js.io/?selectedPath=0x0086:0x0003:0x0001:0.0

AlCalzone, 2021-02-07:
If the device is not working, please use the bug report template and follow the steps there, including providing a logfile.

closed 2021-02-07 by AlCalzone