zwave-js/zwave-js #1637: Add support for Lockwood (Assa Abloy) Keyless Wireless Deadbolt

brettku, 2021-02-07:
**Which device is missing?**

Lockwood (Assa Abloy) Keyless Wireless Deadbolt (YRD240)

**What are the IDs?**

Manufacturer ID: unknown
Product type: unknown
Product ID: unknown

**Did you check if we really don't have it?**

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [x] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [x] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**
https://www.lockweb.com.au/Local/AU/Technical%20Downloads/Lockwood/Secure%20Connect/Keyless%20Wireless%20Digital%20Deadbolt/Lockwood%20Keyless%20Wireless%20Digital%20Deadbolt%20User%20Guide.pdf

brettku, 2021-02-07:
By restarting the Z-Wave JS app and using the lock keypad I was able to get the following response from the log, but it marked it as dead soon after

20:22:53.231 CNTRLR « [Node 024] received unsupported notification {
                          "nodeId": 24,
                          "ccId": "Notification",
                          "ccCommand": "0x05",
                          "alarmType": 21,
                          "alarmLevel": 1
                      }

marcus-j-davies, 2021-02-07:
This device is already  on the DB (but under a label of YRD220).

https://devices.zwave-js.io/?selectedPath=0x0129:0x0000:0x0002:0.0

https://github.com/OpenZWave/open-zwave/blob/f42e4f58572f38f6e4c2b95a74ab445af444f286/config/manufacturer_specific.xml#L273

If you're having problems - please use the bug report template.

marcus-j-davies, 2021-02-07:
@AlCalzone,

Right's to close - if that will help you out.

AlCalzone, 2021-02-07:
> received unsupported notification

That should be changed in the latest zwave-js version.

AlCalzone, 2021-02-07:
I'll give @blhoward2 a chance to look if we need to update config files

PR #1650 (2021-02-10): fix(config): improve Yale Lock support
Cleans up and improves Yale Lock support. Closes #1637 which we already supported anyway
closed 2021-02-08 by blhoward2