zwave-js/zwave-js #1643: Add Support for Ecolink Door Sensors

raccettura, 2021-02-07:
**Which device is missing?**

There's 2 versions, one looks like zwave plus (I've only got the original)

DWZWAVE1
DWZWAVE2.5-ECO

**What are the IDs?**

Manufacturer ID: 0x014a
Product type: 0x0001
Product ID: 0x0002  

**Did you check if we really don't have it?**

- [x ] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [ x] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [x] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**

If so, it helps in creating the config file.

https://products.z-wavealliance.org/products/790?selectedFrequencyId=2
https://products.z-wavealliance.org/products/1498?selectedFrequencyId=2

blhoward2, 2021-02-08:
We support this device already. 

packages/config/config/devices/0x014a/dwzwave2.json

closed 2021-02-08 by blhoward2

raccettura, 2021-02-21:
That only looks like the DWZWAVE2 version of this device.  There is a DWZWAVE1.  

I believe the difference is zwave plus vs zwave.

Testing against 6.4.0 results in:
Unknown manufacturer undefined |   | Unknown product undefined

blhoward2, 2021-02-21:
You have to wake the device up. If it says undefined it will never match any device file.