zwave-js/zwave-js #1644: Configuration sub-fields should be allowed to be batched

blunderingautomator, 2021-02-07:
### Bug Description
I have several LZW36 Ceiling Fan/Light controllers, but other Inovelli devices will have the same issue. 

To configure notifications, you must program a configuration item that combines effect, color, intensity, and duration into a 4-byte configuration item. 

With zwave-js, I noticed that [this configuration ](https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x031e/lzw36.json)was conveniently separated into separate fields:
```json
		"24[0x7f000000]": {
			"label": "Light LED Strip Effect",
			"description": "Light LED Strip Effect. 0 = Off, 1 = Solid, 2 = Slow Blink, 3 = Fast Blink, 4 = Chase, 5 = Pulse Default: 0",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 5,
			"defaultValue": 0,
			"unsigned": true,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Off",
					"value": 0
				},
				{
					"label": "Solid",
					"value": 1
				},
				{
					"label": "Slow Blink",
					"value": 2
				},
				{
					"label": "Fast Blink",
					"value": 3
				},
				{
					"label": "Chase",
					"value": 4
				},
				{
					"label": "Pulse",
					"value": 5
				}
			]
		},
		"24[0xff0000]": {
			"label": "Light LED Strip Effect Duration",
			"description": "Light LED Strip Effect Duration. 1 to 60 = seconds, 61 to 120 minutes, 121 - 254 = hours, 255 = Indefinitely Default: 0",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 255,
			"defaultValue": 0,
			"unsigned": true,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"24[0xff00]": {
			"label": "Light LED Strip Effect Intensity",
			"description": "Light LED Strip Effect Intensity. 0 to 9. 0 = dim, 9 = bright Default: 3",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 9,
			"defaultValue": 3,
			"unsigned": true,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"24[0xff]": {
			"label": "Light LED Strip Effect Color",
			"description": "Light LED Strip Effect Color. Color - 0 - 255. Hue color wheel. value/255 * 360 = Hue color wheel value Range: 0-255 Default: 0",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 255,
			"defaultValue": 0,
			"unsigned": true,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true,
			"options": [
				{
					"label": "Red",
					"value": 0
				},
				{
					"label": "Orange",
					"value": 21
				},
				{
					"label": "Yellow",
					"value": 42
				},
				{
					"label": "Green",
					"value": 85
				},
				{
					"label": "Cyan",
					"value": 127
				},
				{
					"label": "Blue",
					"value": 170
				},
				{
					"label": "Violet",
					"value": 212
				},
				{
					"label": "Pink",
					"value": 234
				}
			]
		},
```

While this is _super_ convenient for one-time configurations, the fact is, this particular configuration parameter is a notification that changes frequently. So what's the problem with this? (Sorry this is taking so long, but I needed some setup info.)

Okay, say I want to fire a Red, Fast Blink notification, then I want to switch to a Yellow, Solid notification. Each time I do this, I need to send out two requests on the Z-Wave mesh. 

1) zwave-js seems to be messing up in the read-modify-write flow because half the time, either the notification type or color is wrong.
2) Multiply this by 4 sub-parameters with 2 indicators on 6 devices, and now we're up to 4x2x6x2=96 calls, which could have been done in 12. It's known that flooding Z-wave is not good, and this behavior doesn't help. 

I have verified what I'm writing over MQTT for this configuration works, and I can flood MQTT with several requests until it finally gets the notification right (usually takes 3-5 tries). 

### Proposal

From the MQTT perspective, give us the option to write the entire configuration parameter from the parameter's parent: 

Allow me to write zwave/nodeId/112/0/25/set to set all 4 sub-parameters by it's parent value instead of individually like below:
[image]

**Device information**

Which device(s) is/are affected (make/model)?  Any devices with sub-parameters (Inovelli products for sure)
What are the node IDs? 2, 4, 13, 14, 18, 22

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [X] Previously working device (other platform)
    - Which platform? Home Assistant built-in Z-Wave
    - Have you made any recent configuration changes to the device? Describe. Reset and readded all nodes. 
    
**Installation information**
How did you install `node-zwave-js`?
- [X] `zwavejs2mqtt` (dev) docker image

**To Reproduce**
Steps to reproduce the behavior:
1. Add a device with sub-parameters
2. Quickly (scripted) program all 4 parameters at once
3. Witness that while the config items are _set_ correctly in MQTT, that the zwavejs2mqtt web page has the 2-112-0-25-255 color incorrect:

MQTT is correct:
[image]

Log indicates the sub-parameters are received from MQTT and sent out correctly to Z-Wave (or whatever layer is being logged here), but when getting updated, they are incorrect:
```
# Correct values below from log
2021-02-07 14:32:53.278 INFO MQTT: Message received on zwave/2/112/0/24/255/set, '0'
2021-02-07 14:32:53.279 INFO ZWAVE: Writing 0 to 2-112-0-24-255 <----------------------------- Correct Here
2021-02-07 14:32:53.284 INFO MQTT: Message received on zwave/2/112/0/25/255/set, '0'
2021-02-07 14:32:53.284 INFO ZWAVE: Writing 0 to 2-112-0-25-255 <----------------------------- Correct Here
2021-02-07 14:32:53.285 INFO MQTT: Message received on zwave/2/112/0/25/2130706432/set, '5'
2021-02-07 14:32:53.285 INFO ZWAVE: Writing 5 to 2-112-0-25-2130706432
2021-02-07 14:32:53.285 INFO MQTT: Message received on zwave/2/112/0/24/2130706432/set, '5'
2021-02-07 14:32:53.285 INFO ZWAVE: Writing 5 to 2-112-0-24-2130706432
# When it reads them back, they're incorrect
2021-02-07 14:32:54.388 INFO ZWAVE: Node 2: value updated: 112-0-24-2130706432 5 => 5
2021-02-07 14:32:54.389 INFO ZWAVE: Node 2: value updated: 112-0-24-16711680 255 => 255
2021-02-07 14:32:54.389 INFO ZWAVE: Node 2: value updated: 112-0-24-65280 9 => 9
2021-02-07 14:32:54.390 INFO ZWAVE: Node 2: value updated: 112-0-24-255 0 => 45 <----------- Incorrect Here
2021-02-07 14:32:54.431 INFO ZWAVE: Node 2: value updated: 112-0-25-2130706432 5 => 5
2021-02-07 14:32:54.432 INFO ZWAVE: Node 2: value updated: 112-0-25-16711680 255 => 255
2021-02-07 14:32:54.432 INFO ZWAVE: Node 2: value updated: 112-0-25-65280 9 => 9
2021-02-07 14:32:54.433 INFO ZWAVE: Node 2: value updated: 112-0-25-255 0 => 45 <----------- Incorrect Here
2021-02-07 14:32:54.491 INFO ZWAVE: Node 2: value updated: 112-0-25-2130706432 5 => 5
2021-02-07 14:32:54.492 INFO ZWAVE: Node 2: value updated: 112-0-25-16711680 255 => 255
2021-02-07 14:32:54.492 INFO ZWAVE: Node 2: value updated: 112-0-25-65280 9 => 9
2021-02-07 14:32:54.493 INFO ZWAVE: Node 2: value updated: 112-0-25-255 45 => 45 <----------- Incorrect Here
2021-02-07 14:32:54.535 INFO ZWAVE: Node 2: value updated: 112-0-24-2130706432 5 => 5
2021-02-07 14:32:54.536 INFO ZWAVE: Node 2: value updated: 112-0-24-16711680 255 => 255
2021-02-07 14:32:54.536 INFO ZWAVE: Node 2: value updated: 112-0-24-65280 9 => 9
2021-02-07 14:32:54.537 INFO ZWAVE: Node 2: value updated: 112-0-24-255 45 => 45 <----------- Incorrect Here
```

**Additional context**
Add any other context about the problem here.

**Logfile:** 
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5939973/zwavejs_1.log)

AlCalzone, 2021-02-07:
It seems that https://github.com/zwave-js/node-zwave-js/issues/1347 would solve this issue, but if possible we should avoid merging partial params with outdated values.

blunderingautomator, 2021-02-07:
Yep, that's exactly what we need. 

Perhaps instead of /set, we use /store, then at any parent level under zwave/ we can call /flush that will take everything in the /stores, compile the writes, then flush them out? Anything missing /store can use the current value, and obviously if the current value matches what is to be written, discard the write (although that goes against what a flush would normally do)?

Thanks for working on zwave-JS! This is a great addition to the community.  

This is somewhat the flow I would think would work:
```
INFO MQTT: Message received on zwave/2/112/0/24/255/store, '0'
INFO MQTT: Message received on zwave/2/112/0/25/255/store, '0'
INFO MQTT: Message received on zwave/2/112/0/25/2130706432/store, '5'
INFO MQTT: Message received on zwave/2/112/0/24/2130706432/store, '5'
INFO MQTT: Message received on zwave/2/112/0/flush
2021-02-07 14:32:53.285 INFO ZWAVE: Writing 67045888 to 2-112-0-24
2021-02-07 14:32:53.285 INFO ZWAVE: Writing 67045888 to 2-112-0-25
INFO MQTT: Setting zwave/2/112/0/24/255/store to ''
INFO MQTT: Setting zwave/2/112/0/25/255/store to ''
INFO MQTT: Setting zwave/2/112/0/24/2130706432/store to ''
INFO MQTT: Setting zwave/2/112/0/25/2130706432/store to ''
```

AlCalzone, 2021-02-07:
/cc @robertsLando for the MQTT part. zwave-js would just accept arrays of valueID -> value and make the best of it

jtronicus, 2021-02-08:
I always envisioned partial parameters being part of a sub-topic, like this:
[image]

In this example, the LED Strip Effect parameter consists of 4 parts: Color, Brightness, Duration, and Effect.  I can update the parameter in 1 of 3 ways.

Send a value to the parameter itself with the calculations already done:
```
Topic: testzwavejs/nodeID_26/configuration/endpoint_0/LED_Strip_Effect/set
Message: 34212522
```

Send a value to the parameter itself, but specify each part:
```
Topic: testzwavejs/nodeID_26/configuration/endpoint_0/LED_Strip_Effect/set
Message: {"Color": 170, "Brightness": 10, "Duration": 10, "Effect": "Chase" }
```

Send 4 message, one for each part:
```
Topic: testzwavejs/nodeID_26/configuration/endpoint_0/LED_Strip_Effect/Color/set
Message: 170

Topic: testzwavejs/nodeID_26/configuration/endpoint_0/LED_Strip_Effect/Brightness/set
Message: 10

Topic: testzwavejs/nodeID_26/configuration/endpoint_0/LED_Strip_Effect/Duration/set
Message: 10

Topic: testzwavejs/nodeID_26/configuration/endpoint_0/LED_Strip_Effect/Effect/set
Message: "Chase"
```

AlCalzone, 2021-06-17:
https://github.com/zwave-js/node-zwave-js/pull/2875 solves this, but you will have to use the `commandClasses` API, not the `setValue` API.

closed 2021-06-18 by AlCalzone