zwave-js/zwave-js #1658: No Up/stop/Down buttons in HA

labaland, 2021-02-08:
**Which device is missing?**
Qubino  Flush Shutter DC - ZMNHOD

**What are the IDs?**
??

**Do you have a link to the PDF manual?**
[https://support.qubino.com/support/solutions/articles/44001757258-qubino-flush-shutter-dc](url)

AlCalzone, 2021-02-08:
1. Please take a second and read the information we put under the different issue templates
[image]
They are there for a reason.

2. If something is not working in HA, start by opening an issue in the HA repo.

closed 2021-02-08 by AlCalzone

marcus-j-davies, 2021-02-08:
@AlCalzone  - You beat me to it,

This already exists : https://devices.zwave-js.io/?selectedPath=0x0159:0x0053:0x0003:0.0.

So shouldn't belong here.