zwave-js/zwave-js #1661: Add bin firmware format.

DelusionalAI, 2021-02-08:
**Describe the bug**

When trying to update the firmware or my inovelli switch, I get an error when trying go update target one. It says it is unable to detect firmware format. 

    
**Installation information**
How did you install `node-zwave-js`?
- [ X] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

The firmware in question can be found below. 
https://support.inovelli.com/portal/en/kb/articles/firmware-change-log-lzw31-sn-dimmer-switch-red-series

AlCalzone, 2021-02-10:
@robertsLando these bin files are for firmware target 1 (i.e. not the z-wave chip) - do you have a way of selecting that?

AlCalzone, 2021-02-10:
#1703 fixes this. Note that you need to select the firmware target 1 when updating the .bin files. Not sure if that is possible yet.

DelusionalAI, 2021-02-10:
Yes you can do that now with https://github.com/zwave-js/zwavejs2mqtt/pull/501 on zwavejs2mqtt.

PR #1703 (2021-02-10): feat: support .bin firmware files
fixes: #1661
closed 2021-02-10 by AlCalzone