zwave-js/zwave-js #1662: Zooz Zen30 manufacturer and device ID mismatch

cbb09, 2021-02-08:
**Describe the bug**

I saw in the zwavejs release notes for 6.1.3 that the Zen30 device was added. I've updated my zwavejs2mqtt docker image to 1.1.0, which includes the latest zwavejs release. However, my Zen30 double switch is still "unknown manufacturer" and "unknown device". I've checked the device info in the zwavejs XML file and it seems it's different than what the device itself reports (see above for the ones the device reports.). I've also tried removing and re-adding the switch to my Z-Wave network, but no luck.

 I've copied the relevant section of the device debug info below.
 "deviceId": "634-40968-40960",
  "manufacturer": "Unknown manufacturer 634",
  "manufacturerId": 634,
  "productType": 40960,
  "productId": 40968,
  zwave protocal is: 5.3.0
  command class: 134
  applicationversion: 1.5.1
  firmwareversion: 1.5

**Device information**

Which device(s) is/are affected (make/model)?  Zooz Zen30
What are the node IDs?  "manufacturerId": 634, "productType": 40960, "productId": 40968,

**Last Known Working Configuration**
- [X] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version? Zwavejs Version 6.1.3; App Version 1.1.0
    - Have you made any recent configuration changes to the device? No

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? 
    
**Installation information**
How did you install `node-zwave-js`?
- [X] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Steps to reproduce the behavior:

**Additional context**
**Logfile:**

AlCalzone, 2021-02-08:
I think that was the mix-up we discussed earlier?

cbb09, 2021-02-08:
Had a closer look - it appears that productID and productType are switched.

blhoward2, 2021-02-08:
Agree. #1660 should close this.

cbb09, 2021-02-08:
Thanks much! I made a quick edit and swapped the IDs in my running docker image and after restarting, the device shows up correctly.

closed 2021-02-09 by blhoward2