zwave-js/zwave-js #1663: Inovelli LZW31-SN Missing Config Parameters

tlykken, 2021-02-08:
**Describe the bug**

The config files for Invoelli LZW31-SN, Red Series Dimmer Switch - is missing some config parameters.

Device Manufacturer - 0x031w
Product Type": 0x0001
Product Id": 0x0001

Missing Parameter (all): Parameter 12 - Association Behavior [(Documented Here)](https://products.z-wavealliance.org/products/3438/configs)

Parameter 12 - Association Behavior 
  | 1 | Local
  | 2 | 3-Way
  | 3 | 3-Way + Local
  | 4 | Z-Wave HUB
  | 5 | Z-Wave HUB + Local
  | 6 | Z-Wave HUB + 3-Way
  | 7 | Z-Wave HUB + Local
  | 8 | Timer
  | 9 | Timer + Local
  | 10 | Timer + 3-Way
  | 11 | Timer + Local
  | 12 | Timer + Z-Wave HUB
  | 13 | Timer + Local
  | 14 | Timer + Z-Wave HUB
  | 15(Default) | All

Missing Parameter (Firmware version 1.47+) - Instant On [Found here](https://support.inovelli.com/portal/en/kb/articles/firmware-change-log-lzw31-sn-dimmer-switch-red-series)
Add parameter 51, to enable instant-on (ie: disable the 700ms delay). Note, if you disable the delay, it will also disable scene control except for Button 1 (ie: tap up 1x or tap down 1x) and button 7 (config button). All other buttons (2-6) will be disabled.
Parameter 51
Size: 1 Byte
Default: 1
Range: 0-1
0: No Delay
1: 700ms Delay

Missing Parameter (Firmware version 1.49+) -  Button Press Delay [Found here](https://support.inovelli.com/portal/en/kb/articles/firmware-change-log-lzw31-sn-dimmer-switch-red-series)
Adding parameter 50 to adjust the delay when parameter 51 is set to 1. 1=100ms, 2=200ms, 3=300ms, etc.

For all configs - 
**Device information**

Which device(s) is/are affected (make/model)?  Inovelli LZW31-SN
What are the node IDs? 
Device Manufacturer - 0x031w
Product Type": 0x0001
Product Id": 0x0001

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [X ] Other: Home Assistant via zwavejs2mqtt

**To Reproduce**
Steps to reproduce the behavior:
Check config file located [here](https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x031e/lzw31-sn.json)

**Additional context**
Add any other context about the problem here.

**Logfile:**

blhoward2, 2021-02-08:
This will be fixed as part of #1600, that's why it's listed there.

closed 2021-02-08 by blhoward2

tlykken, 2021-02-09:
Thank you for the feedback.  The second two parameters are not listed for these devices by the zwave alliance.    These are in newer firmware releases for this switch.   Is there something I can do to help?  I haven't done any work with this team before but I am willing to learn.

blhoward2, 2021-02-09:
Not at this point. We're building batches of files as we go. We'll split the files by firmware version when we get to them, that's why this one isn't already done.

tlykken, 2021-02-09:
Sounds good, thanks for all your hard work on this!