zwave-js/zwave-js #1664: Ct101 Radio Thermostat  not showing command class

neuman1812, 2021-02-08:
I checked the list at #1600  for my device.  I see the Ct100 is listed,  but I have the CT101.  

two issues with it:  In my log I have: 

`Dropping message because it could not be deserialized: The command class Thermostat Fan State is not implemented`

and lovelace doesn't properly display the "Current Temp" in the thermostat card.

[Imgur](https://i.imgur.com/QbBzFiX.png)

Im running Hassio: 

Core: core-2021.2.1
Supervisor: supervisor-2021.02.5
Operating System:  Home Assistant OS 5.11
Host: RPI 4

AlCalzone, 2021-02-08:
> The command class Thermostat Fan State is not implemented

You can request a CC implementation here: https://github.com/zwave-js/node-zwave-js/issues/729

> and lovelace doesn't properly display the "Current Temp" in the thermostat card.

That is an issue with HomeAssistant. Please report it in the [HomeAssistant repo](https://github.com/home-assistant/core)!

> I see the Ct100 is listed, but I have the CT101

@blhoward2 ?

chilicheech, 2021-02-08:
Fan state has already been requested: https://github.com/zwave-js/node-zwave-js/issues/729#issuecomment-733972927 :)

neuman1812, 2021-02-09:
> > and lovelace doesn't properly display the "Current Temp" in the thermostat card.
> 
> That is an issue with HomeAssistant. Please report it in the [HomeAssistant repo](https://github.com/home-assistant/core)!

issue opened: https://github.com/home-assistant/core/issues/46257

Hilo4321, 2021-02-09:
> > I see the Ct100 is listed, but I have the CT101
> 
> @blhoward2 ?

I have 3 CT101 devices that appear identical physically and from OZWAdmin, other than one character in the hex Product Code. One of them shows up in ZJS as a CT100 for me.

blhoward2, 2021-02-09:
You need to post the type and device id if the missing ones so I can take a look, but that wouldn't cause it not to work.

Hilo4321, 2021-02-09:
Googling Device ID seems to indicate that means Node ID, but I can't imagine that's what you want. What do you need?

In my case, there is no config for the device in ZJS for the device ID'd as CT100.

AlCalzone, 2021-02-09:
No we need the product ID and product type - you should be able to see that in zwavejs2mqtt. Not sure about HA.

blhoward2, 2021-02-09:
It'll be a series of hex numbers in zwave2jsmqtt that identify your device 0x0000-0x0000-0x0000. I believe it's at the top of the node menu when you click on a node. No does if or where the HA server shows that.

neuman1812, 2021-02-09:
If it helps,  here's my [network.jsonl ](https://pastebin.com/4Qmzwz1z) .  I believe the Product code/ID is listed in there. 

`asic", "Manufacturer Specific", "Thermostat Mode", "Thermostat Setpoint", "Version"], "mandatoryControlCCs": []}, "isListening": false, "isFrequentListening": true, "isRouting": true, "maxBaudRate": 40000, "isSecure": false, "version": 3, "isBeaming": true, "manufacturerId": 152, "productId": 12, "productType": 25857, "firmwareVersion": "9.0", "deviceConfig": {"manufacturerId": 152, "manufacturer": "Radio Thermostat Company of America (RTC)", "label": "CT101", "description": "Z-Wave Thermostat", "devices": [{"productType": "0x6501", "productId": "0x0000"}, {"productType": "0x6501", "productId": "0x000c"}, {"productType": "0x6501", "productId": "0x000d"}], "`

Hilo4321, 2021-02-09:
OZW ProductCodes:
Properly ID'd CT101: 0x0098 0x000c 0x6501
CT101 ID's as CT100 (no configs in ZJS): 0x0098 0x000c 0x6501

AlCalzone, 2021-02-09:
https://devices.zwave-js.io/?selectedPath=0x0098:0x0015:0x6401:0.0 includes the last ID

Hilo4321, 2021-02-09:
That will be incorrect, as they are pretty obviously different devices.

AlCalzone, 2021-02-09:
Just pointing it out because you said "no configs". The ID was most likely added to the wrong file.

Hilo4321, 2021-02-09:
Ah, sorry, I should clarify that I meant no configuration parameters.

Hilo4321, 2021-02-12:
#1733 will fix misidentified device

PR #1733 (2021-02-12): fix(config): move product Type/Id from CT100 to CT101
One of my 3 CT101 thermostats is incorrectly identified as a CT100. This device is correctly identified in OZW Admin.

fixes: #1664
closed 2021-02-12 by AlCalzone