zwave-js/zwave-js #1665: Renaming devices and ripple rename entities only a one-time-opportunity?

palengh, 2021-02-08:
It seems like name changing of devices only passes along it´s new name to the entities the first time it´s run: "Do you also want to rename the entity IDs of your entities?"

Seemsl like it only works one time, and that I cannot ripple change names after it´s very first run.
Even If I set the name before all entities are descovered.
Is this by design?

AlCalzone, 2021-02-08:
Did you mean to open this in the HomeAssistant Repo?

palengh, 2021-02-08:
This is my experience using zwave-js inside HA. But I am new to reporting these, so be kind ;-) My experience is that I cannot ripple change entity names using the zwave-js

Madelaide, 2021-02-08:
FWIW this issue can be worked around by opening the node editor, deleting the name (cmd-x), saving, then reopening the editor and pasting the name back in (CMD-V) - this is especially useful if new attributes are scanned in and added to the node (which have a default name). This process resets them all to be the same, and then the rename will work.

palengh, 2021-02-08:
> FWIW this issue can be worked around by opening the node editor, deleting the name (cmd-x), saving, then reopening the editor and pasting the name back in (CMD-V) - 

THIS IS WORTH a million. Thanx!

AlCalzone, 2021-02-08:
Entities are a HA concept, zwave-js has nothing to do with it. So if there is a problem with that, consider opening an issue in the HA repo.

closed 2021-02-08 by AlCalzone