zwave-js/zwave-js #1668: Up/Down and currentValue are reversed in Qubino ZMNHCD

deiger, 2021-02-09:
**Describe the bug**
I have two Qubino ZMNHCD roller shutters. In both cases, the Multilevel Switch is reversed:
x-38-0-Down results in opening the shutter, while x-38-0-Up results in closing it. Likewise, x-38-0-currentValue is 0 for an open shutter and 99 for a closed one.
This is not a wiring issue, as both shutters worked correctly in SmartThings.

**Device information**
Qubino ZMNHCD, firmware 1.1 - 0x0159-0x0052-0x0003

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [x] Previously working device (other platform)
    - Which platform?
      SmartThings
    - Have you made any recent configuration changes to the device? Describe.
      No
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [x] Other: `zwavejs2mqtt` [Home Assistant add-on](https://github.com/hassio-addons/addon-zwavejs2mqtt)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `Multilevel Switch` value for a Qubino Flush Shutter in the Control Panel.
2. Click on `Perform a level change (Down)` or `Perform a level change (Up)`.
3. The shutter will do the reversed action, and the currentValue is inverted as well.

**Additional context**
This seems to have long been an issue with roller shutters. In the OpenZwave integration of HomeAssistant, there are configuration flags `invert_openclose_buttons` and `invert_percent` - see [here](https://github.com/home-assistant/core/blob/dev/homeassistant/components/zwave/cover.py#L15-L16). I'm not sure if it is better to force the flip or allow the user to inverse them.

hmmbob, 2021-02-09:
For what it's worth: if I use UP and DOWN commands in Home Assistant for the same device, the cover moves as expected (UP or DOWN). To put it differently: for me it works as expected.

Maybe Smartthings did some magic somewhere, but I'm not so sure about your statement that it isn't a wiring issue (respectfully).

AlCalzone, 2021-02-09:
You deleted the most important part of the issue template - the logfile. 

I have a couple of Fibaro Roller Shutters and they work perfectly and are not inverted (0 = closed, 99 = open). The commands are guaranteed the same for both devices.

The Window Covering CC specs (I know, different one, but they are designed to be compatible) describe:
> Level Change Up = Opening, Level Change Down = Closing
> The term “Closed” represents the lowest light throughput, while “Open” represents the highest light
throughput of the window covering.

Since currentValue and up/down are interpreted in the same inverted way by your device, I'll have to disagree with your assessment that it is not a wiring issue. I'm not sure what SmartThings does differently in this regard, but maybe you had an invert option set somewhere.

If you really need an invert option, I suggest you raise an issue in the [HomeAssistant repo](https://github.com/home-assistant/core).

blhoward2, 2021-02-09:
Just speculating but some of those devices have inversion parameters. Have you checked the configuration for the device? If not, you may want to google around and see if the manual shows such a parameter. Yours may be set funny from SmartThings even if our device file doesn't present the option yet.

AlCalzone, 2021-02-09:
I couldn't find any 🤷🏻‍♂️

deiger, 2021-02-09:
I couldn't find such parameters either.
Also, I did not reset the modules on migration from SmartThings, only excluded and included them - so all set parameters remained the same.

hmmbob, 2021-02-09:
Debug info of my node:
[Qubino shutter module.txt](https://github.com/zwave-js/node-zwave-js/files/5952059/Qubino.shutter.module.txt)

I don't see any config option for this in the file.

deiger, 2021-02-09:
Here is mine: [zmnhcd.txt](https://github.com/zwave-js/node-zwave-js/files/5952145/zmnhcd.txt)

hmmbob, 2021-02-09:
I compared yours and mine, and did not see significant changes. If AlCalzone doesn't see an issue in our dumps, I'm afraid you'll be needing to get your screwdrivers out and switch the wires ;-)

You'll need to switch Q1 and Q2 either at the module or at the motor side.
[image]

blhoward2, 2021-02-09:
You might try hard resetting the device. Sometimes SmartThings changes weird things that may be undocumented.

kpine, 2021-02-09:
If you're talking about SmartThings, I believe you are able to choose a specific Device Handler (or is it autodetected), is that right?

There is a custom Qubino Flush Shutter DH. The [open/close functions](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/ffa0b5ed083c73f071ce90daf769f5b0354bc507/devicetypes/qubino/qubino-flush-shutter.src/qubino-flush-shutter.groovy#L223-L229) are peculiar.
```groovy
def close() {
	setShadeLevel(0x64)
}

def open() {
	setShadeLevel(0x00)
}

def setShadeLevel(level) {
	log.debug "Setting shade level: ${level}"
	encap(zwave.switchMultilevelV3.switchMultilevelSet(value: Math.min(0x63, level)))
}
```

It looks to me it has inverted the open and close values. Open == 0x0, Close == 0x63. So if you're using the DH, then that might explain why open/close behave opposite of past experience.

There is also a generic "Z-Wave Basic Window Shade" [DH](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/36d37c1b431eb81c2722202c66dd07d38f3bd3dd/devicetypes/smartthings/zwave-basic-window-shade.src/zwave-basic-window-shade.groovy#L131-L139) with a configurable [reverse option](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/36d37c1b431eb81c2722202c66dd07d38f3bd3dd/devicetypes/smartthings/zwave-basic-window-shade.src/zwave-basic-window-shade.groovy#L56). It looks to me the default behavior for that DH is still Open == 0x0, Close == 0xFF.

If I understand SmartThings and these DH correctly, then it seems SmartThings has reversed the meanings of the values as defined by the spec, as Al has mentioned?

deiger, 2021-02-09:
@kpine Nice idea. However I've been using the [official DH by Qubino](https://github.com/kjamsek/SmartThings/blob/master/DeviceHandlers/Qubino/FlushShutter/QubinoFlushShutterDeviceHandler.groovy), Which seems to do what it should:
```groovy
def open() {
	log.debug "Qubino Flush Shutter: open()"
	return zwave.switchMultilevelV3.switchMultilevelSet(value: 0xFF, dimmingDuration: 0x00).format()
}

def close() {
	log.debug "Qubino Flush Shutter: close()"
	return zwave.switchMultilevelV3.switchMultilevelSet(value: 0x00, dimmingDuration: 0x00).format()
}

def setSlatLevel(level) {
	log.debug "Qubino Flush Shutter: setSlatLevel()"
	log.debug level
	if(level > 99) level = 99
	//zwave.multiChannelV3.multiChannelCmdEncap(sourceEndPoint:1, destinationEndPoint:2).encapsulate(zwave.switchMultilevelV3.switchMultilevelSet(value: level, dimmingDuration: 0x00)).format()
	if(settings.param72 == null){
		return delayBetween([
			zwave.multiChannelV3.multiChannelCmdEncap(sourceEndPoint:1, destinationEndPoint:2).encapsulate(zwave.switchMultilevelV3.switchMultilevelSet(value: level, dimmingDuration: 0x00)).format(),
			zwave.multiChannelV3.multiChannelCmdEncap(sourceEndPoint:1, destinationEndPoint:2).encapsulate(zwave.switchMultilevelV3.switchMultilevelGet()).format()
		], 1500)
	}else{
		def tempTurningTime = settings.param72.toInteger()*10
		return delayBetween([
			zwave.multiChannelV3.multiChannelCmdEncap(sourceEndPoint:1, destinationEndPoint:2).encapsulate(zwave.switchMultilevelV3.switchMultilevelSet(value: level, dimmingDuration: 0x00)).format(),
			zwave.multiChannelV3.multiChannelCmdEncap(sourceEndPoint:1, destinationEndPoint:2).encapsulate(zwave.switchMultilevelV3.switchMultilevelGet()).format()
		], tempTurningTime)
	}
}
```

I do see though that the [DH by SmartThings](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/qubino/qubino-flush-shutter.src/qubino-flush-shutter.groovy) is messed up:
```groovy
def close() {
	setShadeLevel(0x64)
}

def open() {
	setShadeLevel(0x00)
}
```

I guess my best option is to change the wiring. I do think though that an option for inverting the values is a good one to have.

blhoward2, 2021-02-12:
It looks like this may be an HA issue? I saw this on the Community website:

"When I upgrade to 2021.2.3 , my FGRM222 shutter switches are inverted in the UI ?!..(i use open-zwave)"

deiger, 2021-02-12:
The issue appears exactly the same in zwave2jsmqtt. That's why I opened the issue here.

closed 2021-02-15 by zwave-js-assistant[bot]