zwave-js/zwave-js #1669: Qubino (GOAP) Flush Dimmer ZMNHDA not exposing groups

NINE78, 2021-02-09:
Dear all,

new zwave-js user, running on Home Assistant using supervised zwavejs2mqtt docker. I have a number of Qubino Flush Dimmers (ZMNHDA) devices which do not update the controller. I think it may be due to the fact none of the association groups are recognized.

When adding/refreshing the node through the zwavejs2mqtt interface, the following appears in the log:
2021-02-09 09:05:28.917 WARN ZWAVE: Node 10 doesn't support groups associations

Whereas I need group 4 to report to the controller, as per documentation at https://opensmarthouse.org/zwavedatabase/209.

Most appreciated.

AlCalzone, 2021-02-09:
Please choose an appropriate issue template and fill it out completely when opening an issue. That helps us investigate your issue quicker because all the necessary information is there. Otherwise, the issue will go to the bottom of the pile - and that pile is growing quickly.

> Node 10 doesn't support groups associations

@robertsLando that looks like a message from zwavejs2mqtt

I'm not sure where that comes from since the config file contains this:
```jsonc
	"associations": {
		"1": {
			"label": "multilevel",
			"maxNodes": 4
		},
		"2": {
			"label": "basic on/off triggered by the input I2 state",
			"maxNodes": 4
		},
		"3": {
			"label": "basic on/off triggered by the input I3 state",
			"maxNodes": 4
		},
		"4": {
			"label": "Controller updates",
			"maxNodes": 4,
			"isLifeline": true
		}
	},
```

NINE78, 2021-02-09:
Providing further info:

**Device information**

Qubino (GOAP) ZMNHDA
Node IDs 18, 54, 64 are affected

**Last Known Working Configuration**
- [X] Previously working device (other platform)
    - Which platform? --> Home Assistant using OpenZWave
    - Have you made any recent configuration changes to the device? Describe. --> NO
    
**Installation information**
How did you install `node-zwave-js`?

- [ X] Other: zwavejs2mqtt as a supervised add-on in Home Assistant

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ZWaveJS2MQTT control panel
2. Click on  affected device
3. Click on "Groups"
4. Observe "Group" dropdown says "No data available"

**Additional context**
I've tried excluding/including device, as well as re-interviewing node multiple times

**Logfile:**
Attached
[zwavejs_473.log](https://github.com/zwave-js/node-zwave-js/files/5953122/zwavejs_473.log)

AlCalzone, 2021-02-09:
Can you re-interview one of those three nodes once and send me the log of that? Unfortunately the interview after startup does not include the info I'm looking for.

NINE78, 2021-02-09:
Adding another comment to this (sorry!): the WARN message in the original post is indeed from zwavejs2mqtt, and is caused by a catch from "driver.controller.getAssociationGroups(...)" calls in ZWaveClient.js

NINE78, 2021-02-09:
> Can you re-interview one of those three nodes once and send me the log of that? Unfortunately the interview after startup does not include the info I'm looking for.

The tail of the log should be a re-interview of node 064

AlCalzone, 2021-02-09:
Huh, there it is:
```
18:09:53.256 CNTRLR « [Node 064] node info received
                      supported CCs:
                      · Multilevel Switch
                      · All Switch
                      · Binary Switch
                      · Meter
                      · Multilevel Sensor
                      · Multi Channel
                      · Powerlevel
                      · Association
                      · Configuration
                      · Manufacturer Specific
                      · Version
                      controlled CCs:
                      · Basic
                      · Multilevel Switch

...

18:09:53.997 SERIAL « 0x010a000400400486148500a2                                          (12 bytes)
18:09:54.001 DRIVER « [Node 064] [REQ] [ApplicationCommand]
                      └─[VersionCCCommandClassReport]
                          CC:      Association
                          version: 0
18:09:54.006 CNTRLR   [Node 064]   does NOT support CC Association (0x85)
```

The device contradicts itself. As a workaround we could do the same as in https://github.com/zwave-js/node-zwave-js/pull/1666.
I would advise you to report this to the Qubino support though. Last time I talked to them, they seemed to be quite interested in having their devices work with zwave-js.

NINE78, 2021-02-09:
Aha, good catch! So we could basically hard-code it to support CC Associations, post device interview?

If that could be provided for, that’d be great. I’ve rolled back to OZW in the meantime as any automations based on knowing the state of these devices is broken. Do I need to do anything to move this patch forward?

I’ll reach out to Qubino and see what they come back with.

Most appreciated!!

AlCalzone, 2021-02-09:
If you change the corresponding file, restarting or re-interviewing should do the trick

NINE78, 2021-02-10:
So I added the following to zmnhda.json in /config/0x0159/

```
	"compat": {
		"commandClasses": {
			"add": {
				"0x85": {
					"isSupported": true
				}
			}
		}
	}
}
```

Zwave2mqtt no longer throws the warning upon re-interview, but I'm still not seeing the groups become populated...

Any tips very welcome!!

AlCalzone, 2021-02-10:
New log please :)

NINE78, 2021-02-10:
Please find it attached. Even though the zwavejs2mqtt log no longer shows the warning, the zwavejs log (attached)
[zwavejs_578.log](https://github.com/zwave-js/node-zwave-js/files/5957384/zwavejs_578.log)
 still shows 

```
11:05:07.574 DRIVER « [Node 064] [REQ] [ApplicationCommand]
                      └─[VersionCCCommandClassReport]
                          CC:      Association
                          version: 0
11:05:07.578 CNTRLR   [Node 064]   does NOT support CC Association (0x85)

```

Can you confirm that the above patch I made to zmnhda.json is in fact what's supposed to do the trick?

thanks!!
Nik.

AlCalzone, 2021-02-10:
Yes, the query is still done as part of the Version CC interview, but the support info is overwritten afterwards. Later in the log you see:
```
11:00:22.982 CNTRLR « [Node 064] received information for association group #2:
                      maximum # of nodes: 16
                      currently assigned nodes: 
```

NINE78, 2021-02-10:
Ok,thanks!

The plot thickens, as when I manipulate the physical switch, I see:

```
13:20:12.233 SERIAL « 0x0110000400400a3202213400000000000084                              (18 bytes)
13:20:12.237 CNTRLR   [Node 064] [~] [Meter] value[66049]: 10.5 => 0                    [Endpoint 0]
13:20:12.238 CNTRLR   [Node 064] [~] [Meter] deltaTime[66049]: 0 => 0                   [Endpoint 0]
13:20:12.239 SERIAL » [ACK]                                                                   (0x06)
13:20:12.241 DRIVER « [Node 064] [REQ] [ApplicationCommand]
                      └─[MeterCCReport]
                          type:       Electric
                          scale:      W
                          rate type:  Consumed
                          value:      0
                          time delta: 0 seconds
13:20:12.246 CNTRLR   [Node 064] [~] [Meter] value[66049]: 10.5 => 0                    [Endpoint 1]
13:20:12.253 CNTRLR   [Node 064] [~] [Meter] deltaTime[66049]: 0 => 0                   [Endpoint 1]
```

Endpoint 1 isn't advertised through zwavejs2mqtt (after re-interview) and upon first reception of these updates, entries for 64-38-1 are added. Although I see both updates for Endpoints 0 and 1 in zwavejs, the Endpoint 0 isn't updated in zwavejs2mqtt. Probably something to track further on that end...

Thanks for all your help!

closed 2021-02-10 by AlCalzone