zwave-js/zwave-js #1671: [bug] Eurotronic Spirit devices failed inclusion

nau-tic, 2021-02-09:
### Version

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 1.1.0
zwavejs version: 6.1.3

### Describe the bug

When adding multiple Eurotronic Spirit devices, some of them are included but not working and others fail to be included at all.

### To Reproduce

Steps to reproduce the behavior:

I'm setting up a Z-Wave network from scratch using zwavejs2mqtt version 1.1.0 (tried also with 1.0.5), however not successfully.

Controller is the Aeotec Z-Stick Gen5+ (version supporting Raspberry Pi 4). The clients are all Eurotronic Spirit (supporting Z-Wave Plus).

By doing inclusion per software (zwavejs2mqtt) it seems as some devices are added correctly, whereas others are not managing to get included.

I ruled out a hardware issue, as inclusion with the controller stick itself (when not connected to software) is working fine. Problem if I do this, is that when connecting the stick to zwavejs2mqtt all devices are reported as dead - and logs indicate no reply to ping.

This is the routine I've tried:
- Exclude each one of the devices with the controller stick itself
- Reset the Eurotronic Spirit devices
- Reset the controller stick
- Connect the stick into the computer, create & start new zwavejs2mqtt container 
- Disable MQTT Gateway (I'm only interested on integration with Home Assistant)
- Enable logging to file with "silly"  level both for zwavejs2mqtt and zwavejs
- For each Eurotronic Spirit..
      - Start secure inclusion in zwavejs2mqtt
      - Start inclusion in Eurotronic Spirit device

As a result I get:
- Controller stick is shown in device list, all good
- 1 device is shown as Alive, Securely connected, but without Device ID and marked as unknown manufacturer/product code
- 1 device is shown as Alive, not-securely connected, also without Device ID and marked as unknown manufacturer/product code
- 2 devices are not being included (timeout both on zwavejs2mqtt and on device)

Doing a node re-interview doesn't change anything.
      

### Expected behavior

Be able to add Eurotronic Spirit devices via the web-based node inclusion, see them on the node list, and finally be able to integrate them with Home Assistant.

### Additional context

See attached the corresponding logs:
[zwavejs_1.log](https://github.com/zwave-js/zwavejs2mqtt/files/5946751/zwavejs_1.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/zwavejs2mqtt/files/5946752/zwavejs2mqtt.log)

robertsLando, 2021-02-09:
I'm seeing lot of

`Timed out while waiting for a response from the node`

I think this could be a netowrk/device issue?

AlCalzone, 2021-02-09:
While the driver asks for the securely supported commands, the node requuests a nonce twice in a row. This causes the first one to be invalidated (as the specifications mandate), but the device uses the invalid one to encrypt its response.
As a result, the request for secure commands times out and the interview continues without security - which leaves us with very little information to work with.

```
21:08:26.212 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]

21:08:26.224 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      12
                      └─[SecurityCCNonceReport]
                          nonce: 0x2efae001a3343de4

21:08:26.241 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]

21:08:26.318 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      13
                      └─[SecurityCCNonceReport]
                          nonce: 0x1aa277171a6e9d68

21:08:26.344 DRIVER   Dropping message with invalid data (Reason: Nonce 0x2e expired, cannot decode 
                      security encapsulated command.):
                      0x012f0004000227988104edf07c3f59c09725b8dcb47048c207ef5ef574f6ff97aa27c56f432e
                      00d08c630f6ac227dd0032

21:08:36.218 CNTRLR   [Node 002] Timed out while waiting for a response from the node
21:08:36.219 CNTRLR   [Node 002] Querying securely supported commands timed out, skipping Security i
                      nterview...
```

I suggest you re-interview the device and hope it doesn't do that every time.

nau-tic, 2021-02-09:
Thank you! With a node re-interview the device finally appeared correctly (Node 002).

For Node 003 it took like 10 re-interviews to finally get the the manufacturer/device detected. Errors appeared as:
             - "Security bootstrapping failed, the node is included insecurely: a secure inclusion timer has elapsed"
             - "Timed out while waiting for a response from the node",
             -  Node replies to "CC version for Version" but other messages are either missing of with many messages back
             - "ConfigurationCC: skipping interview because CC version is < 3 and there is no config file"

Now it appears as securely connected. Is that possible? I.e. after including a node, a re-interview can cause to switch to secure inclusion.

For Node 004 and Node 005:
             - Not added when using zwave-js, but works out if done with the controller removed from the usb port using the inclusion by hardware
             - There was no relevant log entry as the controller with zwave-js was unable to include 
             - After connecting the controller to usb again, it appears in the web-UI as dead  (however nodes show connected symbol on the HW display)it
             - A day later I tried to re-interview them. There are several messages without apparent reply, but also actual responses to protocol info. 

I assume it's not a coverage issue as it's about 10-15m distance with almost line-of-sight and the node map shows all devices with 1 hop.

Changing the command timeout to 60s seems not to work, e.g. pings are expired still after 10s (or does this setting change something else?).

In general it seems as nodes reply to protocol info but the following controller's ping is not replied. Any idea what can I check next?

Attached new logs with current status. The UI now looks as:
[image]

PS - any idea why when going to the web-UI, selecting two logs and pressing the download button one gets a "zwavejs2mqtt-store.xlsx" file? It's actually a ZIP file I can unzip and see the content.

[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5954135/zwavejs2mqtt.log)
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5954136/zwavejs_1.log)

nau-tic, 2021-02-10:
Hello,

updated today to zwave-js 6.2.0  // zwave2mqtt 1.1.1. This solves the issue. After a node re-interview all devices are properly detected, appearing alive and working.

[image]

closed 2021-02-10 by nau-tic