zwave-js/zwave-js #1674: Ability to set RF region for Z-Wave 700 series USB sticks like the SLUSB001A

Hedda, 2021-02-09:
Please consider adding support for adjusting/setting "RF region" for Z-Wave 700 series USB sticks/dongles like the SLUSB001A.

Adapters like SLUSB001A (SLUSB7000A) UZB-7 USB stick should be able to support all radio frequencies by setting "RF regions".

As I understand it, the host must change the actual "RF region" to be used. 

https://www.silabs.com/community/wireless/z-wave/knowledge-base.entry.html/2019/03/01/z-wave_700_how_toc-iQk2

https://www.silabs.com/community/wireless/z-wave/forum.topic.html/uzb_7_backwards_compatiblewithuzb3-mFu8

RF region information will then be stored permanently on the SLUSB UZB-7 stick after being set by the host controller:

https://www.silabs.com/wireless/z-wave/global-regions

PS: This is a follow-up feature request to https://github.com/zwave-js/node-zwave-js/issues/1357 which was about adding support for SLUSB001A (SLUSB7000A) UZB-7 USB stick.

closed 2021-05-04 by AlCalzone

Hedda, 2021-09-06:
According to this reddit post you can change region, however unlike SLUSB001A, Aeotec has tuned antenna hardware for each region, which should mean that changing from EU frequency to the USA frequency or vice versa would not be recommended because their frequencies difference so much, but smaller changes like from EU frequency to Russia frequency should work fine.

https://www.reddit.com/r/homeassistant/comments/pcqkt4/has_anyone_been_successful_in_altering_the_region/

> The steps to change it are:
> 
> In zwavejs2mqtt, go to Actions at the top of the nodes page, then Advanced. Go to Driver Function. Delete the example code and paste the below, changing the region code to your needs.
> 
> To list your region:
> 
> const region = await driver.controller.getRFRegion();
> 
> console.log(\Controller region is ${region}\);
> 
> To change your region:
> 
> const region = 0x00; // Europe
> 
> const result = await driver.controller.setRFRegion(region);
> 
> console.log(\Setting region resulted in ${result});
> 
> Region codes:
> 
> Europe = 0x00
> 
> USA = 0x01 [This is what Aeotec ships set as for US]
> 
> USA (Long Range) = 0x09
> 
> Australia/New Zealand = 0x02
> 
> Hong Kong = 0x03
> 
> India = 0x05
> 
> Israel = 0x06
> 
> Russia = 0x07
> 
> China = 0x08
> 
> Japan = 0x20
> 
> Korea = 0x21
> 
> Unknown = 0xfe
> 
> Default (EU) = 0xff
> 
> Note: These come from the spec, so not all may be actually supported.

@altShiftDev Probably best you post a new discussion asking about that in https://github.com/zwave-js/node-zwave-js/discussions

node-zwave-js v7.4.0 release did list "**_Implement get/setPowerlevel, get/setRFRegion controller methods_**" reffering to https://github.com/zwave-js/node-zwave-js/pull/2500

https://github.com/zwave-js/node-zwave-js/releases/tag/v7.4.0

If you do post https://github.com/zwave-js/node-zwave-js/discussions then link to https://github.com/zwave-js/node-zwave-js/issues/1674 and https://github.com/zwave-js/node-zwave-js/pull/2500 as well as https://github.com/zwave-js/node-zwave-js/issues/1245 as reference.

AlCalzone, 2021-09-06:
As a matter of fact, I did a few days ago. I switched mine from EU to US and back. While on the US frequency, I didn't see any traffic, so it definitely worked.

PR #2500 (2021-05-04): feat: implement get/setPowerlevel, get/setRFRegion, auto-enable TX status reports
With this PR, we automatically configure supporting sticks to give extended TX status reports we'll be using in a later step.
Furthermore, this adds the methods get/setPowerlevel, get/setRFRegion to configure the stick's radio.

fixes: #1673
fixes: #1674
fixes: #1357