zwave-js/zwave-js #1677: [bug] Issue with multiple reports from specific devices

billiaz, 2021-02-09:
I am facing an issue with my Silabs 700 stick (UZB7)
some nodes do send multiple notification reports sometimes a few others up to 8-10 at a time.

I am attaching the logs
Nodes 113, 97,88  and 35 are having this issue

Running Zwavejs2mqtt from master build today. tried also 1.0.4 and 1.0.1, still same issue
z-wavejs library is 6.1.3

Tried:
reinterview, healnode and heal network. 
I am one step before re-including device. and check if this changes anything.

Log file from beginning (startup)
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/5953885/zwavejs2mqtt-store.zip)

deccos, 2021-02-09:
I am experiencing something similar.  I have 2 specific nodes- both mains powered Aeotecc Sensor 6's that are way chattier than everything else, using an Aeotec USB stick for teh controller, all running in Docker on a NUC.  

One of the nodes (node 5 in enclosed log) is accounting for something like 65% of all log entries in my zwavejs2mqtt log.  Both it and the zwavejs_1.log file enclosed
[zwavejs2mqtt-store (2).zip](https://github.com/zwave-js/node-zwave-js/files/5953958/zwavejs2mqtt-store.2.zip)

AlCalzone, 2021-02-10:
@deccos Most of what I see in your file are the automatic reports which can be configured. You should lower the report interval if that is too frequent for you.

@billiaz Please update, you're running a 6.0.0 beta.

deccos, 2021-02-10:
yeah, i excluded node 5 last night and reset it before reinclusion.  it no longer shows as being chatty in the logs.  instead, Node11 is chatty - my second Aeotec Sensor 6.  I must have twekaed config settings on them at some point in the past.  I had been through a Fibaro home centre controller, then the aeotec stick on zwave, OZW and lately zwave-js.  my own fault really !.

now to reset node 11...

thanks for the help @AlCalzone

billiaz, 2021-02-10:
@AlCalzone faced this on 6.1.3 too no difference there, I will try 6.2.0

AlCalzone, 2021-03-03:
This seems to happen with many different devices. Usually, excluding, resetting and re-including resolves the problem. So I assume it has to do with how the devices were originally included.

closed 2021-03-03 by AlCalzone