zwave-js/zwave-js #1678: Heatit Z-Dim doesn't restore "current value" when switching on "level"

jmgiaever, 2021-02-09:
Hi,

I own 4 HeatIT Z-Dim dimmers. I normally use the level entity (multilevel switch up/down in zwavejs) to turn ON or OFF the lights, and not the binary ON/OFF. 

While this works well for 1 device it doesn't work correctly with the 3 others. The lights are turning ON visually, but their «current value» aren't updated with the targeting value, so programmatically they appear as OFF.

**Device information**

Which device(s) is/are affected (make/model)?  HeatIT Z-DIM ThermoFloor

What are the node IDs?
1. 37
2. 42
3. 43
4. 44

**Last Known Working Configuration**
- [x ] New device

- [ ] Previously working device (other platform)
    - Which platform? 
       - OZW Daemon
    - Have you made any recent configuration changes to the device?
    - **No, all of them uses them default config.**
I've tried to compare the configurations, but they all looks the same.
    
**Installation information**
How did you install `node-zwave-js`?
- [x ] Other: Snap package of the latest build of Zwavejs2mqtt (1.1.0)

**Additional context**
Add any other context about the problem here.

**Logfile:**

Here's the code when I press the two identical toggles on node 42 (working correctly) and node 43 (which isn't working correctly):

```
23:14:57.203 SERIAL » 0x010e00132a076c01ba032601ff259375                                  (16 bytes)
23:14:57.205 DRIVER » [Node 042] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      147
                      └─[SupervisionCCGet]
                        │ session id:      58
                        │ request updates: true
                        └─[MultilevelSwitchCCSet]
                            target value: 255
23:14:57.209 SERIAL « [ACK]                                                                   (0x06)
23:14:57.215 SERIAL « 0x0104011301e8                                                       (6 bytes)
23:14:57.215 SERIAL » [ACK]                                                                   (0x06)
23:14:57.216 DRIVER « [RES] [SendData]
                        was sent: true
23:14:57.380 SERIAL « 0x011800139300001101bebe7f7f7f0101031d0000000101000016              (26 bytes)
23:14:57.381 SERIAL » [ACK]                                                                   (0x06)
23:14:57.383 DRIVER « [REQ] [SendData]
                        callback id:     147
                        transmit status: OK
23:14:57.504 SERIAL « 0x010d0004002a056c023aff00be00cc                                    (15 bytes)
23:14:57.506 SERIAL » [ACK]                                                                   (0x06)
23:14:57.508 DRIVER « [Node 042] [REQ] [ApplicationCommand]
                      └─[SupervisionCCReport]
                          session id:          58
                          more updates follow: false
                          status:              Success
                          duration:            [Duration: 0seconds]
23:14:58.686 SERIAL « 0x010d0004002a052603323200be0042                                    (15 bytes)
23:14:58.688 CNTRLR   [Node 042] [~] [Multilevel Switch] targetValue: 255 =&gt; 50         [Endpoint 0]
2021-02-09 23:14:58.689 INFO ZWAVE: Node 42: value updated: 38-0-targetValue 255 =&gt; 50
23:14:58.690 CNTRLR   [Node 042] [~] [Multilevel Switch] duration: {"value":0,"unit":"s [Endpoint 0]
                      econds"} =&gt; {"value":0,"unit":"seconds"}
[... 176 lines cut ...]
23:15:14.134 DRIVER « [REQ] [SendData]
                        callback id:     153
                        transmit status: OK
23:15:14.155 SERIAL « 0x01210004002b19988158afb3d1e61f00625612638d19899c0b7166d38fc7fcdec (35 bytes)
                      40038
23:15:14.157 SERIAL » [ACK]                                                                   (0x06)
23:15:14.159 DRIVER « [Node 043] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[SupervisionCCReport]
                            session id:          59
                            more updates follow: false
                            status:              Success
                            duration:            [Duration: 0seconds]
23:15:47.221 SERIAL « 0x010a00040013029840c300fb                                          (12 bytes)
23:15:47.223 SERIAL » [ACK]                                                                   (0x06)
23:15:47.225 DRIVER « [Node 019] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
23:15:47.236 SERIAL » 0x01110013130a98805b57fd8d121160f6059a8a                            (19 bytes)
23:15:47.237 DRIVER » [Node 019] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      154
                      └─[SecurityCCNonceReport]
                          nonce: 0x5b57fd8d121160f6
```

AlCalzone, 2021-02-09:
There must be **some** difference between the devices. Node 42 reports the change on its own, Node 43 doesn't.

While we are normally optimistic and update the `currentValue` with the `targetValue` when the set command succeeded, that is not possible for the value `255`, which does not convey a level, but just means "on".

I guess we could poll even when supervision tells us that the "on" command worked, but since it works for one of your devices I'd try to figure out why. Have you compared the associations too?

jmgiaever, 2021-02-09:
Hi,

for what I'm aware there aren't. They came with the same order - who knows, and how can I check this? 

They all worked by toggling the level in OZW. The level value was restored and they appeared to be ON. 

I'll do some more toggling while watching the logs to see if something's comes up :)

jmgiaever, 2021-02-09:
Here's some toggling on node 43, one of the nodes that doesn't report correctly (sorry for the other garbage in here from other nodes):

```
00:19:14.886 DRIVER « [REQ] [SendData]
callback id: 80
transmit status: OK
00:19:14.906 SERIAL « 0x01120004002b0a9880e72934bfe27047cac6004c (20 bytes)
00:19:14.908 SERIAL » [ACK] (0x06)
00:19:14.910 DRIVER « [Node 043] [REQ] [ApplicationCommand]
└─[SecurityCCNonceReport]
nonce: 0xe72934bfe27047ca
00:19:14.916 SERIAL » 0x012300132b1c988140b1d14f03799cc4e200e7538423323c88e77bcb1e2e90f53 (37 bytes)
dc5255155
00:19:14.916 DRIVER » [Node 043] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 81
└─[SecurityCCCommandEncapsulation]
│ nonce id: 231
└─[SupervisionCCGet]
│ session id: 35
│ request updates: false
└─[MultilevelSwitchCCStartLevelChange]
startLevel: 0 (ignored)
direction: up
00:19:14.922 SERIAL « [ACK] (0x06)
00:19:14.929 SERIAL « 0x0104011301e8 (6 bytes)
00:19:14.929 SERIAL » [ACK] (0x06)
00:19:14.930 DRIVER « [RES] [SendData]
was sent: true
00:19:14.958 SERIAL « 0x011800135100000300c37f7f7f7f010103000000000201000065 (26 bytes)
00:19:14.959 SERIAL » [ACK] (0x06)
00:19:14.961 DRIVER « [REQ] [SendData]
callback id: 81
transmit status: OK
00:19:14.981 SERIAL « 0x010a0004002b029840c300c3 (12 bytes)
00:19:14.982 SERIAL » [ACK] (0x06)
00:19:14.984 DRIVER « [Node 043] [REQ] [ApplicationCommand]
└─[SecurityCCNonceGet]
00:19:14.995 SERIAL » 0x011100132b0a98806d6c0973988c2f8c05525f (19 bytes)
00:19:14.996 DRIVER » [Node 043] [REQ] [SendData]
│ transmit options: 0x05
│ callback id: 82
└─[SecurityCCNonceReport]
nonce: 0x6d6c0973988c2f8c
00:19:14.999 SERIAL « [ACK] (0x06)
00:19:15.004 SERIAL « 0x0104011301e8 (6 bytes)
00:19:15.005 SERIAL » [ACK] (0x06)
00:19:15.006 DRIVER « [RES] [SendData]
was sent: true
00:19:15.030 SERIAL « 0x011800135200000300c47f7f7f7f010103000000000201000061 (26 bytes)
00:19:15.031 SERIAL » [ACK] (0x06)
00:19:15.033 DRIVER « [REQ] [SendData]
callback id: 82
transmit status: OK
00:19:15.061 SERIAL « 0x01210004002b199881654be276cc6691357c917f0950826d1ffcda577d3d86f9c (35 bytes)
600f6
00:19:15.063 SERIAL » [ACK] (0x06)
00:19:15.066 DRIVER « [Node 043] [REQ] [ApplicationCommand]
└─[SecurityCCCommandEncapsulation]
│ sequenced: false
└─[SupervisionCCReport]
session id: 35
more updates follow: false
status: Success
duration: [Duration: 0seconds]
2021-02-10 00:19:15.071 INFO ZWAVE: Success zwave api call writeValue { success: true, message: 'Success zwave api call', result: undefined }
00:19:31.737 SERIAL « 0x01120004001c0a32022144000000000000cd0067 (20 bytes)
00:19:31.739 CNTRLR [Node 028] [~] [Meter] value[65537]: 0 => 0 [Endpoint 0]
2021-02-10 00:19:31.740 INFO ZWAVE: Node 28: value updated: 50-0-value-65537 0 => 0
00:19:31.742 CNTRLR [Node 028] [~] [Meter] deltaTime[65537]: 0 => 0 [Endpoint 0]
[... 836 lines cut ...]
00:22:24.551 SERIAL « 0x010d0004002a056c022dff00bf00da (15 bytes)
00:22:24.553 SERIAL » [ACK] (0x06)
00:22:24.555 DRIVER « [Node 042] [REQ] [ApplicationCommand]
└─[SupervisionCCReport]
session id: 45
more updates follow: false
status: Success
duration: [Duration: 0seconds]
2021-02-10 00:22:24.560 INFO ZWAVE: Success zwave api call writeValue { success: true, message: 'Success zwave api call', result: undefined }
00:22:25.779 SERIAL « 0x010d0004002a052603636300bf0043 (15 bytes)
00:22:25.782 CNTRLR [Node 042] [~] [Multilevel Switch] targetValue: 0 => 99 [Endpoint 0]
2021-02-10 00:22:25.782 INFO ZWAVE: Node 42: value updated: 38-0-targetValue 0 => 99
00:22:25.784 CNTRLR [Node 042] [~] [Multilevel Switch] duration: {"value":0,"unit":"s [Endpoint 0]
econds"} => {"value":0,"unit":"seconds"}
2021-02-10 00:22:25.785 INFO ZWAVE: Node 42: value updated: 38-0-duration [Duration: 0seconds] => [Duration: 0seconds]
00:22:25.787 CNTRLR [Node 042] [~] [Multilevel Switch] currentValue: 0 => 99 [Endpoint 0]
2021-02-10 00:22:25.788 INFO ZWAVE: Node 42: value updated: 38-0-currentValue 0 => 99
00:22:25.789 SERIAL » [ACK] (0x06)
00:22:25.795 DRIVER « [Node 042] [REQ] [ApplicationCommand]
└─[MultilevelSwitchCCReport]
current value: 99
target value: 99
duration: [Duration: 0seconds]

```

jmgiaever, 2021-02-09:
> Have you compared the associations too?

Checked it now; `Lifeline -> Node_1` on all devices

jmgiaever, 2021-02-10:
Not sure if these will help, but found a «debug info» tab:
```
{
  "id": 42,
  "deviceId": "411-8704-3",
  "manufacturer": "ThermoFloor",
  "manufacturerId": 411,
  "productType": 3,
  "productId": 8704,
  "name": "Dining-table-light",
  "loc": "Kitchen",
  "values": [
    {
      "id": "42-38-0-targetValue",
      "nodeId": 42,
      "commandClass": 38,
      "commandClassName": "Multilevel Switch",
      "endpoint": 0,
      "property": "targetValue",
      "propertyName": "targetValue",
      "type": "number",
      "readable": true,
      "writeable": true,
      "label": "Target value",
      "stateless": false,
      "min": 0,
      "max": 99,
      "list": false,
      "value": 99,
      "lastUpdate": 1612914295459,
      "newValue": 99
    },
    {
      "id": "42-38-0-duration",
      "nodeId": 42,
      "commandClass": 38,
      "commandClassName": "Multilevel Switch",
      "endpoint": 0,
      "property": "duration",
      "propertyName": "duration",
      "type": "duration",
      "readable": true,
      "writeable": true,
      "label": "Transition duration",
      "stateless": false,
      "list": false,
      "value": {
        "value": 0,
        "unit": "seconds"
      },
      "lastUpdate": 1612914295462,
      "newValue": {
        "value": 0,
        "unit": "seconds"
      }
    },
    {
      "id": "42-38-0-currentValue",
      "nodeId": 42,
      "commandClass": 38,
      "commandClassName": "Multilevel Switch",
      "endpoint": 0,
      "property": "currentValue",
      "propertyName": "currentValue",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Current value",
      "stateless": false,
      "min": 0,
[... 2406 lines cut ...]
  "failed": false,
  "lastActive": 1612914289490,
  "interviewCompleted": true,
  "firmwareVersion": "1.0",
  "isBeaming": true,
  "isSecure": true,
  "keepAwake": false,
  "maxBaudRate": null,
  "isRouting": true,
  "isFrequentListening": false,
  "isListening": true,
  "status": "Alive",
  "interviewStage": "Complete",
  "productLabel": "HeatIt ZDim",
  "productDescription": "Knob Smart Dimmer",
  "zwaveVersion": 4,
  "deviceClass": {
    "basic": 4,
    "generic": 17,
    "specific": 1
  },
  "hexId": "0x019b-0x2200-0x0003",
  "_name": "Stairway-lights (Kitchen)"
}
```

jmgiaever, 2021-02-10:
Sorry for all the info, but tried to do a device reset and got an error saying that "node 1 do not support associations". I'm having the [UZB-1](https://z-wave.me/products/uzb/) dongle from Zwave.me.

I haven't added any associations to this (or any) node.

AlCalzone, 2021-02-10:
Please attach logs as files. These inline logs make it **really** hard to digest the issue.
@robertsLando where can he check in z2m how the associations are set up?

robertsLando, 2021-02-10:
Control Panel > Expand a node in the table > Group Associations Tab

jmgiaever, 2021-02-10:
> Please attach logs as files. These inline logs make it **really** hard to digest the issue.
> @robertsLando where can he check in z2m how the associations are set up?

Hi, [I already checked](https://github.com/zwave-js/node-zwave-js/issues/1678#issuecomment-776324307) it for the dimmers. Node 1 (the controller) is in the Lifeline group of the devices.

There is no group association options for the controller, so nothing comes up there.

AlCalzone, 2021-02-10:
Okay before we start guessing more, please send me your cache files (`<some-hex-key>.json` and `.values.jsonl`)

jmgiaever, 2021-02-10:
Yes, are they safe to publish here or email them to you? 

I also see that I have 3 different sets, probably because I tried to rebuild the whole network yesterday. How can I find out which is «current»?

AlCalzone, 2021-02-10:
You can send them via email

jmgiaever, 2021-02-10:
@AlCalzone Sent! :)

AlCalzone, 2021-02-10:
Ok a couple of observations:
* Both 42 and 43 support Z-Wave+ Info V2, which means they **have to** report all status updates via the lifeline.
* Node 42 reports Association CC as version 2, node 43 as version 0 (i.e. unsupported). The same happens for a couple of other CCs.
* For Node 37 we don't have any version information besides the CCs
* The associations are the same for all 4 nodes.

I'm guessing that you're losing some messages, maybe due to RF issues. Since your log snippet in the OP is very short, I can't verify though.
Could you make and attach another zwave-js logfile (only that, no logs from zwavejs2mqtt please) including the entire startup and some switching attempts?

jmgiaever, 2021-02-10:
Hi, will do! :)

I'm not sure if it's noteworthy, but three of the dimmers is approx 1 meter away from the controller. And node 42 and node 43 is located in a rack, with node 43 (which is also problematic). 

Node 37 on the other hand is maybe 3 meters away, in the basement on an angle of 45⁰. 

The Z-wave graph shows that all of these devices has 1 hop. That shouldn't be problematic, but I do not understand how/why, since they're so close to the controller.

I'm just upgrading zwavejs2mqtt to the recent version. Will enable logging and save it to file as soon as the compilation is done. :)

You'll hear from me asap.

jmgiaever, 2021-02-10:
Log entry sent! ~9MB so it might take few minutes to arrive. 

At (time)
1. ~18:05 All powered devices interviewed and battery devices is restored from cache
2. ~18:17 I'll do a re-interview of all powered devices
3. ~18:18 I'll start toggling. I've tried to repeat the same process for each device, each time. E.g do the same on node 42, 43, 37, 44 before trying to do something else. 

For one of the iteration I set the target value, and "the level" then responds correctly.

AlCalzone, 2021-02-10:
Hah - Node 42 is the only one of those 4 that is not included securely. I'll do some investigations what that could imply in regards to the Z-Wave+ CC V2 requirements.

jmgiaever, 2021-02-10:
Ah, I'll see that now. It wasn't visual from the UI earlier, but[ with the upgrade](https://github.com/zwave-js/zwavejs2mqtt/releases/tag/v1.1.1) I just did... I'll see it too. The logs aren't self-explanatory to me :package:

jmgiaever, 2021-02-10:
Saw the discussion on this matter; Don't know if it's worth mentioning, but with OZW they were working with S0 and they were one of the snappiest devices I had in my system, however they had a habit of reporting their state to often. Are they supposed to get an ACK back to confirm that the data was received.

PS! When I tried to look up a thread that I once read regarding the spamming issue, I found this regarding [OZW + Multilevel CC](https://github.com/OpenZWave/open-zwave/issues/1321). Can this help in any way?

AlCalzone, 2021-02-10:
OZW used to poll a lot. When using Multilevel Switch CC with Supervision (which OZW does not support), that is usually not necessary, because we know the device has reached its target state. The only exception is ON (255) where we don't know which value the device has chosen.
But since the Z-Wave+ V2 devices **must** report their changes (and 42 does), I want to figure out why the secure nodes don't. Such modern devices shouldn't require polling workarounds...

jmgiaever, 2021-02-10:
Thanks for the clarification :) 

I don't mind having a door or window sensor insecure, since that's something everyone without any technical background can see from the street (haha, read a post where this was your argument about a shutter - good one!). 

But we are planning to establish this in all our short term rentals and don't want personal data (motion sensors etc) to be readable to an outside party.

We also strive to keep costs and environmental damages to the lowest, so anything that can be flipped (turned on/off) should be secure. Some of our rentals is up to 48 000 kWh a year. :|

Again! Thanks for all your work. I've put up a monthly sponsor.

AlCalzone, 2021-02-14:
Okay from what I've found out: the 3 devices included with Security S0 are not behaving correctly. If you don't mind, I'd re-include them insecurely and try Security again once Security S2 is implemented.

jmgiaever, 2021-02-14:
Hmm, I think it's strange that they worked flawlessly on OZW. :/ 

What the status on S2 if you dont mind me asking?

AlCalzone, 2021-02-14:
That's probably because OZW verified all changes through polling by default, whether that is necessary or not. The ultimate goal of zwave-js is to get certified, so polling Z-Wave+ V2 devices would be a no-go.

I believe I've solved the hard parts of S2 already. Need to put it all together and test extensively. Unfortunately, the last couple of months were busy investigating the influx of issues from new HASS and zwavejs2mqtt users, so S2 got stuck on the sidelines.

jmgiaever, 2021-02-14:
I can imagine! Excited to follow the path of Zwave-js. Let me know if you need real life testing when you'll get that far.

closed 2021-02-18 by zwave-js-assistant[bot]

jmgiaever, 2021-02-22:
Hi @AlCalzone. I excluded all of the HeatIt ZDim devices and made a new/second Z-wave network with these devices only. 

Node 42 (it has a new ID in the new network, but using this for reference) still works correctly in secure mode. Isn't it odd that this node works correctly, but the other ones don't?

jmgiaever, 2021-02-22:
Now also tried to exclude one of the weird devices and include it insecurely. Same behavior as when it was added securely.

jmgiaever, 2021-02-22:
```
It looks like all the data received is the same, except the last report. Node 42 sends a «MultilevelSwitchCCReport» while the other's sends a «MeterCCReport». Node 42 does not send this... However this is always 0.

Node 2 = 42 (before)
Node 3 = 43 (before)

2021-02-22 17:15:47.900 INFO ZWAVE: Node 2: value updated: 38-0-targetValue 0 => 99
2021-02-22 17:15:47.901 INFO ZWAVE: Node 2: value updated: 38-0-duration [Duration: 0seconds] => [Duration: 0seconds]
2021-02-22 17:15:47.902 INFO ZWAVE: Node 2: value updated: 38-0-currentValue 0 => 99
17:15:47.902 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[MultilevelSwitchCCReport]
                            current value: 99
                            target value:  99
                            duration:      [Duration: 0seconds]
```

```
2021-02-22 17:16:31.775 INFO ZWAVE: Node 3: value updated: 50-0-value-65537 0 => 0
2021-02-22 17:16:31.776 INFO ZWAVE: Node 3: value updated: 50-0-deltaTime-65537 0 => 0
17:16:31.776 DRIVER « [Node 003] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[MeterCCReport]
                            type:       Electric
                            scale:      kWh
                            rate type:  Consumed
                            value:      0
                            time delta: 0 seconds
```

AlCalzone, 2021-02-22:
Is there anything you can configure on these devices, like which reports are getting sent?
If not, I'd advise you to bug the support, they should have some more insight.

jmgiaever, 2021-02-22:
Everything is equal in the configuration part of Zwavejs2mqtt UI, and nothing that i related to reporting. 

However... I [found a post](https://www.hjemmeautomasjon.no/forums/topic/7398-heatit-z-dim-2019-b-vs-2020-a-manglede-rapportering/?tab=comments#comment-84714) (written in Norwegian, my own lang.) from last Friday. He's writing that there's a new 2020A version which doesn't report changes in HomeSeer, while the 2019 version does.

I'm not sure which versions I have; however I bought them by the end of 2020. I removed the cover of the rack and can see some differences here. The QR code of node 42 (top) and node 43 (bottom) differs, and the text have different sizes. 

[image]

So I guess node 42 is probably 2019 version while the others I have is 2020. 

I've tried to find a complete manual for the 2020 version to see if it has more settings/configurations that is not included in the Zwavejs config. But I can only find the QuickStart guide for it, and its just a simple install guide mainly describing how to put it to inclusion mode.

I'm wondering if there is firmwares available somewhere. Tried to google but cant find any.

AlCalzone, 2021-02-22:
I've removed your images as you did not blur the DSK under the QR codes.

> So I guess node 42 is probably 2019 version while the others I have is 2020. 

That could explain it. Like I said, it is probably best to ask the Heatit support what's going on.

jmgiaever, 2021-02-22:
I've written them and linked to this post + the forum post that I found. 

Thanks for your patience with me :)

jmgiaever, 2021-02-26:
Just leaving a comment that the new OTA from Thermo-Floor / HeatIt solved this issue. I'm not able to attach the file here, but reaching out to them took a day until I had the file.

irowan, 2021-04-23:
> Just leaving a comment that the new OTA from Thermo-Floor / HeatIt solved this issue. I'm not able to attach the file here, but reaching out to them took a day until I had the file.

Hi! 

Can you send it to me as well? I contacted you through Home Assistant Norge on Facebook, but the messaage probably went to the spam folder. :)