zwave-js/zwave-js #1690: Infinite searching for parameters.

Petro31, 2021-02-10:
**Describe the bug**

When searching the LZW45 for parameters, the system enters an infinite loop.  It cycles all parameters indefinitely, adding megabytes of text in the log every minute.  This could be a firmware issue on the LZW45, however the system should have a safeguard against this.

The bug is one or the other:

* When an unknown parameter is met, the system restarts at parameter 1.  I don't know if this is true, but the logs indicate this *might* be the case.

or 

* If a device incorrectly tells the system that the next parameter is the first parameter, the system should not start over at the first parameter.

Here's a snippit of the area in the log:

```
07:12:30.304 CNTRLR Â» [Node 003] querying parameter #51 information...
07:12:30.307 DRIVER Â» [Node 003] [REQ] [SendData]
                      â”‚ transmit options: 0x25
                      â”‚ callback id:      78
                      â””â”€[ConfigurationCCPropertiesGet]
                          parameter #: 51
07:12:30.315 DRIVER Â« [RES] [SendData]
                        was sent: true
07:12:30.328 DRIVER Â« [REQ] [SendData]
                        callback id:     78
                        transmit status: OK
07:12:30.340 DRIVER Â« [Node 003] [REQ] [ApplicationCommand]
                      â””â”€[ConfigurationCCPropertiesReport]
                          parameter #:         51
                          next param #:        240
                          value size:          1
                          value format:        UnsignedInteger
                          min value:           0
                          max value:           1
                          default value:       1
                          alters capabilities: false
                          readonly:            false
                          advanced:            false
                          bulk support:        false
07:12:30.345 DRIVER Â» [Node 003] [REQ] [SendData]
                      â”‚ transmit options: 0x25
                      â”‚ callback id:      79
                      â””â”€[ConfigurationCCNameGet]
                          parameter #: 51
07:12:30.355 DRIVER Â« [RES] [SendData]
                        was sent: true
07:12:30.367 DRIVER Â« [REQ] [SendData]
                        callback id:     79
                        transmit status: OK
07:12:30.379 DRIVER Â« [Node 003] [REQ] [ApplicationCommand]
                      â””â”€[ConfigurationCCNameReport]
                          parameter #:       51
                          name:              Button Delay
                          reports to follow: 0
07:12:30.383 DRIVER Â» [Node 003] [REQ] [SendData]
                      â”‚ transmit options: 0x25
                      â”‚ callback id:      80
                      â””â”€[ConfigurationCCInfoGet]
                          parameter #: 51
07:12:30.391 DRIVER Â« [RES] [SendData]
                        was sent: true
07:12:30.404 DRIVER Â« [REQ] [SendData]
                        callback id:     80
                        transmit status: OK
07:12:30.416 DRIVER Â« [Node 003] [REQ] [ApplicationCommand]
                      â””â”€[ConfigurationCCInfoReport]
                          parameter #:       51
                          info:              Button Delay
                          reports to follow: 0
07:12:30.418 CNTRLR Â« [Node 003] received information for parameter #51:
[... 118 lines cut ...]
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?

I've verified this on all installations, so I'll check every box.

- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Start any zwave js network with an included LZW45.  Or include an LZW45 on your network.

**Additional context**
Add any other context about the problem here.

**Logfile:**

Petro31, 2021-02-10:
Possible fixes:

* Create a list of known parameters while searching, if we reach the same param twice, break.

* Create a dictionary of parameters with a count for each parameter.  When the count of each one reaches 5, we know we are in an infinite loop, break

* create a timeout for each node, something like 2 minutes.  (personally not in favor of this one)

AlCalzone, 2021-02-10:
Fix up at #1692. Instead of only checking for `nextParam === 0`, we now also check for `next param < current param` to stop the loop.

Petro31, 2021-02-10:
I'll update my stream tonight and give this a whirl.

PR #1692 (2021-02-10): fix(cc): avoid infinite loop when scanning config params
fixes: #1690
closed 2021-02-10 by AlCalzone