zwave-js/zwave-js #1691: Sleeping devices take a long time to report "Ready" at start

steve28, 2021-02-10:
**Describe the bug**

When zwave-js starts, battery/sleeping devices are not reported ready (RestartFromCache) for a while. It appears it takes zwave-js a while to decide the device is asleep and revert to the cached info.  During this period, such devices then show as "unavailable" in HA. This is amplified for large networks.  Example - my 50+ device network with many batt devices takes ~15-20 min before all batt devices are available in Home Assistant.

I would expect a sleepy device to go almost immediately to RestartFromCache.

**Device information**

All batt powered devices.

**Last Known Working Configuration**
N/A
    
**Installation information**
How did you install `node-zwave-js`?
- [X] `zwavejs2mqtt` (latest) docker image (1.1.1)
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Steps to reproduce the behavior:
1. Restart server

**Additional context**
N/A

**Logfile:**
N/A

AlCalzone, 2021-02-10:
Duplicate of #1657

closed 2021-02-10 by AlCalzone

AlCalzone, 2021-02-11:
fixed in https://github.com/zwave-js/node-zwave-js/pull/1714