zwave-js/zwave-js #1698: Feature Request: Add method to update Driver logConfig

raman325, 2021-02-10:
**Is your feature request related to a problem? Please describe.**
As far as I can tell, the only way to change the log level is to restart the driver which triggers a full re-interview of all nodes. It appears that many Home Assistant users are changing log levels often to help troubleshoot integration issues or device specific issues. It would be great if we could update the log level without restarting the driver so we didn't have to trigger a full re-interview

**Describe the solution you'd like**
Add a method to the `Driver` class that can update the `logConfig`, particularly `level`, without having to destory and recreate the Driver.

**Describe alternatives you've considered**
None at this time

**Additional context**
Add any other context or screenshots about the feature request here. If you want to propose an API, share your thoughts how it could be used (syntax).

AlCalzone, 2021-02-10:
Damn, I think we had that and it got lost. @robertsLando?

AlCalzone, 2021-02-10:
The log container has this: https://github.com/zwave-js/node-zwave-js/blob/6cc6466e2ce050b95364b1740cc6d123af3cdba1/packages/core/src/log/shared.ts#L129

raman325, 2021-02-10:
hmm OK, I think this solves the problem then. I believe we'd just need to call `driver.logContainer.updateConfiguration` then? Seems like something we can expose in the server and then provide a Home Assistant service to call it

AlCalzone, 2021-02-10:
I think we need a method on the driver to update the config for all loggers

robertsLando, 2021-02-11:
> I think we need a method on the driver to update the config for all loggers

Yep absolutely, now I don't remember how it works but I thought it worked in that way, doesn't it?

AlCalzone, 2021-02-11:
We had that at the start but seem to have removed it when refactoring to log containers

PR #1754 (2021-02-14): feat: add ability to update logging configuration on the fly
fixes: #1698
closed 2021-02-14 by AlCalzone