zwave-js/zwave-js #1701: Name: Garage Door Controller  Model Number: ZW062

johnniemalm, 2021-02-10:
**Which device is missing?**

Name: Garage Door Controller

Model Number: ZW062

**What are the IDs?**

Manufacturer ID: ???????  
Product type: 0x0003 
Product ID: 0x003E 

**Did you check if we really don't have it?**

- [ ] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [ ] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [ ] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**
https://aeotec.freshdesk.com/support/solutions/articles/6000168304-garage-door-controller-technical-specifications-

https://products.z-wavealliance.org/products/1497?selectedFrequencyId=1

If so, it helps in creating the config file.

AlCalzone, 2021-02-10:
You forgot to check the checkboxes. If you had opened the link for the configuration DB and searched for ZW062, you would have found this: https://devices.zwave-js.io/?jumpTo=0x0086:0x0003:0x003e:0.0

closed 2021-02-10 by AlCalzone

johnniemalm, 2021-02-11:
But this is only as switch , not as garage-door ?

AlCalzone, 2021-02-11:
That has nothing to do with the config file. The barrier operator CC was only implemented very recently and HA needs to add support on their end too.

johnniemalm, 2021-02-11:
Ok, thanks for explaning this to me 👍