zwave-js/zwave-js #1711: Z-wave JS exits with RangeError [ERR_INVALID_BUFFER_SIZE]: Buffer size must be a multiple of 16-bits

shamimrahim, 2021-02-11:
**Describe the bug**

A clear and concise description of what the bug is. Describe what causes the bug, what you observe, and what happens as a result.

Once the docker container starts, it begins to query devices for approximately 10 minutes. Then it exits with the error : 
RangeError [ERR_INVALID_BUFFER_SIZE]: Buffer size must be a multiple of 16-bits. It then exits and stops the container. Running on Ubuntu VM (ESXI). Clean install as well as using the Home Assistant integration. 

**Device information**

Which device(s) is/are affected (make/model)?  

- Aeotec Gen 5 USB (Z-Stick)

What are the node IDs?

N/A. The container is stopped before it finishes interrogating all devices (approx 40 zwave devices)

**Last Known Working Configuration**
- [x] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [ ] Previously working device (other platform)
    - Which platform? Currently working fine with Homeassistant Zwave 1.4 integration. All node devices working correctly. 
    - Have you made any recent configuration changes to the device? 
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [x] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
N/A
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**
Add any other context about the problem here.

**Logfile:**

AlCalzone, 2021-02-11:
Can you re-upload the original logfile (the one you downloaded from the container) as a text file please? docx is not a good format for this.

shamimrahim, 2021-02-11:
Apologies. Please find attached in .txt format. 

[zwaveJSError.txt](https://github.com/zwave-js/node-zwave-js/files/5964307/zwaveJSError.txt)

shamimrahim, 2021-02-11:
Sorry, this is the log file from the Home Assistant integration. I have since uninstalled the addon. If you need the log file from the integration, i can reinstall and extract logs as required. Thanks.

AlCalzone, 2021-02-11:
I now know where it is happening, but having the `debug` level log too would be good to know for sure why.

shamimrahim, 2021-02-11:
Thank you @AlCalzone. I will reinstall and get you the logs.

shamimrahim, 2021-02-11:
Hi, the docker container does exits too quickly, and i did not have enough time to attach and dig around for the log file (could you let me know where it would be located?). I have attached the containers log file. Hope this is what you needed. 
[zwaveJS_log2.txt](https://github.com/zwave-js/node-zwave-js/files/5964471/zwaveJS_log2.txt)

AlCalzone, 2021-02-11:
Sorry, the loglevel is too low. It needs to be `debug` or I can't see which data is being received.

shamimrahim, 2021-02-11:
> Sorry, the loglevel is too low. It needs to be `debug` or I can't see which data is being received.

HI, i have enabled debug log level in Homeassistant and attach the container logs. Apologies for my ignorance if this is not what you wanted. If there is another way to obtain the logs that you need, please let me know. 

[zwaveJS_log3.txt](https://github.com/zwave-js/node-zwave-js/files/5965221/zwaveJS_log3.txt)

AlCalzone, 2021-02-14:
Sorry, that's still not the `debug` loglevel.

AlCalzone, 2021-02-14:
@shamimrahim PR #1753 should fix this, but I'm still interested in seeing the raw data that caused the crash. Please just post it here, even if the issue is closed by then.

shamimrahim, 2021-02-14:
thanks, @AlCalzone, i am struggling to find out how to enable debug logging. I have been installing it via the Home Assistant integration. 

If you could direct me how to do this, I'll definitely post the logs you require. 

Thanks again.

AlCalzone, 2021-02-14:
Not very familiar with HomeAssistant. Do you have access to the zwavejs2mqtt dashboard? If so, the link above should give you a hint.

shamimrahim, 2021-02-14:
Thanks. I haven't tried it via the zwave2mqtt addon. I'll try and spin it up in a separate docker container and give it and extract the logs

AlCalzone, 2021-02-14:
careful, `zwavejs2mqtt !== zwave2mqtt`.

shamimrahim, 2021-02-14:
ah! thanks for that! I actually had not made that distinction!

shamimrahim, 2021-02-14:
HI @AlCalzone, I managed to use the zwaveJS2mqtt control panel to set the log level before initialising the stick as you suggested. Worked a treat. 

I attach the the log file downloaded from the control panel (zwavejs_1_2_store.log) as well as a copy and paste of the docker logs via terminal (zwaveJS_log). 

Hope I have finally got the right ones! (but please let me know if you need anything else). Thanks

[zwavejs_1-2_store.log](https://github.com/zwave-js/node-zwave-js/files/5978242/zwavejs_1-2_store.log)
[zwaveJS_log4.txt](https://github.com/zwave-js/node-zwave-js/files/5978243/zwaveJS_log4.txt)

AlCalzone, 2021-02-14:
You got the right ones.

But the message in question is not written to disk yet when the container crashes 😫
I'll release a new version tonight or tomorrow. With that, the container should no longer crash and you should be able to extract the complete log.

shamimrahim, 2021-02-14:
oh ok! That's a shame. I'll send you same logs again once you've released the new version.

closed 2021-02-14 by AlCalzone

shamimrahim, 2021-02-15:
hi @AlCalzone, i noticed there was a new docker container today. Did the same test and still errored out like last time. I've attached the log file for you anyway. 
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5983695/zwavejs2mqtt.log)

AlCalzone, 2021-02-15:
Wrong log again ;)

shamimrahim, 2021-02-18:
> Wrong log again ;)

oops sorry. Here are the logs again. Both the .log files form the 'store' directory. As well as the docker container log in the .txt. Thanks

[docker_console_log.txt](https://github.com/zwave-js/node-zwave-js/files/6003998/docker_console_log.txt)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6003999/zwavejs2mqtt.log)
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6004000/zwavejs_1.log)

AlCalzone, 2021-02-18:
Okay that report claims it contains utf16, but includes an ascii string. Might be that the application used to set the string did it wrong and the device did not check the contents. My added validation now catches this, so if your container crashes, it means that the container is not up to date.

shamimrahim, 2021-02-18:
> Okay that report claims it contains utf16, but includes an ascii string. Might be that the application used to set the string did it wrong and the device did not check the contents. My added validation now catches this, so if your container crashes, it means that the container is not up to date.

ok, great, I'll try it again. I checked the docker hub page and it says that the container was updated a few seconds ago so am assuming that was your upload. I'll report back shortly.

AlCalzone, 2021-02-18:
I don't have any control over the docker images. I just develop the driver.
@robertsLando?

shamimrahim, 2021-02-18:
ah ok, yes, it's @robertsLando.  i am pulling it using the instructions on https://zwave-js.github.io/zwavejs2mqtt/#/getting-started/quick-start. 

I just tried it again and it's erroring out as before. So maybe the docker image hasn't been updated yet? 

[docker_log.txt](https://github.com/zwave-js/node-zwave-js/files/6004280/docker_log.txt)
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6004282/zwavejs_1.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6004284/zwavejs2mqtt.log)

robertsLando, 2021-02-18:
the docker image should be available now

shamimrahim, 2021-02-18:
ok, i've just tried again and get's stuck quick quickly. I've attached the docker log 
[docker2_log.txt](https://github.com/zwave-js/node-zwave-js/files/6004578/docker2_log.txt)

robertsLando, 2021-02-19:
Hemmm I have no idea why you get that error.... If that would be really like that my mailbox would be filled by bug like that. I suggest you to remove all your local images of zwavejs2mqtt and try to pull a new one to be sure you are using the correct one

shamimrahim, 2021-02-19:
> Hemmm I have no idea why you get that error.... If that would be really like that my mailbox would be filled by bug like that. I suggest you to remove all your local images of zwavejs2mqtt and try to pull a new one to be sure you are using the correct one

Yes, that occurred to me too! 
I built a new VM just to be sure and spun up the new docker. It's been running for the last 20 minutes with no issues whatsoever, found all my devices and is looking great so far! I've attached my logs for your reference in case you need them. Let me know if you need anything else. 

I really appreciate your kind help. I'm not sure when the home assistant integration will receive this update, but it looks brilliant so far. 

Thanks again, @AlCalzone!

[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6010434/zwavejs2mqtt.log)
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6010435/zwavejs_1.log)

shamimrahim, 2021-02-19:
Sorry i forgot to also thank you, @robertsLando. Many apologies and thank you very much also!

PR #1753 (2021-02-14): fix: assert that buffer length is a multiple of 2
fixes: #1711