zwave-js/zwave-js #1717: Add "boolean" config type

cgarwood, 2021-02-11:
**Is your feature request related to a problem? Please describe.**
Some devices have configuration values that are boolean on/off or enabled/disabled. These are currently defined as `type` = `number`, `allowManualEntry` = `false`, with `states: {0: "Disabled", 1: "Enabled"}` 

**Describe the solution you'd like**
Having a config type of `boolean` could help other applications more easily determine what type of UI to use for a config option. For example, instead of a dropdown to select between Enabled/Disabled, a toggle switch could be used instead.

The type of `boolean` would also more accurately describe what the value is. Granted technically boolean _is_ a number, just 0 or 1. When I see type: number I would expect it to be more than that 🙂

**Describe alternatives you've considered**
Coding the UI in the individual applications to be able to convert type: number with a min/max of 0/1 to use a toggle instead of a dropdown, but doing it at the application level doesn't benefit other ZWaveJs users.

AlCalzone, 2021-02-11:
I've had that at the back of my mind for a while - also for the values that can just do 1 and 0. The difficulty will be to determine for which params 0 means `true`, for which params 1 means `true`, and if that value is semantically compatible to the parameter meaning - that wildly differs between devices :(

cgarwood, 2021-02-11:
Yeah, for Z-Wave being a standard there's a lot of workarounds and guess-work involved :(

blhoward2, 2021-02-12:
Determining that would be easier though after the templating. To start we could just parse the files and swap in the new option entry wherever there are two options that contain enable(d) and disable(d), then use that to setup the value. It would need to support any value equaling true/false. There will be a delta that needs cleaned up but that's true now.