zwave-js/zwave-js #1721: Vivint Element Thermostat values not updating

srirams, 2021-02-11:
**Describe the bug**

I am able to set thermostat setpoints, but the values are not updated in zwavejs. Temperature and humidity values are not updated either. Initial values seem to be correct upon restart. 
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5968107/zwavejs2mqtt.log)

**Device information**

Which device(s) is/are affected (make/model)?  Radio Thermostat Company of America (RTC) 
What are the node IDs? 20

**Installation information**

How did you install `node-zwave-js`?
- [x] `zwavejs2mqtt` (latest) docker image

srirams, 2021-02-11:
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5968110/zwavejs2mqtt.log)

AlCalzone, 2021-02-12:
Wrong log :( 
The issue template gives you some pointers, but you deleted half of it.

srirams, 2021-02-12:
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5972735/zwavejs_1.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5972736/zwavejs2mqtt.log)

[d6b47744.json.txt](https://github.com/zwave-js/node-zwave-js/files/5972744/d6b47744.json.txt)
[d6b47744.values.jsonl.txt](https://github.com/zwave-js/node-zwave-js/files/5972745/d6b47744.values.jsonl.txt)

Attached correct log now. I promise, I did read the troubleshooting steps... just got confused between zwavejs2mqtt log and zwavejs log files....

I manually changed the thermostat setpoint a 9:21 and 9:25, and it seems zwavejs picked it up, as well as temperature and humidity updates, but not zwavejs2mqtt, so it may be an issue there...

AlCalzone, 2021-02-12:
Hmm, the reports are received on endpoint 0 and not mapped to endpoint 1. I'll have to check out why that is.

jjlawren, 2021-02-13:
I have a CT101 thermostat where I've noticed somewhat similar behavior.

Immediately after the thermostat node is interviewed, it has a single heat setpoint on endpoint 0. The values will update normally in both zwavejs2mqtt and Home Assistant as the thermostat operates. Commands sent to the thermostat over Z-Wave are reflected immediately.

However, if the thermostat is operated manually (and other unknown events?), a second heat setpoint is created at endpoint 1. This change to the setpoint _is_ reflected in both zwavejs2mqtt and HA, but no further changes are reflected. Commands sent via Z-Wave update the thermostat, but those values are not reflected in either zwavejs2mqtt or HA.

Once in this state, a re-interview of the node will return normal operation. After the re-interview is completed, the setpoint on endpoint 1 disappears.

Complete zwave-js debug logs with operation comments added:
```
# Manual control from 67 -> 66

16:25:52.711 SERIAL « 0x010b0004080a054303010942fd                                        (13 bytes)
16:25:52.712 CNTRLR   [Node 010] [~] [Thermostat Setpoint] setpoint[1]: 67 => 66        [Endpoint 0]
16:25:52.715 CNTRLR   [Node 010] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 0] [internal]
                       1 => 1
16:25:52.715 SERIAL » [ACK]                                                                   (0x06)
16:25:52.715 DRIVER « [Node 010] [REQ] [ApplicationCommand]
                      │ type: multicast
                      └─[ThermostatSetpointCCReport]
                          setpoint type: Heating
                          value:         66 °F
16:25:52.717 CNTRLR   [Node 010] [Thermostat Setpoint] setpoint[1]: metadata updated    [Endpoint 1]
16:25:52.718 CNTRLR   [Node 010] [+] [Thermostat Setpoint] setpoint[1]: 66              [Endpoint 1]
16:25:52.718 CNTRLR   [Node 010] [+] [Thermostat Setpoint] setpointScale[1]: [Endpoint 1] [internal]
                       1
16:25:52.724 SERIAL « 0x010b0004000a054303010942f5                                        (13 bytes)
16:25:52.725 CNTRLR   [Node 010] [~] [Thermostat Setpoint] setpoint[1]: 66 => 66        [Endpoint 0]
16:25:52.725 CNTRLR   [Node 010] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 0] [internal]
                       1 => 1
16:25:52.725 SERIAL » [ACK]                                                                   (0x06)
16:25:52.725 DRIVER « [Node 010] [REQ] [ApplicationCommand]
                      └─[ThermostatSetpointCCReport]
                          setpoint type: Heating
                          value:         66 °F
16:25:52.725 CNTRLR   [Node 010] [~] [Thermostat Setpoint] setpoint[1]: 66 => 66        [Endpoint 1]
16:25:52.726 CNTRLR   [Node 010] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 1] [internal]
                       1 => 1
                       
# Z-Wave control from 66 -> 67
                       
16:27:05.304 SERIAL » 0x010c00130a05430101094325d112                                      (14 bytes)
16:27:05.305 DRIVER » [Node 010] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      209
                      └─[ThermostatSetpointCCSet]
                          setpoint type: Heating
                          value:         67 °F
16:27:05.307 SERIAL « [ACK]                                                                   (0x06)
16:27:05.312 SERIAL « 0x0104011301e8                                                       (6 bytes)
16:27:05.313 SERIAL » [ACK]                                                                   (0x06)
16:27:05.313 DRIVER « [RES] [SendData]
                        was sent: true
16:27:05.330 SERIAL « 0x01070013d100000238                                                 (9 bytes)
16:27:05.330 SERIAL » [ACK]                                                                   (0x06)
16:27:05.331 DRIVER « [REQ] [SendData]
                        callback id:     209
                        transmit status: OK
16:27:10.335 SERIAL » 0x010a00130a0343020125d258                                          (12 bytes)
16:27:10.335 DRIVER » [Node 010] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      210
                      └─[ThermostatSetpointCCGet]
                          setpoint type: Heating
16:27:10.337 SERIAL « [ACK]                                                                   (0x06)
16:27:10.343 SERIAL « 0x0104011301e8                                                       (6 bytes)
16:27:10.343 SERIAL » [ACK]                                                                   (0x06)
16:27:10.343 DRIVER « [RES] [SendData]
                        was sent: true
16:27:10.360 SERIAL « 0x01070013d20000023b                                                 (9 bytes)
16:27:10.360 SERIAL » [ACK]                                                                   (0x06)
16:27:10.361 DRIVER « [REQ] [SendData]
                        callback id:     210
                        transmit status: OK
16:27:10.407 SERIAL « 0x010b0004000a054303010943f4                                        (13 bytes)
16:27:10.408 CNTRLR   [Node 010] [~] [Thermostat Setpoint] setpoint[1]: 67 => 67        [Endpoint 0]
16:27:10.408 CNTRLR   [Node 010] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 0] [internal]
                       1 => 1
16:27:10.408 SERIAL » [ACK]                                                                   (0x06)
16:27:10.409 DRIVER « [Node 010] [REQ] [ApplicationCommand]
                      └─[ThermostatSetpointCCReport]
                          setpoint type: Heating
                          value:         67 °F
```

AlCalzone, 2021-02-13:
Thanks, can you also post your cache file `<homeid>.json`?

jjlawren, 2021-02-13:
The cache file doesn't seem to include the new endpoints immediately after they're created. Is there a way to force a write of the current state?

jjlawren, 2021-02-13:
Here are the cache and a debug log which shows (node 10):
1. The interview process
2. Updating the setpoint over Z-Wave
3. Updating the setpoint from the thermostat
4. Additional setpoint changes over Z-Wave (none reflected in zwavejs2mqtt or HA)

[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5976029/zwavejs_1.log)
[fb4f583f.json.txt](https://github.com/zwave-js/node-zwave-js/files/5976030/fb4f583f.json.txt)

jjlawren, 2021-02-13:
I've also noticed that a second operating mode endpoint is reported after a while which causes that value to stop updating. I haven't found a reliable way to force it to appear yet.

AlCalzone, 2021-02-14:
Okay, so what I'm seeing in the log:
1. The device has two endpoints, but those only support Multilevel Sensor and Association, nothing else.
2. Setting and verifying the setpoint seems to work
3. The setpoint report is received on endpoint 0, but zwave-js wrongly maps it to endpoint 1
4. At that point, zwave-js finds the same value ID on endpoint 1 and hides the changes on endpoint 0.

PR #1749 (2021-02-14): fix: check endpoint support before mapping unsolicited reports
fixes: #1721
closed 2021-02-14 by AlCalzone