zwave-js/zwave-js #1722: Zwave Combo - Two version, one works fine other only shows battery level

kan84, 2021-02-12:
**Describe the bug**

I have two version of First Alert Zwave combo Smoke and CO alarm system. 
-Version 2 works perfectly fine and shows multiple entities for smoke and CO along with other entities.
-Version 1 only shows two entites and both are battery related

V2 Details  
-https://www.amazon.com/gp/product/B00KMHXFAI
-Node 13
-Json File Attached 

V1 Details
-https://www.amazon.com/gp/product/B08FFB233Y
-Node 15
-Json Attached

**Device information**

Which device(s) is/are affected (make/model)?  
What are the node IDs?
Node 15 - V1 have issue with not showing CO and Smoke alarm. Only shows battery entites
Node 13 - V2 shows both along with other entities

**Last Known Working Configuration**
- Was using with smartthings and both worked. Excluded and added to zwavejs2mqtt and Homeassistant
- [ ] Previously working device (node-zwave-js)
    -V1 never showed alarm just battery
    -V2 shows all alarms
    -ZwaveJS Version 6.1.3 
    -HA Version :2021.2.2
- [ ] Previously working device (other platform)
    - Smartthings
    - Excluded both from smartthings and added to zwavejs
    
**Installation information**
How did you install `node-zwave-js`?
- [ X] `zwavejs2mqtt` (latest) docker image

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Homeassistant
2. Entites you can see both Smoke/CO sensors
3. But for V1 only batterty entties shows whereas V2 all entities shows including CO/Smoke alarm

**Logfile:**
[zwavejs.log](https://github.com/zwave-js/node-zwave-js/files/5969234/zwavejs.log)
[FirstAlert_v1.txt](https://github.com/zwave-js/node-zwave-js/files/5969189/FirstAlert_v1.txt)
[FirstAlert_v2.txt](https://github.com/zwave-js/node-zwave-js/files/5969191/FirstAlert_v2.txt)

blhoward2, 2021-02-12:
It's seeing the notification command class. When you say you are only seeing battery entities, where is that? In HA? Are you using the zwave-js server or mqtt discovery?

kan84, 2021-02-12:
1. In Home Assistant
2. ZwaveJS2mqtt
3. Ok will try enabling MQTT and see waht happens. WIll update once I have something.
4. This is what I see in debug log when I manually triggered the Alarm
Debug
2021-02-11 23:04:29.818 INFO ZWAVE: Node 15: value updated: 113-0-alarmType 13 => 12
2021-02-11 23:04:29.821 INFO ZWAVE: Node 15: value updated: 113-0-alarmLevel 255 => 255

blhoward2, 2021-02-12:
It looks like we're setting that device up correctly, but the zwave-js server isn't creating an entity for it. They'd have to fix that.

kan84, 2021-02-12:
There were few entities disabled by default at start enabled them and this is what I see.
[image]

Compared to V2, which also have quite a few disabled entities.

blhoward2, 2021-02-12:
That looks correct. The 255 is the alarm value. The 12/13 is whether it's a real alarm or test if memory serves.

kan84, 2021-02-12:
Thanks for the information, I believe FirstAlert did not implement the binary sensor at hardware level or something which probably was fixed in V2. For those facing same issue, they can create a binary sensor or automation based on the following two entities which are by default disabled when you add the device for some reason but you can enable it from settings in Home Assistant

sensor.living_room_fire_alarmtype  - Alarm Type 
sensor.living_room_fire_alarmlevel - Alarm Level

Alarm Type        Alarm Level               Alarm
13                             255                Default State
12                             255                Smoke Test
1                               255                Smoke Alarm
2                               255                CO Alarm

closed 2021-02-12 by kan84