zwave-js/zwave-js #1738: Aeotec ZW132 dual nano switch - Missing consumption report

razvanws, 2021-02-12:
**Describe the bug**

Aeotec ZW132 switch doesn't send Meter report.

**Device information**

Which device(s) is/are affected (make/model)?  
What are the node IDs?

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [x] Previously working device (other platform)
    - Which platform?
       Openzwave / zwave2mqtt
    - Have you made any recent configuration changes to the device? Describe.
       No

**Installation information**
How did you install `node-zwave-js`?
- [x] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Turn on/off ZW132 binary switches and check the log file. The report meter is missing.
The report is also not sent at specified interval. Check the attached screenshot with the parameter settings.

**Additional context**
If I trigger "Refresh values", the values are updated.

**Logfile:**
19:08:18.456 SERIAL » 0x010e00130d07600d00022501ff2589f0                                  (16 bytes)
19:08:18.458 DRIVER » [Node 013] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      137
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      0
                        │ destination: 2
                        └─[BinarySwitchCCSet]
                            target value: true
19:08:18.461 SERIAL « [ACK]                                                                   (0x06)
19:08:18.467 SERIAL « 0x0104011301e8                                                       (6 bytes)
19:08:18.468 SERIAL » [ACK]                                                                   (0x06)
19:08:18.469 DRIVER « [RES] [SendData]
                        was sent: true
19:08:18.483 SERIAL « 0x010700138900000260                                                 (9 bytes)
[image]
[image]
19:08:18.484 SERIAL » [ACK]                                                                   (0x06)
19:08:18.485 DRIVER « [REQ] [SendData]
                        callback id:     137
                        transmit status: OK
19:08:18.489 CNTRLR   [Node 013] [~] [Binary Switch] currentValue: false => true        [Endpoint 2]
19:08:19.500 SERIAL » 0x010d00130d06600d00022502258a0d                                    (15 bytes)
19:08:19.501 DRIVER » [Node 013] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      138
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      0
                        │ destination: 2
                        └─[BinarySwitchCCGet]
19:08:19.508 SERIAL « [ACK]                                                                   (0x06)
19:08:19.514 SERIAL « 0x0104011301e8                                                       (6 bytes)
19:08:19.515 SERIAL » [ACK]                                                                   (0x06)
19:08:19.516 DRIVER « [RES] [SendData]
                        was sent: true
19:08:19.531 SERIAL « 0x010700138a00000263                                                 (9 bytes)
19:08:19.532 SERIAL » [ACK]                                                                   (0x06)
19:08:19.533 DRIVER « [REQ] [SendData]
                        callback id:     138
                        transmit status: OK
19:08:19.544 SERIAL « 0x010d0004000d07600d02002503ff4a                                    (15 bytes)
19:08:19.546 CNTRLR   [Node 013] [~] [Binary Switch] currentValue: true => true         [Endpoint 2]
19:08:19.549 SERIAL » [ACK]                                                                   (0x06)
19:08:19.551 DRIVER « [Node 013] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      2
                        │ destination: 0
                        └─[BinarySwitchCCReport]
                            current value: true
19:08:20.247 SERIAL « 0x010d0004000d07600d02002003ff4f                                    (15 bytes)
19:08:20.248 SERIAL » [ACK]                                                                   (0x06)
19:08:20.251 DRIVER « [Node 013] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      2
                        │ destination: 0
                        └─[BasicCCReport]
                            current value: 99
19:08:20.254 CNTRLR   [Node 013] [~] [Binary Switch] currentValue: true => true         [Endpoint 2]

AlCalzone, 2021-02-12:
I don't see any screenshots 🤷🏻‍♂️

razvanws, 2021-02-12:
Please see below the association json file. I also updated the screenshots in the first post.

[e7851489.txt](https://github.com/zwave-js/node-zwave-js/files/5974464/e7851489.txt)

AlCalzone, 2021-02-13:
It looks like the associations are set up correctly - the device sends an unsolicited Binary Switch Report:
```
19:08:19.551 DRIVER « [Node 013] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      2
                        │ destination: 0
                        └─[BinarySwitchCCReport]
                            current value: true
```
Since that is the case, I believe this is ultimately a configuration issue. From your screenshot it looks like the device is configured to send a meter report every 600 seconds.

BTW @blhoward2 these config parameters are named terribly. Is that file on your list?

blhoward2, 2021-02-13:
Yes

razvanws, 2021-02-19:
After debugging it more, there are 3 problems actually:

1. The associations are not setup correctly (I will explain below why).
2. Even if I set the associations correctly, after restarting zwavejs, they are again incorrect
3. Even if the associations are set correctly and I can see the reports in the log file, the reports are not published to mqtt.

**Problem 1 and 2 details**
In case of those ZW132 devices, it seems that when you setup the lifeline group, you are forced to enter a Channel ID. If you don't enter anything as Channel ID, it defaults to 0. The problem is that when channel ID is 0, the power meter reports are not sent.
In order for the power meter reports to be sent, I must enter Channel ID -1. When I enter Channel ID -1, the association look like in the attached image. However, if zwavejs is restarted, Channel ID is set to 0 again and the reports are not sent anymore.

**Problem 3 details**
After I'm setting the Channel ID for group association to -1, I can see the reports in zwavejs logs (see below). However, those reports are not published to mqtt. The reports below are not pushed to mqtt:

```
11:55:10.746 SERIAL « 0x01100004000d0a3202217400001c8d000018                              (18 bytes)
11:55:10.749 CNTRLR   [Node 013] [~] [Meter] value[66049]: 0 => 7.309                   [Endpoint 0]
11:55:10.750 CNTRLR   [Node 013] [~] [Meter] deltaTime[66049]: 0 => 0                   [Endpoint 0]
11:55:10.751 SERIAL » [ACK]                                                                   (0x06)
11:55:10.753 DRIVER « [Node 013] [REQ] [ApplicationCommand]
                      └─[MeterCCReport]
                          type:       Electric
                          scale:      W
                          rate type:  Consumed
                          value:      7.309
                          time delta: 0 seconds

11:55:20.746 SERIAL « 0x01100004000d0a3202217400001c8f00001a                              (18 bytes)
11:55:20.748 CNTRLR   [Node 013] [~] [Meter] value[66049]: 7.309 => 7.311               [Endpoint 0]
11:55:20.750 CNTRLR   [Node 013] [~] [Meter] deltaTime[66049]: 0 => 0                   [Endpoint 0]
11:55:20.750 SERIAL » [ACK]                                                                   (0x06)
11:55:20.753 DRIVER « [Node 013] [REQ] [ApplicationCommand]
                      └─[MeterCCReport]
                          type:       Electric
                          scale:      W
                          rate type:  Consumed
                          value:      7.311
                          time delta: 0 seconds
```

[image]

[image]

AlCalzone, 2021-02-19:
>  Channel ID -1

@robertsLando @marcelveldt do any of you know what is meant by this? Channel ID is not a `zwave-js` concept.

razvanws, 2021-02-19:
In order to work around for the Channel ID -1 not sticking after the restart of zwavejs, you can set the lifeline association twice, once with 0, once with -1. 
I don't know what are the implications of doing it but it seems that it will keep the Channel ID -1 after restart and the power consumption reports are sent after restart as well.

The reports are still not published to mqtt though.

[image]

robertsLando, 2021-02-19:
Sorry that's just a wrong label, that's the endpoint

AlCalzone, 2021-02-19:
But what is endpoint -1?

robertsLando, 2021-02-19:
> But what is endpoint -1?

User error I think

razvanws, 2021-02-19:
By the way, I noticed that some devices, such as Aeotec ZW111 doesn't have the option to specify an Endpoint. I believe that ZW132 should be the same, most probably that endpoint shouldn't exist for this device and that's why somehow it works with -1?

AlCalzone, 2021-02-19:
Ohhh - so ZW111's association shouldn't be an endpoint association, but a node association (no target endpoint)?
Yeah, we have a flag for that.

razvanws, 2021-02-19:
If I leave the endpoint field empty, it defaults to 0 and then the reports are not received. That's why I came up with -1, just to see what happens.

Anyway, if you're saying that ZW111 has a flag for not showing the endpoint, the same flag should be used for ZW132 as well.

AlCalzone, 2021-02-19:
> If I leave the endpoint field empty, it defaults to 0 and then the reports are not received.

That's a bug in zwavejs2mqtt @robertsLando !

AlCalzone, 2021-02-19:
ZW111 config is fixed by https://github.com/zwave-js/node-zwave-js/pull/1821
Are you 100% sure about ZW132? That is a dual channel switch, so it must have multiple endpoints.

razvanws, 2021-02-19:
I'm sure. I tried with other values as well (2, 3 etc.) and they are not accepted. The only values that are accepted are 0 (which doesn't work because the reports are not received) and -1 (which works perfectly, all the reports are received). The only other problem is that the reports are not pushed to mqtt (for ZW132).

AlCalzone, 2021-02-19:
Wait, I think I misread your comment... So lets get this straight:
ZW111 uses a node association by default and works. This is probably due to the fact that it only has a single endpoint.
ZW132 uses an endpoint association which does not work. A node association works.

Correct? If so, the above fix should be applied to the ZW132 config, not the 111 one.
Once we've fixed that, we should investigate this - but it might be that the changes will fix it:
> The only other problem is that the reports are not pushed to mqtt (for ZW132).

---

> I tried with other values as well (2, 3 etc.) and they are not accepted. 

Yep, because the controller does not have any endpoints - you're setting the target, not the source.

robertsLando, 2021-02-19:
https://github.com/zwave-js/zwavejs2mqtt/pull/650

AlCalzone, 2021-02-19:
> type="number"

Can you leave a number field empty? Or does it have a default value then?

razvanws, 2021-02-19:
If I leave it empty, after I press the "Add" button, I see the association with "Endpoint: 0". 
As if the backend will automatically set it as 0 (you can see it in one of the screnshots above).

razvanws, 2021-02-19:
> Correct? If so, the above fix should be applied to the ZW132 config, not the 111 one.

Yes, it should be applied to ZW132, not ZW111. ZW111 doesn't have any problem, it works perfectly as it is.

robertsLando, 2021-02-19:
check the logs, there you will see the payload sent to the backend, you will see that the endpoint is not there

razvanws, 2021-02-19:
You're right, just checked it, the UI doesn't send any payload to the backend (see the screenshot below).

[image]

However, you can see below that the backend replies with endpoint: 0, see below:

[image]

AlCalzone, 2021-02-19:
It returns both. One with and one without endpoint.

razvanws, 2021-02-19:
I did one more test. I deleted all the associations and added only the lifeline again without touching the endpoint box. All went well until I restarted zwavejs2mqtt container. After restart, it comes back with endpoint: 0.

AlCalzone, 2021-02-19:
Yup, zwave-js tries to fix the lifeline association when it interviews the device. Without the config change, it thinks that it should be an endpoint association, so it removes the (what it thinks) incorrect node association.

razvanws, 2021-02-19:
Ok then. Thanks a lot for the effort. I will test it again and follow up after the PR above is integrated and released.

razvanws, 2021-02-27:
I just tested the fix and it's working good as far as lifeline association is concerned. 

However, the metering reports are not published to mqtt. See below an example of meter report not being published. Do you want me to open a new bug for it?

```
11:12:32.763 SERIAL « 0x01100004000d0a3202217400000000000089                              (18 bytes)
11:12:32.765 CNTRLR   [Node 013] [~] [Meter] value[66049]: 0 => 0                       [Endpoint 0]
11:12:32.765 CNTRLR   [Node 013] [~] [Meter] deltaTime[66049]: 0 => 0                   [Endpoint 0]
11:12:32.766 SERIAL » [ACK]                                                                   (0x06)
11:12:32.767 DRIVER « [Node 013] [REQ] [ApplicationCommand]
                      └─[MeterCCReport]
                          type:       Electric
                          scale:      W
                          rate type:  Consumed
                          value:      0
                          time delta: 0 seconds
```

AlCalzone, 2021-02-27:
@robertsLando

robertsLando, 2021-03-01:
@razvanws please open a bug on zwavejs2mqtt and also attach logs of both

bburan, 2021-03-02:
I don't know if this is related, but I also have a ZW132 switch and I have it configured to provide a meter report whenever the wattage changes (parameter 90 = Enable, parameter 91 = 5W, parameter 92=5%). Zwave JS log properly provides the report:
```
05:19:59.018 SERIAL « 0x0110000400050a3202217400007d79000085                              (18 bytes)                                                
05:19:59.018 CNTRLR   [Node 005] [~] [Meter] value[66049]: 33.814 => 32.121             [Endpoint 0]                                                 
05:19:59.019 CNTRLR   [Node 005] [~] [Meter] deltaTime[66049]: 0 => 0                   [Endpoint 0]                                                 
05:19:59.019 SERIAL » [ACK]                                                                   (0x06)                                                
05:19:59.019 DRIVER « [Node 005] [REQ] [ApplicationCommand]                                                                                         
                      └─[MeterCCReport]                                                                                                              
                          type:       Electric                                                                                                       
                          scale:      W                                                                                                              
                          rate type:  Consumed                                                                                                       
                          value:      32.121                                                                                                         
                          time delta: 0 seconds                                                                                                      
```
However, neither Home Assistant or the ZwaveJS control panel seem to pick up this value change. Screenshot from Zwave JS control panel:
[image]
Screenshot from Home Assistant showing the entity values:
[image]

AlCalzone, 2021-03-02:
@bburan please test https://github.com/zwave-js/node-zwave-js/pull/1966

bburan, 2021-03-05:
@AlCalzone I can confirm this fixed my issue as well. Thank you!

killtux, 2023-02-08:
i dont know if its related to this issue but i am having problems getting the ZW132 Dual Switch to send reports for device 1 and 2. It only reports for the "main" device (consumption for both together) but not for the invidual ones.

if i hit the refresh button at the meter dialog in z-wave js, then it updates the values for the invidual devices to.

[image]

[image]

any idea what could be the issue with this? is there any way to "pull" the values/refresh itself?

AlCalzone, 2023-03-06:
@killtux 
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of you re-interviewing the device, loglevel `debug` and attach it here as a file (drag & drop into the text field).

PR #1822 (2021-02-19): fix(config): use Node Associations for ZW132 Lifeline
fixes: #1738
closed 2023-03-17 by zwave-js-assistant[bot]