zwave-js/zwave-js #1742: Adding Zooz Zen30 causes docker image to crash

bryantlee, 2021-02-12:
**Describe the bug**

When adding a Zooz Zen30 switch, zwavejs crashes and docker container stops.  When starting the container again I can see a new node but it has no device information.  When interacting with the node, I can see scene events in logs, but device has no entities in HASS.

**Device information**

Which device(s) is/are affected (make/model)?  
Zooz Zen30
What are the node IDs?
23

**Last Known Working Configuration**
- [X] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [X] Other: HASS Docker image

**To Reproduce**
Steps to reproduce the behavior:
Exclude device, Include device
**Additional context**

**Logfile:**
`17:25:40.599 CNTRLR   starting inclusion process...
17:25:40.616 CNTRLR   handling add node request (status = Ready)
17:25:40.617 CNTRLR     the controller is now ready to add nodes
17:25:42.899 CNTRLR   handling add node request (status = NodeFound)
17:25:43.067 CNTRLR   handling add node request (status = AddingSlave)
17:25:47.314 CNTRLR   handling add node request (status = ProtocolDone)
17:25:47.315 CNTRLR   stopping inclusion process...
17:25:47.375 CNTRLR   handling add node request (status = Done)
17:25:47.376 CNTRLR   done called for 23
17:25:47.377 CNTRLR   finished adding node 23:
                        basic device class:    Routing Slave
                        generic device class:  Multilevel Switch
                        specific device class: Multilevel Power Switch
                        supported CCs: 
                        · Basic (0x20)
                        · Multilevel Switch (0x26)
                        · All Switch (0x27)
                        · Z-Wave Plus Info (0x5e)
                        · Binary Switch (0x25)
                        · Association (0x85)
                        · Multi Channel Association (0x8e)
                        · Association Group Information (0x59)
                        · Transport Service (0x55)
                        · Version (0x86)
                        · Manufacturer Specific (0x72)
                        · Device Reset Locally (0x5a)
                        · Powerlevel (0x73)
                        · Configuration (0x70)
                        · Central Scene (0x5b)
                        · Multi Channel (0x60)
                        · Security 2 (0x9f)
                        · Supervision (0x6c)
                        · Firmware Update Meta Data (0x7a)
                        controlled CCs: 
17:25:47.378 CNTRLR » [Node 023] Assigning return route to controller...
17:25:47.389 CNTRLR   the inclusion process was stopped
17:25:47.520 CNTRLR   [Node 023] Configuring Z-Wave+ Lifeline association...
17:25:51.670 CNTRLR   [Node 023] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
17:25:56.303 CNTRLR   [Node 023] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
17:26:00.903 CNTRLR   [Node 023] Node 23 did not respond after 3 attempts, it is presumed dead
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
ZWaveError: Failed to send the command after 3 attempts (Status NoAck)
    at Object.sendDataErrorToZWaveError (/usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/src/lib/driver/StateMachineShared.ts:112:12)
    at /usr/lib/node_modules/@zwave-js/server/node_modules/zwave-js/src/lib/driver/SendThreadMachine.ts:437:4
    at /usr/lib/node_modules/@zwave-js/server/node_modules/xstate/lib/utils.js:410:33
    at Array.reduce (<anonymous>)
    at Object.updateContext (/usr/lib/node_modules/@zwave-js/server/node_modules/xstate/lib/utils.js:400:25)
    at Object.resolveActions (/usr/lib/node_modules/@zwave-js/server/node_modules/xstate/lib/actions.js:409:19)
    at StateNode.resolveTransition (/usr/lib/node_modules/@zwave-js/server/node_modules/xstate/lib/StateNode.js:787:35)
    at StateNode.transition (/usr/lib/node_modules/@zwave-js/server/node_modules/xstate/lib/StateNode.js:733:21)
    at StateNode.resolveRaisedTransition (/usr/lib/node_modules/@zwave-js/server/node_modules/xstate/lib/StateNode.js:738:22)
    at StateNode.resolveTransition (/usr/lib/node_modules/@zwave-js/server/node_modules/xstate/lib/StateNode.js:873:39)
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.`

AlCalzone, 2021-02-13:
Thanks for the report. Is it reproducible when you exclude and add the device again?

bryantlee, 2021-02-13:
> Thanks for the report. Is it reproducible when you exclude and add the device again?

Yes it is. Same behavior. manually starting the container, exclude, then include will result in same behavior.

AlCalzone, 2021-02-13:
Okay just to make sure we don't miss anything, can you set the log level to debug and post the updated log snippet?

closed 2021-02-14 by AlCalzone

bryantlee, 2021-02-17:
Sorry I didn't think logs uploaded before this was closed.  I have updated to the latest version and the container no longer crashes, but still fails to add the device (I will pull logs and open a new issue)

PR #1757 (2021-02-14): fix(driver): guard Z-Wave+ bootstrapping against transmission errors
fixes: #1742