zwave-js/zwave-js #1744: Inovelli LZW36 (Red Series Fan + Light) is missing meter settings for electric consumption

carldanley, 2021-02-13:
**Describe the bug**

Inovelli LZW36 is missing meter settings. It is supposed to have an `Electric Consumed [W]` and `Electric Consumed [kWh]` just like the Inovelli LZW31-SN  and the Inovelli LZW30-SN.

**Device information**

Which device(s) is/are affected (make/model)? `Inovelli LZW36`
What are the node IDs? `Device ID: 798-1-14 (0x031e-0x0001-0x000e)`

**Last Known Working Configuration**
- [x] Previously working device (other platform)
    - Which platform? `openzwave + homeassistant`
    - Have you made any recent configuration changes to the device? Describe. `I have not.`
    
**Installation information**
How did you install `node-zwave-js`?
- [x] `zwavejs2mqtt` (latest) docker image Specifically, [SHA digest: c482b2b83757870a4f937048db2f663cc8734b76eabc1860021338a3b834233d](https://hub.docker.com/layers/zwavejs/zwavejs2mqtt/latest/images/sha256-c482b2b83757870a4f937048db2f663cc8734b76eabc1860021338a3b834233d?context=explore). I believe this is the `1.1.1` release (as the digests match).

**To Reproduce**
Brand new install of homeassistant; completely empty. Added zwavejs2mqtt (using websockets). I just let zwave do it's thing ;)

AlCalzone, 2021-02-13:
Please make a zwave-js log, loglevel `debug` and attach it here as a file.
Please also post the cache files `<homeid>.json` and `.values.jsonl`.

AlCalzone, 2021-02-15:
Also, please update to `6.3.0` and re-interview your device - this might be related to https://github.com/zwave-js/node-zwave-js/pull/1749

BluGeni, 2021-02-15:
I dont know if it is related but if I add an inovelli switch from zwavejs2mqtt I dont get kW but I do get kWh. If I add the switch from HA through the zwavejs integration, I do get kW and kWh.

AlCalzone, 2021-02-15:
Might be a coincidence. If the node does not respond to the one query for some reason it could be that the value doesn't exist yet when zjs2m or HA create the entities.
I don't know how dynamically they react to new values that come in later.

closed 2021-02-26 by zwave-js-assistant[bot]