zwave-js/zwave-js #1752: Nortek Security & Control LLC Garage Door Controller GD00Z-7 MISSING CONTROL BUTTONS

trentjw, 2021-02-14:
MISSING BUTTONS TO CONTROL GARAGE DOOR DEVICE FILE IS NOT BEING BUILT PROPER
Nortek Security & Control LLC Garage Door Controller GD00Z-7
it shows up as a sensor only, does not show up as a controller to operate the door network key is working as my kwikset locks are working as intended
**What are the IDs?**

Manufacturer ID: 0x14f
Product type: 0x4744
Product ID: 0x3531

XML file
`
<Node id="51" name="" location="" basic="4" generic="64" specific="7" roletype="5" devicetype="7680" nodetype="0" type="Secure Barrier AddOn" listening="true" frequentListening="false" beaming="true" routing="true" max_baud_rate="40000" version="4" secured="true" query_stage="Complete"> <Manufacturer id="14f" name="Linear (Nortek Security Control LLC)"> <Product type="4744" id="3531" name="GD00Z-7 Garage Door Opener Remote Controller" /> </Manufacturer> <CommandClasses> <CommandClass id="32" name="COMMAND_CLASS_BASIC" version="1" request_flags="4"> <Instance index="1" /> <Value type="byte" genre="basic" instance="1" index="0" label="Basic" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" /> </CommandClass> <CommandClass id="34" name="COMMAND_CLASS_APPLICATION_STATUS" version="1" request_flags="4" issecured="true"> <Instance index="1" /> </CommandClass> <CommandClass id="90" name="COMMAND_CLASS_DEVICE_RESET_LOCALLY" version="1" issecured="true"> <Instance index="1" /> </CommandClass> <CommandClass id="94" name="COMMAND_CLASS_ZWAVEPLUS_INFO" version="1" request_flags="4" innif="true"> <Instance index="1" /> <Value type="byte" genre="system" instance="1" index="0" label="ZWave+ Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="1" /> <Value type="short" genre="system" instance="1" index="1" label="InstallerIcon" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-32768" max="32767" value="7680" /> <Value type="short" genre="system" instance="1" index="2" label="UserIcon" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-32768" max="32767" value="7680" /> </CommandClass> <CommandClass id="102" name="COMMAND_CLASS_BARRIER_OPERATOR" version="1" issecured="true"> <Instance index="1" /> <Value type="list" genre="user" instance="1" index="1" label="Barrier State Label" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="6"> <Item label="Closed" value="0" /> <Item label="Closing" value="1" /> <Item label="Stopped" value="2" /> <Item label="Opening" value="3" /> <Item label="Opened" value="4" /> <Item label="Unknown" value="5" /> </Value> <Value type="list" genre="config" instance="1" index="2" label="Supported Signals" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="3" size="4"> <Item label="None" value="0" /> <Item label="Audible" value="1" /> <Item label="Visual" value="2" /> <Item label="Both" value="3" /> </Value> <Value type="bool" genre="config" instance="1" index="3" label="Audible Notification" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" /> <Value type="bool" genre="config" instance="1" index="4" label="Visual Notification" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" /> </CommandClass> <CommandClass id="113" name="COMMAND_CLASS_ALARM" version="8" request_flags="2" issecured="true"> <Instance index="1" /> <Value type="byte" genre="user" instance="1" index="0" label="Alarm Type" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" /> <Value type="byte" genre="user" instance="1" index="1" label="Alarm Level" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" /> <Value type="byte" genre="user" instance="1" index="2" label="SourceNodeId" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" /> <Value type="byte" genre="user" instance="1" index="9" label="Access Control" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="254" /> <Value type="byte" genre="user" instance="1" index="10" label="Burglar" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="254" /> </CommandClass> <CommandClass id="114" name="COMMAND_CLASS_MANUFACTURER_SPECIFIC" version="1" request_flags="4" innif="true"> <Instance index="1" /> </CommandClass> <CommandClass id="115" name="COMMAND_CLASS_POWERLEVEL" version="1" request_flags="4" issecured="true"> <Instance index="1" /> <Value type="list" genre="system" instance="1" index="0" label="Powerlevel" units="dB" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="1"> <Item label="Normal" value="0" /> <Item label="-1dB" value="1" /> <Item label="-2dB" value="2" /> <Item label="-3dB" value="3" /> <Item label="-4dB" value="4" /> <Item label="-5dB" value="5" /> <Item label="-6dB" value="6" /> <Item label="-7dB" value="7" /> <Item label="-8dB" value="8" /> <Item label="-9dB" value="9" /> </Value> <Value type="byte" genre="system" instance="1" index="1" label="Timeout" units="seconds" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" /> <Value type="button" genre="system" instance="1" index="2" label="Set Powerlevel" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" /> <Value type="byte" genre="system" instance="1" index="3" label="Test Node" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" /> <Value type="list" genre="system" instance="1" index="4" label="Test Powerlevel" units="dB" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="1"> <Item label="Normal" value="0" /> <Item label="-1dB" value="1" /> <Item label="-2dB" value="2" /> <Item label="-3dB" value="3" /> <Item label="-4dB" value="4" /> <Item label="-5dB" value="5" /> <Item label="-6dB" value="6" /> <Item label="-7dB" value="7" /> <Item label="-8dB" value="8" /> <Item label="-9dB" value="9" /> </Value> <Value type="short" genre="system" instance="1" index="5" label="Frame Count" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="-32768" max="32767" value="0" /> <Value type="button" genre="system" instance="1" index="6" label="Test" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" /> <Value type="button" genre="system" instance="1" index="7" label="Report" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" /> <Value type="list" genre="system" instance="1" index="8" label="Test Status" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="1"> <Item label="Failed" value="0" /> <Item label="Success" value="1" /> <Item label="In Progress" value="2" /> </Value> <Value type="short" genre="system" instance="1" index="9" label="Acked Frames" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-32768" max="32767" value="0" /> </CommandClass> <CommandClass id="133" name="COMMAND_CLASS_ASSOCIATION" version="1" request_flags="4" issecured="true"> <Instance index="1" /> <Associations num_groups="1"> <Group index="1" max_associations="1" label="Group 1" auto="true"> <Node id="1" /> </Group> </Associations> </CommandClass> <CommandClass id="134" name="COMMAND_CLASS_VERSION" version="1" request_flags="4" innif="true"> <Instance index="1" /> <Value type="string" genre="system" instance="1" index="0" label="Library Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="3" /> <Value type="string" genre="system" instance="1" index="1" label="Protocol Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="4.60" /> <Value type="string" genre="system" instance= [... cut ...] y="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="3.00" /> </CommandClass> <CommandClass id="152" name="COMMAND_CLASS_SECURITY" version="1" request_flags="4" innif="true"> <Instance index="1" /> <Value type="bool" genre="system" instance="1" index="0" label="Secured" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" /> </CommandClass> </CommandClasses> </Node>
`
If so, it helps in creating the config file.

AlCalzone, 2021-02-14:
Dude! Just use the bug report template, fill it out don't delete it!
Next time I'm gonna block you.

closed 2021-02-14 by AlCalzone

trentjw, 2021-02-14:
Fine won't submit it anymore if you want to fix the faulty config file for zwave js for this device you can pull the xml file from one of the closed ones my zwave 1.4 is working fine so I don't need to use this anyways