zwave-js/zwave-js #1761: Inovelli LZW31-SN firmware update does not complete due to 'waiting for activation' status

prwood80, 2021-02-14:
**Describe the bug**

Updating firmware on Inovelli LZW31-SN results in result 0xFD (OK_WaitingForActivation).  The activation never occurs, the node is refreshed and the update does not appear to succeed.

**Device information**

Inovelli LZW31-SN, current firmware version is 1.47.  Updating to v1.52

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [x] Previously working device (other platform)
    - Which platform? Z-Wave PC Controller 5
    - Have you made any recent configuration changes to the device? No
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [x] Manual Docker build
    - `node-zwave-js` branch: master
    - `zwavejs2mqtt` branch: master
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Steps to reproduce the behavior:
1. Initiate 'Begin Firmware Update' for node on the internal Control Panel web page.
2. Select Target 0 & otz file
3. Wait for update to complete
4. Observe Controller Status reports completion with status code 0xFD
5. Refresh node
6. Observe node firmware is not updated

**Additional context**

**Logfile:**
Could not successfully start a firmware update with the logging enabled.  Will continue to try.  Even after restarting the docker container and waiting for all nodes to complete the interview stage, I received a number of errors (firmware update in progress, node does not support firmware update [though it seemed to be identified and healthy])

AlCalzone, 2021-02-14:
AFAIK that means you have to manually push a button on the device to continue.
It is basically a safety measure so you can't just update the firmware without physical access to the device.

prwood80, 2021-02-14:
I'll try that.  I didn't see any documentation from Inovelli in those regards (and the Z-wave PC Controller 5 doesn't require that).

AlCalzone, 2021-02-15:
> the Z-wave PC Controller 5 doesn't require that

I might be mistaken then.

AlCalzone, 2021-02-15:
Nevermind, I found it:

> 4.17.5 Firmware Update Activation Set Command
This command is used to initiate the programming of a previously transferred firmware image.
Refer to the Firmware Update Meta Data Status Report Command Status code 0xFD.
This command MAY be issued directly by a controlling node or MAY be scheduled for later execution via
the Schedule Command Class.

We're not doing that currently.

BluGeni, 2021-02-15:
I was able to update the firmware without the activation but im on the latest docker build so I cant do the .bin update yet. Waiting for the official image release of that and I can test further. I am upgrading from the base firmware - I think 1.32 or something close to that. I cant say the switch really works 100% after doing just the ota target 0 update but I assume its because I need to do the other firmware update still.

BluGeni, 2021-02-15:
I am now on the latest zwavejs (6.3.0) and can upgrade .bin but I get 
`16:24:33.258 CNTRLR « [Node 003] Firmware update failed with status Error_InsufficientMemory
2021-02-15 16:24:33.260 INFO ZWAVE: Controller status: Node 3 firmware update FINISHED: Status 7, Time: undefined`

I am targeting 1 with the .bin file.

AlCalzone, 2021-02-15:
Was firmware 0 activated? Might be that that one is still in memory waiting for activation.
Otherwise I can only guess why that is happening.

prwood80, 2021-02-15:
I had the same issue with target 1, but my assumption was that memory was full due to the non-activated target 0 update. I believe @AlCalzone assessment is correct in that we'll need to implement the firmware activation set command. 

On the surface, it doesn't seem like that should be hard, so I may play around with it.

AlCalzone, 2021-02-15:
The command is implemented, it just doesn't get used 😅

prwood80, 2021-02-15:
Understood. I meant I was going to play with the handling of the FD status by sending the command, process the response, then refreshing the node. Maybe it's more complicated?

AlCalzone, 2021-02-15:
I think it is actually pretty simple. I've just never encountered a device that requires it so it must have slipped through the cracks.

prwood80, 2021-02-16:
After digging into this a little further, I think the problem is that 'activation' option in FirmwareUpdateMetaDataCCRequestGetOptions does not appear to get set.  If activation is undefined, FirmwareUpdateMetaDataCCRequestGet serialize will not add the payload[9] word with the appropriate bit set.

I don't really have experience with TypeScript, but I couldn't find any place in the code where the value is set, and it does not appear to have a default.  From the comment in Node.ts, it appears the expectation is that this bit will be set to 0 (ie. activate immediately).  As a test, I set 'activation: false' in the call to requestUpdate (Node.ts).  With this change, the firmware status returns 0xFF (and update is successful).  Is this an acceptable change?  On the surface, the change seems reasonable because there are other checks in requestUpdate to ensure command 4 (or higher) is supported.

However, target 1 does not update.  It still reports error 7 (which may be a separate issue).  I tried removing power from the switch, re-interviewing it, and updating target 1 again but I only get status 7, so more digging will be required.

AlCalzone, 2021-02-16:
I think two things need to be done:
a) The firmware update method should set `activation: false` by default
b) We should give API users a way to set it to true **and** add an API call for the activation.

Regarding target 1: That's a `.bin` file, correct? I'm not sure if there is any conversion that needs to be done here, like with the `.hex` files. Maybe that file contains more info than just the firmware.

DelusionalAI, 2021-02-16:
I was able to update Target 1 on mine no problem, that .bin file should just be firmware. Its the same file you would pass when using the Zwave PC controller 5.

PR #1788 (2021-02-17): feat(cc): disallow delay of firmware activation
With this PR, we explicitly set the activation feature to false during firmware updates.

fixes: #1761
closed 2021-02-17 by AlCalzone