zwave-js/zwave-js #1764: bug: heal seems to fail with silicon labs 700 controller

scyto, 2021-02-15:
**Describe the bug**

When i perform a heal it doesn't appear to work based on the logs..

**Device information**

All devices and all node IDs.
What are the node IDs?

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [X ] Previously working device (other platform)
    - zwavejs2mqtt with razberry pi hat zwave controler - multiple versions
    - swapped to  the silicon labs 
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [X] Manual Docker build
    - `node-zwave-js` branch: master
    - `zwavejs2mqtt` branch: master
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [] Other: 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '.ontrol pane;'
2. Click on 'heal network'
4. See error in logs

**Additional context**
Add any other context about the problem here.

**Logfile:**

AlCalzone, 2021-02-15:
Looks like the command was changed and the callback isn't correctly detected.

scyto, 2021-02-15:
I assume nothing I can to do to help?

AlCalzone, 2021-02-15:
Not at the moment

kpine, 2021-02-16:
I initiated a heal on my network with the silicon labs 700-series controller. There didn't seem to be any problem that I could tell. I've attached my log for comparison.

[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5991537/zwavejs_1.log)

AlCalzone, 2021-02-16:
In the log above, we get the confirmation that the healing process was started, but for some reason the "Completed" message never comes. I'm investigating whether that is something we need to handle differently.

kpine, 2021-02-16:
Two of my nodes are battery devices that never wake up, perhaps that's why?

AlCalzone, 2021-02-16:
Sorry, I was talking about scyto's log

scyto, 2021-02-16:
@kpine yeah your log  is different
I wonder if there is a difference between your SLUSB001A and mine?

I am in middle of trying to produce complete log, but after an hour I am at this, no idea how long it will take @AlCalzone does the attempts have a time out or will it go until i stop heal?
```
15:15:44.944 CNTRLR   [Node 014] refreshing neighbor list failed: Timeout while waiting for a callback from the controller
15:15:44.944 CNTRLR » [Node 014] refreshing neighbor list (attempt 4)...
                                        
```

kpine, 2021-02-16:
Do you know what version of firmware your stick has? I can't seem to find this information in zwave-js, but [according to](https://github.com/zwave-js/node-zwave-js/issues/1357#issuecomment-761520370) the PC Controller software mine is 7.11.

scyto, 2021-02-17:
Yes 7.11 here too.
I am running zwavejs in docker on a raspberry pi4 if that matters.
Very odd, i could try switching from one of the dockerfile.contrib builds to one off docker hub?

Full log from start of heal to stop FWIW which isn't much more than the fragment, interestingly the first time i ever tried this i swear it said failed in the log and the UI.  But that was on previous zwavejs version.

[network.heal.log](https://github.com/zwave-js/node-zwave-js/files/5992366/network.heal.log)

UI at end showed 
[image]

scyto, 2021-02-17:
yup same on issue for me on

  App Version = 1.1.1
  Zwavejs Version = 6.3.0

So that eliminates the self-built docker container being the issue.

AlCalzone, 2021-02-17:
@scyto same thing still happens in the last log 🤷🏻‍♂️

scyto, 2021-02-17:
@AlCalzone yup, I am confused why @kpine has no issue with his stick doing heal.

AlCalzone, 2021-02-17:
@scyto how large is your network? How many mains-powered nodes, how many frequent listening (FLIRS), how many sleeping nodes (old school battery powered)?

scyto, 2021-02-17:
@AlCalzone 19 device nodes, 16 mains powered, 3 FLIRS locks, and no battery devices.
All the  Jasco devices were bough circa 2013 as were all locks except the BE469ZP one.
The Fibaro is from 2013.
The aeotec devices are mixes of ages.

[image]

scyto, 2021-02-17:
Today I:

updated to latest public dev container
    - App Version 1.1.1
    - Zwavejs Version 6.4.0
    - Zwavejs-server Version 1.0.0-beta.
 
at the suggestion of @billiaz updated controller FW to 7.15 and raised gain to +4

looks like this fixed it:
```
13:24:30.297 SERIAL » 0x010600460a0171c5                                                   (8 bytes)
13:24:30.298 DRIVER » [Node 010] [REQ] [AssignReturnRoute]
                        payload: 0x0a0171
13:24:30.314 SERIAL « [ACK]                                                                   (0x06)
13:24:30.316 SERIAL « 0x0104014601bd                                                       (6 bytes)
13:24:30.317 SERIAL » [ACK]                                                                   (0x06)
13:24:30.319 DRIVER « [RES] [AssignReturnRoute]
                        has started: true
13:24:31.095 SERIAL « 0x01210046710000000000000000000000000000000000000000000000000000000 (35 bytes)
                      000e9
13:24:31.100 SERIAL » [ACK]                                                                   (0x06)
13:24:31.102 DRIVER « [REQ] [AssignReturnRoute]
                        callback id:     113
                        transmit status: OK
13:24:31.112 SERIAL » 0x010600460b0172c7                                                   (8 bytes)
13:24:31.113 DRIVER » [Node 011] [REQ] [AssignReturnRoute]
                        payload: 0x0b0172
13:24:31.123 SERIAL « [ACK]                                                                   (0x06)
13:24:31.125 SERIAL « 0x0104014601bd                                                       (6 bytes)
13:24:31.126 SERIAL » [ACK]                                                                   (0x06)
13:24:31.128 DRIVER « [RES] [AssignReturnRoute]
                        has started: true
13:24:32.688 SERIAL « 0x01210046720000000000000000000000000000000000000000000000000000000 (35 bytes)
                      000ea
13:24:32.689 SERIAL » [ACK]                                                                   (0x06)
13:24:32.690 DRIVER « [REQ] [AssignReturnRoute]
                        callback id:     114
                        transmit status: OK
13:24:32.699 SERIAL » 0x010600460c0173c1                                                   (8 bytes)
13:24:32.700 DRIVER » [Node 012] [REQ] [AssignReturnRoute]
                        payload: 0x0c0173
13:24:32.717 SERIAL « [ACK]                                                                   (0x06)
13:24:32.720 SERIAL « 0x0104014601bd                                                       (6 bytes)
13:24:32.720 SERIAL » [ACK]                                                                   (0x06)
13:24:32.722 DRIVER « [RES] [AssignReturnRoute]
                        has started: true
13:24:32.794 SERIAL « 0x01210046730000000000000000000000000000000000000000000000000000000 (35 bytes)
                      000eb
13:24:32.795 SERIAL » [ACK]                                                                   (0x06)
```

AlCalzone, 2021-02-17:
Cool!

scyto, 2021-02-17:
yeah, if those per controller plugins happen (i saw an issue somewhere on that) we might want to expose gain for controllers that support it? 
Want me to close?

closed 2021-02-17 by AlCalzone

kpine, 2021-02-17:
> updated controller FW to 7.15 and raised gain to +4

Do you have a reference on how to perform the update?

scyto, 2021-02-17:
> > updated controller FW to 7.15 and raised gain to +4
> 
> Do you have a reference on how to perform the update?

high level only:
- install simplicity studio and zwave tools (why do they make that so hard)
- do a backup of the stick
- when in the controller top level UI click OTW icon
- navigate to C:\SiliconLabs\SimplicityStudio\v5\developer\sdks\gecko_sdk_suite\v3.1\protocol\z-wave\Apps\bin\gbl
- choose (for US) ZW_SerialAPI_Controller_7.15.2_214_EFR32ZG14_REGION_US.gbl
- wait, and its done

For gain:
- install simplicity studio and zwave tools
- on the controller top level menu pick the 'radio menu' sorry forgot what it is called and it isn't loaded
- change the first box to +4
[image]

scyto, 2021-02-17:
[image]

hanskroner, 2021-02-18:
> For gain:
> 
> * install simplicity studio and zwave tools
> * on the controller top level menu pick the 'transmit settings'
> * change the first box to +4

It looks like you're based in the US. You should know the FCC has power output limitations on the 916MHz ISM channels used by Z-Wave in that region - they don't allow for anything higher than +0dBm.

My guess is that your controller is situated somewhere where there's "a lot" of RF interference. Older 500-series hardware uses region-specific SAW filters to combat this. The newer Z-Wave 700 hardware has better RF blocking performance, and the region-specific SAW filters aren't typically needed anymore (which is great for manufacturers, as it means they no longer have to make region-specific hardware). Maybe in your particular case, the controller is located too close to another RF source - is it possible for you to try repositioning it?

If there is an interference issue, just raising the output power will leave you with flaky connections anyway. Being able to "shout louder" is only half of the equation - you also need to be able to "listen better", which you don't have a setting for.

scyto, 2021-02-18:
@hanskroner  thanks for the insight. interesting.

All i can say is that no i placed this in a part of my house with the least RF i know, away from all other devices. (i have a stupid long cable so i could do just this :-) during testing.

That according to the log the stick was only having an issue when waiting for a response back - the logs were clear it was waiting for packet back and once i made *all* the changes the problem went away, but i made multiple changes at same time (stupid me).

I will back off the db setting and see if problem returns, if not then we may conclude the issue was something else.

If the problem returns I can also try a short usb extension cable and see if distancing the USB stick from the raspberry pi makes a difference (in case the pi is generating RF - maybe the poe hat wit fan and oled screen is causing an issue.....)

scyto, 2021-02-18:
i backed off the +4db and heal is still working so at this point ¯\(°_o)/¯
i will stop saying the +4db is what fixed it for me

scyto, 2021-02-19:
backing off the +4db was a big mistake, issues came back (healing, missing lifelines, etc)
someone else pointed out that +4db with 33 measured is actually -2.9 db so well under the 0db FCC rules.

hanskroner, 2021-02-20:
> backing off the +4db was a big mistake, issues came back (healing, missing lifelines, etc)
> someone else pointed out that +4db with 33 measured is actually -2.9 db so well under the 0db
I'm not intimately familiar with how the FCC testing is performed, I just remember the +0dBm figure. If you didn't need to tweak anything on your 500-series stick, that was outputting +0dBm as well, which is why my mind went to "interference issues." If you're missing Lifeline reports, an interference issue sounds even more likely - your stick isn't transmitting anything in that scenario, only receiving. Bumping your stick's output power doesn't affect the output power of any other nodes in the network.

> I get this is transmission - for beam formed packets used to wake up locks transmit is key....
Z-Wave doesn't do beam forming. The "beaming" used to wake-up FLiRS nodes is, in the US, essentially a second-long jam with a repeating pattern that FLiRS nodes scan for either 1 or 4 times per second, depending on the way they're setup. The output power these "beams" are transmitted at is the same as all other regular frames.

It's also only the last "hop" in a route that beams the FLiRS node, so your USB stick is responsible for beaming your lock only if the lock is in direct range of your stick.

scyto, 2021-02-20:
It is in direct range.