zwave-js/zwave-js #1765: Nodes constantly become dead

oleynikandrey, 2021-02-15:
Hello, I have pretty small z-wave network: z-way controller (RaZberry Controller 2016 ZWave+), 2 channel wall switch (Shenzhen Neo Electronics EU-2) and dimmer (McoHome Technology MH-P220). All devices are located in one room.

I'm constantly see error messages about timeouts, unexpected responses or nodes becoming dead.

Here is a part of the log:
```
Feb 12 15:09:24 pi npm[11047]: 15:09:24.383 CNTRLR « [Node 003] Received update for a Supervision session
Feb 12 15:09:34 pi npm[11047]: 15:09:34.341 CNTRLR   [Node 002] Timed out while waiting for a response from the node
Feb 12 15:09:35 pi npm[11047]: 15:09:35.596 CNTRLR « [Node 003] Received update for a Supervision session
Feb 12 15:09:38 pi npm[11047]: 15:09:38.180 CNTRLR « [Node 003] Received update for a Supervision session
Feb 12 15:09:39 pi npm[11047]: 15:09:39.684 CNTRLR   [Node 003] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
Feb 12 15:09:50 pi npm[11047]: 15:09:50.919 CNTRLR   [Node 003] Timed out while waiting for a response from the node
Feb 12 15:09:50 pi npm[11047]: 2021-02-12 15:09:50.921 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 71 => 85
Feb 12 15:09:52 pi npm[11047]: 2021-02-12 15:09:52.187 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:09:52 pi npm[11047]: 2021-02-12 15:09:52.966 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:09:53 pi npm[11047]: 2021-02-12 15:09:53.123 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:09:53 pi npm[11047]: 2021-02-12 15:09:53.304 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:09:56 pi npm[11047]: 15:09:56.778 CNTRLR   [Node 003] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
Feb 12 15:09:57 pi npm[11047]: 2021-02-12 15:09:57.608 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:09:58 pi npm[11047]: 2021-02-12 15:09:58.507 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:09:58 pi npm[11047]: 15:09:58.715 CNTRLR   [Node 003] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
Feb 12 15:09:58 pi npm[11047]: 2021-02-12 15:09:58.757 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:09:58 pi npm[11047]: 2021-02-12 15:09:58.883 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:09:59 pi npm[11047]: 2021-02-12 15:09:59.504 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:09:59 pi npm[11047]: 2021-02-12 15:09:59.532 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:09:59 pi npm[11047]: 2021-02-12 15:09:59.918 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:09:59 pi npm[11047]: 2021-02-12 15:09:59.966 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:10:00 pi npm[11047]: 2021-02-12 15:10:00.105 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:10:00 pi npm[11047]: 2021-02-12 15:10:00.191 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 85
Feb 12 15:10:00 pi npm[11047]: 15:10:00.265 CNTRLR   [Node 003] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
Feb 12 15:10:10 pi npm[11047]: 15:10:10.813 CNTRLR   [Node 003] Timed out while waiting for a response from the node
Feb 12 15:10:11 pi npm[11047]: 2021-02-12 15:10:11.138 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 85 => 99
Feb 12 15:10:12 pi npm[11047]: 15:10:12.210 CNTRLR « [Node 003] Received update for a Supervision session
Feb 12 15:10:15 pi npm[11047]: 15:10:15.240 CNTRLR   [Node 002] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
Feb 12 15:10:19 pi npm[11047]: 15:10:19.999 CNTRLR   [Node 002] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
Feb 12 15:10:24 pi npm[11047]: 15:10:24.792 CNTRLR   [Node 002] Node 2 did not respond after 3 attempts, it is presumed dead
Feb 12 15:10:24 pi npm[11047]: 2021-02-12 15:10:24.804 INFO ZWAVE: Node 2 is now dead
Feb 12 15:10:24 pi npm[11047]: 15:10:24.805 CNTRLR   [Node 002] The node is now dead.
Feb 12 15:10:33 pi npm[11047]: 2021-02-12 15:10:33.954 ERROR ZWAVE-SERVER: Unexpected error Failed to send the command after 3 attempts (Status NoAck)
Feb 12 15:10:33 pi npm[11047]: ZWaveError: Failed to send the command after 3 attempts (Status NoAck)
Feb 12 15:10:33 pi npm[11047]:     at Object.sendDataErrorToZWaveError (/opt/zwavejs2mqtt/node_modules/zwave-js/src/lib/driver/StateMachineShared.ts:112:12)
Feb 12 15:10:33 pi npm[11047]:     at /opt/zwavejs2mqtt/node_modules/zwave-js/src/lib/driver/SendThreadMachine.ts:437:4
Feb 12 15:10:33 pi npm[11047]:     at /opt/zwavejs2mqtt/node_modules/xstate/lib/utils.js:410:33
Feb 12 15:10:33 pi npm[11047]:     at Array.reduce (<anonymous>)
Feb 12 15:10:33 pi npm[11047]:     at Object.updateContext (/opt/zwavejs2mqtt/node_modules/xstate/lib/utils.js:400:25)
Feb 12 15:10:33 pi npm[11047]:     at Object.resolveActions (/opt/zwavejs2mqtt/node_modules/xstate/lib/actions.js:409:19)
Feb 12 15:10:33 pi npm[11047]:     at StateNode.resolveTransition (/opt/zwavejs2mqtt/node_modules/xstate/lib/StateNode.js:787:35)
Feb 12 15:10:33 pi npm[11047]:     at StateNode.transition (/opt/zwavejs2mqtt/node_modules/xstate/lib/StateNode.js:733:21)
Feb 12 15:10:33 pi npm[11047]:     at StateNode.resolveRaisedTransition (/opt/zwavejs2mqtt/node_modules/xstate/lib/StateNode.js:738:22)
Feb 12 15:10:33 pi npm[11047]:     at StateNode.resolveTransition (/opt/zwavejs2mqtt/node_modules/xstate/lib/StateNode.js:873:39)
Feb 12 15:10:33 pi npm[11047]:     at StateNode.transition (/opt/zwavejs2mqtt/node_modules/xstate/lib/StateNode.js:733:21)
Feb 12 15:10:33 pi npm[11047]:     at /opt/zwavejs2mqtt/node_modules/xstate/lib/interpreter.js:602:34
Feb 12 15:10:33 pi npm[11047]:     at Object.provide (/opt/zwavejs2mqtt/node_modules/xstate/lib/serviceScope.js:11:18)
Feb 12 15:10:33 pi npm[11047]:     at Interpreter.nextState (/opt/zwavejs2mqtt/node_modules/xstate/lib/interpreter.js:601:38)
Feb 12 15:10:33 pi npm[11047]:     at /opt/zwavejs2mqtt/node_modules/xstate/lib/interpreter.js:124:39
Feb 12 15:10:33 pi npm[11047]:     at Scheduler.process (/opt/zwavejs2mqtt/node_modules/xstate/lib/scheduler.js:60:13)
Feb 12 15:10:33 pi npm[11047]: 15:10:33.964 DRIVER   unexpected response, discarding...
Feb 12 15:10:33 pi npm[11047]: 15:10:33.966 DRIVER   unexpected response, discarding...
Feb 12 15:10:33 pi npm[11047]: 15:10:33.968 DRIVER   unexpected response, discarding...
Feb 12 15:10:33 pi npm[11047]: 15:10:33.972 DRIVER     no handlers registered!
Feb 12 15:10:33 pi npm[11047]: 15:10:33.974 DRIVER     no handlers registered!
Feb 12 15:10:33 pi npm[11047]: 15:10:33.977 DRIVER     no handlers registered!
Feb 12 15:10:33 pi npm[11047]: 15:10:33.985 CNTRLR » [Node 002] pinging the node...
Feb 12 15:10:44 pi npm[11047]: 15:10:44.117 CNTRLR   [Node 003] Timed out while waiting for a response from the node
Feb 12 15:10:48 pi npm[11047]: 15:10:48.371 CNTRLR   [Node 002] Node 2 did not respond after 1 attempts, it is presumed dead
Feb 12 15:10:48 pi npm[11047]: 15:10:48.397 CNTRLR   [Node 002] ping failed: Failed to send the command after 1 attempts (Status No
Feb 12 15:10:48 pi npm[11047]:                       Ack)
Feb 12 15:10:48 pi npm[11047]: 2021-02-12 15:10:48.400 ERROR ZWAVE-SERVER: Unexpected error The message cannot be sent because node 2 is dead
```

Could someone help me to find a reason for such behaviour?

AlCalzone, 2021-02-15:
`NoAck` is usually an indication for bad connectivity. There's also a crash in here which I do not like, but I can't see where it is coming from. I'd need to see zwave-js logs.

oleynikandrey, 2021-02-15:
Here is the log: [zwavejs.log](https://github.com/zwave-js/node-zwave-js/files/5980871/zwavejs.log)

AlCalzone, 2021-02-15:
Sorry, that log doesn't help --> https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report
You could try performing a network heal - maybe the nodes try to use a route that no longer exists.

oleynikandrey, 2021-02-15:
Here is all I've got:
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/5981221/zwavejs2mqtt-store.zip)
[nodes.zip](https://github.com/zwave-js/node-zwave-js/files/5981226/nodes.zip)

It's strange but I can't reproduce exception yet.

AlCalzone, 2021-02-17:
Something is really weird with your system. I can't tell whether it is the node or the stick - in any way, healing **could** work.

Often, the driver requests something, the device sends a response but shortly afterwards the controller claims that the device did not acknowledge the request.

oleynikandrey, 2021-02-19:
@AlCalzone, thank you for your time!

closed 2021-02-19 by oleynikandrey

lolouk44, 2021-09-15:
I have been having the same issue.
I also have a very small network: Aeotec Z-Stick Gen5, Yale Conexis L1 lock, and TZ68 socket (which I bought as a "bridge" between the stick and the lock).
I had built some automations in HA to refresh values / heal,
Normally when the lock shows up as dead, I can manually unlock and relock the lock and it shows back up as alive.

Today while the lock was showing as dead, I stopped zwave-JS and reverted to an old zwave integrationin HA. The lock showed up as working straight away without me touching it.
I will for now stick with the "old" zwave integration in HA, though I'm happy to bring Zwave-JS back if you need me to troubleshoot this

jooka1978, 2021-11-26:
Hi,

I have exactly the same problem. The zvawe2mqtt shows the nodes dead. If I manually turn on the switch it immediately comes back live again, but in 10-20 min it is dead again.
Can you please help? 
I can send logs (just please tell tell me where to find and what exactly to send).
Thanks

lolouk44, 2021-11-26:
A couple of things that greatly improved it for me:
1. Power the host off AND unplug the zwave usb stick, the plug back and restart 
2. I don't use batteries in the lock anymore, I instead use a power adapter

jooka1978, 2021-11-26:
> A couple of things that greatly improved it for me:
> 
> 1. Power the host off AND unplug the zwave usb stick, the plug back and restart
> 2. I don't use batteries in the lock anymore, I instead use a power adapter

Thanks, I will give it a try. My Shenzhen Neo Electronics EU-2 switches are in-wall swiches so powered by the electrical grid.

AlCalzone, 2021-11-26:
Something else you could check out or change is the stick location.

Z-Wave sticks in particular are prone to interference by USB ports, especially by USB3 ports. We recommend putting the stick in a suitable location:

-   on an **USB extension cord** (this works wonders!)
-   away from other USB ports
-   away from metallic surfaces
-   and especially not in the back of a server rack

Sn3akyP3t3, 2022-03-23:
Thanks for this obvious, but overlooked option!
`USB extension cord`

ddumas83, 2023-04-18:
> Thanks for this obvious, but overlooked option! `USB extension cord`

Wow... I was puzzled by the fact my z-wave devices couldn't communicate if I put them more than 2m away or behind a wall..
My Raspberry Pi 4 was standing on a wooden surface with the USB Z-wave stick directly attached.
I used to have a Raspberry pi 2 running Domoticz which worked ok, but also seemed to have communication flaws (and version was outdated).
This seems very simple, but it solved my problems immediately!
Thank you very much!