zwave-js/zwave-js #1766: [feat] Add support for Eaton Z-Wave Outlet Model:  RFTR9605-TAW

JWise1203, 2021-02-15:
Please add support for this device. It currently shows up as Unknown for the Product and Code columns.

Also, Home Assistant control is buggy.

I have attached the JSON file

[node_5.txt](https://github.com/zwave-js/zwavejs2mqtt/files/5978053/node_5.txt)

robertsLando, 2021-02-15:
I have moved the issue to the zwave-js repo

AlCalzone, 2021-02-15:
@blhoward2 we don't have this device:
Product Manufacturer: 0x001a
Product Type: 0x5244
Product ID: 0x0000

AlCalzone, 2021-02-15:
And @JWise1203 please further specify what you mean with
> Home Assistant control is buggy.

That is not a very specific problem description :)

blhoward2, 2021-02-16:
This file did actually exist, they share a product type/id between two very similar outlets. 96505/96605

closed 2021-02-16 by AlCalzone

JWise1203, 2021-02-16:
Thanks everyone, since the Product ID was different, will zwave-js be updated, or is there something I need to do on my end?

blhoward2, 2021-02-16:
It'll be updated to list both but it's just a label. It makes no functional difference.

PR #1774 (2021-02-16): feat(config): add eaton outlet to device file
Closes #1766