zwave-js/zwave-js #1768: Heatit Z-Smoke Detector  - missing device configuration

dffffffff, 2021-02-15:
Heatit Z-Smoke Detector 230V

Manufacturer ID: 0x019B 
Product ID: 0x3601 
Product Type ID: 0x0003   

- [ x] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [x ] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [ x] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

PDF Manual - https://media.heatit.com/709

https://products.z-wavealliance.org/products/3933?selectedFrequencyId=1
https://products.z-wavealliance.org/Products/3933/XML

I have no prior knowledge of how to add device configurations, I'm happy to help if there docs i can follow.

AlCalzone, 2021-02-15:
The label is shit, but we do have it:
https://devices.zwave-js.io/?jumpTo=0x019b:0x0003:0x3601:0.0

blhoward2, 2021-02-16:
It looks like Thermofloor is another example where we need to correct the brand.

dffffffff, 2021-02-16:
I will make an attempt at improving the file and make a PR

AlCalzone, 2021-02-16:
@blhoward2 is already on it. See linked PR

dffffffff, 2021-02-16:
See if any of this is helpful - https://github.com/dffffffff/node-zwave-js/commit/0001fad47af5517cf519d9bc4fd0e4abf812f02b

dffffffff, 2021-02-17:
Updated due to a typo.

https://github.com/dffffffff/node-zwave-js/blob/patch-4512689.json/packages/config/config/devices/0x019b/4512689.json

PR #1776 (2021-02-18): fix(config): correct heatit brand names
Closes #1768. Fixes Heatit brand names in some files, cleans up values.
closed 2021-02-18 by AlCalzone