zwave-js/zwave-js #1769: MCO home mh-c421 can't put up or down

S1M8N, 2021-02-15:
**Describe the bug**

Hello everyone,

**I published a post a month ago about configuring mco home modules (https://github.com/zwave-js/node-zwave-js/issues/1401). I added my mh-c421 module and everything seems to be good (just the description which is not perfectly correct it is not a Micro Shutter but a shutter panel ... this is not very important). However, I noticed that it was still not possible to raise or lower the shutter as I wanted (it is always considered as an on / off switch and when I activate it, the shutter goes down but I do not have no option to bring it up ...).**

If I use the parameters of the multilevel Switch I have this message:
[image]

Since I mentioned them in my old post, there are already some log files and some answers to possible questions.

If the problem is just my home, can you tell me how to fix it?

I remain available for any questions or to give you any file.

thanks a lot for your help

Have good day

Simon

AlCalzone, 2021-02-15:
I can't find Node 26 in your other logs. Please just make a new one and post your current cache file (`<homeid>.json`) here, so the information is not spread through multiple issues.

S1M8N, 2021-02-15:
> I can't find Node 26 in your other logs. Please just make a new one and post your current cache file (`<homeid>.json`) here, so the information is not spread through multiple issues.

Node 23 should be taken from the old post, node 26 is the same, I just excluded my device several times. But if you really want the logs, I can send them to you tomorrow.

Thank you

S1M8N, 2021-02-21:
I have that : 
[zwave_1 log.txt](https://github.com/zwave-js/node-zwave-js/files/6017700/zwave_1.log.txt)

It's ok ?

Thank you

AlCalzone, 2021-02-23:
No, sorry. See my other comment:

> [...] and post your current cache file (`<homeid>.json`) here

It wouldn't hurt if you also post a log where you re-interview the node. Since it was already queried before, I cannot see much in the logs.

S1M8N, 2021-02-24:
> No, sorry. See my other comment:
> 
> > [...] and post your current cache file (`<homeid>.json`) here
> 
> It wouldn't hurt if you also post a log where you re-interview the node. Since it was already queried before, I cannot see much in the logs.

@AlCalzone,

I have that : 

**[Log.zip](https://github.com/zwave-js/node-zwave-js/files/6038032/Log.zip)**

For the "**<homeid.json>.json**", if it's not in my .zip file please give me the exactely file location.

Thank you for give me your time !

S1M8N, 2021-02-24:
ok and on your side this can still be settled definitively on your side or temporarily?

AlCalzone, 2021-02-24:
The PR enables a workaround for the device. That's a bug in the device, not zwave-js.

S1M8N, 2021-02-24:
> The PR enables a workaround for the device. That's a bug in the device, not zwave-js.

PR #1890 (2021-02-24): fix(config): force Multi Channel CC to be supported for MH-C421
fixes: #1769
closed 2021-02-24 by AlCalzone