zwave-js/zwave-js #1771: [bug] couldn't send secure command to FE599NX lock until controller was next to lock.

scyto, 2021-02-15:
**Describe the bug**

I have four locks. 3 are working perfectly.  The front door has a Kwikset and Schlage FE599NX
After multiple interviews the FE599NX would not allow me to unlock door or set user codes.
After moving controller right next to lock, doing another interview i was able to to change lock from secured to unsecured.
I don't seem to have had other general communication errors with the FE599NX so don't understand why it wasn't accepting secure command until i moved it and re-interviewed.  Original inclusion was done with controller and lock next to each other.

**Device information**

Schlage FE599 NX
Node 004

**Last Known Working Configuration**
- [X] unknown
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [X] Manual Docker build
    - `node-zwave-js` branch: master
    - `zwavejs2mqtt` branch: master
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
see description

**Additional context**
Add any other context about the problem here.

**Logfile:**
[Copy of Copy of zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/5983871/Copy.of.Copy.of.zwavejs_1.log)

AlCalzone, 2021-02-15:
Looking at the log, after the re-interview it works as expected. My guess is that before moving it had a flaky connection to the controller and some of the commands used to determine whether the device is secure timed out.

Once you move it back, make sure to perform a network heal (once I've solved #1764 😅)

scyto, 2021-02-15:
Deal, once 1764 is closed and fixes the head i can retest and the close this one :-p

scyto, 2021-02-17:
fixed by this https://github.com/zwave-js/node-zwave-js/issues/1764#issuecomment-780865266

closed 2021-02-17 by scyto