zwave-js/zwave-js #1773: bug: Kwikset 910 lock doesn't appear to send any form of status change on door lock/unlock

scyto, 2021-02-15:
**Describe the bug**

when i physically lock or unlock the 910 it appears to send zero notifications
watching zwavejs_1.log with tail -f show no traffic occurring at all 

**Device information**

Kwikset 910
NODE 003

**Last Known Working Configuration**
- [X] unknown (think it worked years ago but can't be sure my memory is accurate)
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [X] Manual Docker build
    - `node-zwave-js` branch: master
    - `zwavejs2mqtt` branch: master
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Steps to reproduce the behavior:
1. restart container
2. wait for all interviews to complete
3. tail -f zwavejs_1.log at silly 
3. re-interview node 003
4. check lock works by changing secured/unsecured
5. verify messages flow in log as expected
6. walk over to door and hand turn bolt OR use external keypad to change secure state
7. observe not a single thing is appended to the zwave log (this is where the log ends at step 5 as nothing came into log)

**Additional context**
could this be related to https://github.com/zwave-js/node-zwave-js/issues/1764?

App Version
1.1.1+94ff7ed.94ff7ed

Zwavejs Version
6.3.0-8f22052

**Logfile:**
[910issue.log](https://github.com/zwave-js/node-zwave-js/files/5984551/910issue.log)

firstof9, 2021-02-15:
Here is a list of the alarm_type and their human readable meanings we've collected for the KeyMaster project.

```python
        0: "No Status Reported",
        9: "Lock Jammed",
        17: "Keypad Lock Jammed",
        21: "Manual Lock",
        22: "Manual Unlock",
        23: "RF Lock Jammed",
        24: "RF Lock",
        25: "RF Unlock",
        26: "Auto Lock Jammed",
        27: "Auto Lock",
        32: "All Codes Deleted",
        161: "Bad Code Entered",
        167: "Battery Low",
        168: "Battery Critical",
        169: "Battery Too Low To Operate Lock",
        16: "Keypad Unlock",
        18: "Keypad Lock",
        19: "Keypad Unlock",
        162: "Lock Code Attempt Outside of Schedule",
        33: "Code Deleted",
        112: "Code Changed",
        113: "Duplicate Code",
```
The alarm_level has always been the user code slot that's activated the alarm_type, for my 910 it's _always_ slot `1` unless unlocked via keypad externally, all other events trigger alarm_level `1`

I hope this is useful in some way.

scyto, 2021-02-16:
@firstof9 I think you mean to post that to  https://github.com/zwave-js/node-zwave-js/issues/1713 ?
This issue is about the alarm codes never showing up in the logs with my device.

firstof9, 2021-02-16:
Oops sorry. thanks.

scyto, 2021-02-16:
no thank  you for taking the time to make that table, its needed!

blhoward2, 2021-02-16:
As I believe was noted on Discord, this is most likely to happen when the controller isn't in the lifeline group. Have you verified that your stick is in Group 1?

scyto, 2021-02-16:
I looked as suggested on discord. It had no groups with define node in the groups tab. So I added this, it was the only group it offered. Didn't see, to fix anything.[image]

scyto, 2021-02-16:
seems after that once the associate group was set  a reboot and re-interview got it working.
interesting that the above wasn't set automatically like all the other locks?

blhoward2, 2021-02-16:
Reopening this because we actually need to fix the issue. The device files don't have association definitions and it appears these locks don't auto-associate group 1 as a lifeline. 

I'll update the device file to fix this.

scyto, 2021-02-16:
The model I have is definitely not zwave plus if that helps.

AlCalzone, 2021-02-16:
> The model I have is definitely not zwave plus if that helps.

That explains why the lifeline does not get configured automatically then.

cppsse, 2021-02-16:
I was having what I think is same problem with my Kwikset 888 deadbolt.  Manual lock/unlock would not send an event to HA.

AlCalzone, 2021-02-16:
You can always add that association manually though.

cppsse, 2021-02-16:
> You can always add that association manually though.

I might give that a try now that I know it is an option.  Thank you!

alterbridge86, 2021-02-16:
> > The model I have is definitely not zwave plus if that helps.
> 
> That explains why the lifeline does not get configured automatically then.

Just adding in that I stumbled upon this issue because my Yale YRD216 is experiencing the same issue, no Alarm Level/Type reports being received. This IS a zwave Plus device, and I still didn't get a lifeline auto configured. I also have a Kwikset 914C which is zwave plus and it too did not have a lifeline configured.

closed 2021-02-17 by AlCalzone

bryantlee, 2021-02-19:
@AlCalzone hate to bump this - how do I know what version this fix got merged into?  Just not sure if we should have seen a fix or if its still needs troubleshooting.  I am on 6.4.0

blhoward2, 2021-02-19:
These would be in the dev image of zwave2jsmqtt the day after they were merged. And in any subsequent release of node. Note that you need to know which version of node is being pulled in. The HA integration is further behind, but you'd need to ask them.

AlCalzone, 2021-02-19:
You can see that on the merge commits:
[image]
No tag here, so not released officially yet.

scyto, 2021-02-20:
At the risk of making myself look stupid, i am not sure us regular folks can see that.... i am pretty certain these made it into 6.

[image]
[image]

AlCalzone, 2021-02-20:
If you scroll down in the PR description you'll see something like "AlCalzone merged commit 1234567 into master". Click that commit and you'll see it.

scyto, 2021-02-20:
Thanks, I had no idea that black number was clickable. I had been furiously clicking hide and I hide details.
For reference for others click the black number highlighted.
[image]

PR #1792 (2021-02-17): fix(config): add Association Groups to Kwikset locks
Closes #1773