zwave-js/zwave-js #1781: Pending Node Interviews, waiting for the lowest NodeID to be interviewed.

marcus-j-davies, 2021-02-16:
**Describe the bug**
It seems, for battery operated nodes that have a pending interview, they will not get interviewed, until the lowest NodeID, has been woken up, and interviewed.

As an example, if Nodes 3,4 have a pending interview, waking up node 4, does not result in an interview,
until node 3 has been woken up/interviewed. Once that happens, the interview for 4 then immediately starts. 

**Device information**
2 x Aeotec Door/Window Sensor 7 (but may affect others)
https://devices.zwave-js.io/?jumpTo=0x0371:0x0002:0x0007:0.0

**Last Known Working Configuration**
- [ ] New device
- [x] Previously working device (node-zwave-js)
    - I have only noticed this since 6.3.0 (but can't be sure)
    - Only updating the lib version

**Installation information**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [x] Other: Node-Red Plugin

**To Reproduce**
1 - Ensure battery operated nodes are a sleep - to cause an interview to be queued.
2 - wake up a node - but ensure its not the lowest ID node pending an interview (it will not get interviewed).
3 - Wake up the lowest NodeID pending an interview, (it will get interviewed, including the previous awaken one)

Line : 1203 -> Waking up Node 4 (no interview)
Line : 1217 -> Waking up Node 3 (interview, +interview for Node 4)

[zwave-2336.log](https://github.com/zwave-js/node-zwave-js/files/5988415/zwave-2336.log)

Hopefully have submitted this all correctly?

AlCalzone, 2021-02-16:
@marcus-j-davies in a few minutes a debug build will be available in https://github.com/zwave-js/node-zwave-js/pull/1783
Please install that and send me another log.

marcus-j-davies, 2021-02-16:
@AlCalzone ,
[zwave-7504.log](https://github.com/zwave-js/node-zwave-js/files/5988604/zwave-7504.log)

marcus-j-davies, 2021-02-16:
@AlCalzone 
https://github.com/zwave-js/node-zwave-js/pull/1784 fixes the bug (log Attached).

Oddly (but maybe more device location specific)
the interview for the first awaken node, is super fast (<=5 seconds) Node 3 ~7 seconds

I checked the cache, to ensure it really was interviewed - it was

[zwave-3120.log](https://github.com/zwave-js/node-zwave-js/files/5988926/zwave-3120.log)

AlCalzone, 2021-02-16:
Yup, Node 4 has a really low delay in responses.

PR #1784 (2021-02-16): fix: improve communication with sleeping nodes when WakeUp CC is unsupported or support not yet known
This PR implements what was discussed in https://github.com/zwave-js/node-zwave-js/discussions/1537 - meaning that messages to sleeping nodes will now be sent if they don't support the Wake Up CC

Messages to sleeping nodes will now also be added to the wakeup queue if we don't know yet if they support Wake Up.
fixes: #1781 
closes: #1783
closed 2021-02-16 by AlCalzone