zwave-js/zwave-js #1794: Config: Add check for duplicate parameters

AlCalzone, 2021-02-17:


AlCalzone, 2021-02-17:
@blhoward2 We can't do this - at least not easily.
https://github.com/json5/json5/pull/217

One other way would be to use the JSON schema to validate the files, but that won't pick up duplicate keys in templates.

blhoward2, 2021-02-18:
Why not just build a Set of the parameter numbers since they have to be unqiue, then compare the size to the number of parameters? If it is less there is a duplicate value somewhere? That wouldn't identify which is the duplicate but you could at least display an error message and fail.

AlCalzone, 2021-02-18:
The thing is, after `JSON5.parse` we don't see the duplicate. I could change it so the comments get stripped and the file parsed with the built-in JSON class, which would throw on duplicates.

blhoward2, 2021-02-18:
Ah, yeah I was thinking about after it's already in memory. I'm not sure it's a huge deal.

AlCalzone, 2021-02-18:
Should be easy enough to do with a regex 🤷🏻‍♂️

blhoward2, 2021-02-18:
I wouldn't worry about it. I can just not make mistakes, lol. Hopefully the templating would cut down on it too.

closed 2021-11-21 by AlCalzone