zwave-js/zwave-js #1797: Zipato Smoke sensors not interviewed

Taraman17, 2021-02-17:
I migrated from OZW Beta to ZWave-JS in Homeassistant on a Rapi with a standard Supervisor installation.
My Smoke sensors () are not correctly interviewed (except of one).
The working one is correctly identified, so the model is supported in ZwaveJS.

When waking up one of the devices, The server gets the notification, but does not react to it.

I did not do anything different when the working one was interviewed. I did it shortly after starting the server, but I also tried that with the others - maybe I just hit the one right moment then...

**Device information**
Zipato HS1SA Somke sensor
NodeIDs: 
35,36,38,39,40,41,45,46
Working one: 44

**Last Known Working Configuration**
- [X ] Previously working device (other platform)
    - Homeassistant with OZW Beta
    - Migrated to ZwaveJS standard addon.
    
**Installation information**
How did you install `node-zwave-js`?
- [X] Other: Homeassistant Z-Wave JS Addon via Supervisor

**To Reproduce**
Steps to reproduce the behavior:
1. Wakeup device
2. open log
3. scroll to end

**Logfile:**
Since I don't know how  to get the complete log from the Supervisor addon, here are the respective lines for now:

The correctly discovered node shows the following when waking up:
```
09:23:23.592 CNTRLR « [Node 044] received wakeup notification
09:23:23.594 CNTRLR   [Node 044] The node is now awake.
09:23:23.866 CNTRLR « [Node 044] received response for battery information:
                      level:                           70
09:23:23.990 CNTRLR   [Node 044] WakeUpCC: doing a partial interview...
09:23:23.993 CNTRLR » [Node 044] retrieving wakeup interval from the device...
09:23:24.074 CNTRLR « [Node 044] received wakeup configuration:
                      wakeup interval: 86400 seconds
                      controller node: 1
09:23:24.087 CNTRLR   [Node 044] BinarySensorCC: doing a partial interview...
09:23:24.089 CNTRLR » [Node 044] querying current value for Smoke...
09:23:24.173 CNTRLR « [Node 044] received current value for Smoke: false
09:23:24.213 CNTRLR   [Node 044] AssociationCC: doing a partial interview...
09:23:24.216 CNTRLR » [Node 044] querying association group #1...
09:23:24.298 CNTRLR « [Node 044] received information for association group #1:
                      maximum # of nodes: 1
                      currently assigned nodes: 1
09:23:24.299 CNTRLR » [Node 044] querying association group #2...
09:23:24.367 CNTRLR « [Node 044] received information for association group #2:
                      maximum # of nodes: 5
                      currently assigned nodes: 
09:23:24.369 CNTRLR » [Node 044] querying association group #3...
09:23:24.466 CNTRLR « [Node 044] received information for association group #3:
                      maximum # of nodes: 5
                      currently assigned nodes: 
09:23:24.478 CNTRLR   [Node 044] AssociationGroupInfoCC: doing a partial interview...
09:23:24.482 CNTRLR   [Node 044] NotificationCC: doing a partial interview...
09:23:24.487 CNTRLR   [Node 044] Interview stage completed: CommandClasses
09:23:24.489 CNTRLR   [Node 044] Interview stage completed: OverwriteConfig
09:23:24.491 CNTRLR » [Node 044] requesting node neighbors...
09:23:24.530 CNTRLR « [Node 044]   node neighbors received: 1, 26
09:23:24.531 CNTRLR   [Node 044] Interview stage completed: Neighbors
09:23:24.533 CNTRLR   [Node 044] Interview completed
09:23:25.538 CNTRLR » [Node 044] Sending node back to sleep...
09:23:25.608 CNTRLR   [Node 044] The node is now asleep.
```

All other Smoke sensors when waking up:
```
11:54:31.271 CNTRLR « [Node 022] received wakeup notification
11:57:41.851 CNTRLR « [Node 036] received wakeup notification
11:57:41.937 CNTRLR   [Node 036] The node is awake.
11:58:31.249 CNTRLR « [Node 022] received wakeup notification
```

AlCalzone, 2021-02-17:
This **might** be a duplicate of #1781, which was fixed in v6.3.1.
Can you please test that version?

Taraman17, 2021-02-17:
I doubt it, since I tried to wake up node 35, which is the lowest one that is not interviewed yet.

Or does that include nodes, that are already interviewed?

AlCalzone, 2021-02-17:
> And how do I test the latest releases in Homeassistant? Is it in the Beta channel?

I don't think that is possible. Running zwavejs2mqtt standalone would be easier:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/docker_custom_builds

And I need a full log (debug level) which includes the serial data.

Taraman17, 2021-02-17:
I fear, that is beyond my possibilities right now. I have Homeassistant running on a Raspi and need that for daily use. If I mess around with it too much, the WAF would drop inacceptably.
What I could do is to switch to zwavejs2mqtt if I could get the logs out of there to help and wait until the next release is live to try if the problem is solved.

AlCalzone, 2021-03-03:
Anything new? HASS has updated its zwave-js version in the meantime.

Taraman17, 2021-03-04:
Updated this morning and I have ZwaveJStoMQTT running now and thus have logs available.

When I wake up the device, it is still not interviewed. (Also a Z-Wave.me Wallcontroller).

But with the new version, when I select "re-interview node" and wake it up just thereafter, it gets interviewed.
Attached is the Log from the addon. Nodes in question are 17 (Wall Controller) & 38 (Smoke Sensor).

[zwavejs_355.log](https://github.com/zwave-js/node-zwave-js/files/6082762/zwavejs_355.log)

AlCalzone, 2021-03-04:
Thanks that helped. Essentially the driver forgot to assume sleeping nodes to be asleep after a restart if the interview was **not** completed before.
Can you try the linked PR?

closed 2021-03-04 by AlCalzone

Taraman17, 2021-03-05:
> Can you try the linked PR?

Unfortunately not.
My Homeassistant on Pi is my production system and I don't have another one to play around with.

PR #1990 (2021-03-04): fix: assume sleeping nodes to be asleep restart w/ incomplete interview
This PR improves on #1784 which was missing the situation where sleeping nodes would still be in ProtocolInfo stage after a restart. We now assume sleeping nodes to always be asleep after restarting from cache, independently of the interview stage.

Fixes: #1797