zwave-js/zwave-js #1803: Zooz Zen77 not updating when operated manually

mattstocum, 2021-02-17:
**Describe the bug**

When operating the Zooz Zen77 manually (at the switch, and not via a Z-Wave command) the device status is not updated in Zwave-JS. Other Zooz Zen switches do update automatically when operated manually.

**Device information**

Which device(s) is/are affected (make/model)?   Zooz Zen77 firmware 1.2
What are the node IDs? 35

**Last Known Working Configuration**
- [X] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [X] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Steps to reproduce the behavior:
1. Click on or off button on Zen77

**Additional context**
In the log below, 13:26:00.833 is when I pressed the button to turn the switch on. At 13:26.02:905 I'm manually sending a pollValue to the switch triggered by the CentralSceneCCNotification. I do believe this is a firmware bug, and have opened ticket #ZOO00050170 with Zooz.

**Logfile:**
13:26:00.833 SERIAL « 0x010b00040023055b0306800109                                        (13 bytes)
13:26:00.834 SERIAL » [ACK]                                                                   (0x06)
13:26:00.835 DRIVER « [Node 035] [REQ] [ApplicationCommand]
                      └─[CentralSceneCCNotification]
                          sequence number: 6
                          key attribute:   KeyPressed
                          scene number:    1
13:26:00.836 CNTRLR   [Node 035] [!] [Central Scene] scene[001]: 0
13:26:00.838 CNTRLR « [Node 035] received CentralScene notification {
                          "nodeId": 35,
                          "ccId": "Central Scene",
                          "ccCommand": "0x03",
                          "payload": "0x068001"
                      }
13:26:02.712 SERIAL « 0x010e0004001c0832022174000028238f                                  (16 bytes)
13:26:02.713 CNTRLR   [Node 028] [~] [Meter] value[66049]: 0.012 => 10.275              [Endpoint 0]
13:26:02.715 CNTRLR   [Node 028] [~] [Meter] deltaTime[66049]: 0 => 0                   [Endpoint 0]
13:26:02.716 SERIAL » [ACK]                                                                   (0x06)
13:26:02.718 DRIVER « [Node 028] [REQ] [ApplicationCommand]
                      └─[MeterCCReport]
                          type:       Electric
                          scale:      W
                          rate type:  Consumed
                          value:      10.275
                          time delta: 0 seconds
13:26:02.905 SERIAL » 0x01090013230226022522e7                                            (11 bytes)
13:26:02.906 DRIVER » [Node 035] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      34
                      └─[MultilevelSwitchCCGet]
13:26:02.919 SERIAL « [ACK]                                                                   (0x06)
13:26:02.920 SERIAL « 0x0104011301e8                                                       (6 bytes)
13:26:02.920 SERIAL » [ACK]                                                                   (0x06)
13:26:02.921 DRIVER « [RES] [SendData]
                        was sent: true
13:26:03.141 SERIAL « 0x0107001322000017de                                                 (9 bytes)
13:26:03.141 SERIAL » [ACK]                                                                   (0x06)
13:26:03.142 DRIVER « [REQ] [SendData]
                        callback id:     34
                        transmit status: OK
13:26:03.279 SERIAL « 0x010b00040023052603636300f3                                        (13 bytes)
13:26:03.280 CNTRLR   [Node 035] [~] [Multilevel Switch] targetValue: 0 => 99           [Endpoint 0]
13:26:03.282 CNTRLR   [Node 035] [~] [Multilevel Switch] duration: {"value":0,"unit":"s [Endpoint 0]
                      econds"} => {"value":0,"unit":"seconds"}
13:26:03.284 CNTRLR   [Node 035] [~] [Multilevel Switch] currentValue: 0 => 99          [Endpoint 0]
13:26:03.286 SERIAL » [ACK]

AlCalzone, 2021-02-17:
You could try assigning the controller node to one of the other associations:
[image]
Maybe it does not report the Multilevel Switch values on the Lifeline.

Also, having a complete log (as an attached file) of a re-interview would be good.

mattstocum, 2021-02-17:
`19:57:38.872 CNTRLR   [Node 035] Beginning interview - last completed stage: None
19:57:38.872 CNTRLR   [Node 035] new node, doing a full interview...
19:57:38.873 CNTRLR » [Node 035] querying protocol info...
19:57:38.879 SERIAL » 0x010400412399                                                       (6 bytes)
19:57:38.880 DRIVER » [REQ] [GetNodeProtocolInfo]
                        payload: 0x23
19:57:38.884 SERIAL « [ACK]                                                                   (0x06)
19:57:38.887 SERIAL « 0x01090141d39c01041101ec                                            (11 bytes)
19:57:38.889 SERIAL » [ACK]                                                                   (0x06)
19:57:38.890 DRIVER « [RES] [GetNodeProtocolInfo]
                        payload: 0xd39c01041101
19:57:38.894 CNTRLR « [Node 035] received response for protocol info:
                      basic device class:    Routing Slave
                      generic device class:  Multilevel Switch
                      specific device class: Multilevel Power Switch
                      is a listening device: true
                      is frequent listening: false
                      is a routing device:   true
                      is a secure device:    unknown
                      is a beaming device:   true
                      maximum baud rate:     40000 kbps
                      version:               4
19:57:38.895 CNTRLR   [Node 035] Interview stage completed: ProtocolInfo
19:57:38.895 CNTRLR » [Node 035] pinging the node...
19:57:38.900 SERIAL » 0x0108001323010025d635                                              (10 bytes)
19:57:38.901 DRIVER » [Node 035] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      214
                      └─[NoOperationCC]
19:57:38.902 SERIAL « [ACK]                                                                   (0x06)
19:57:38.909 SERIAL « 0x0104011301e8                                                       (6 bytes)
19:57:38.909 SERIAL » [ACK]                                                                   (0x06)
19:57:38.909 DRIVER « [RES] [SendData]
                        was sent: true
19:57:39.125 SERIAL « 0x01070013d60000162b                                                 (9 bytes)
19:57:39.125 SERIAL » [ACK]                                                                   (0x06)
19:57:39.125 DRIVER « [REQ] [SendData]
                        callback id:     214
                        transmit status: OK
19:57:39.132 CNTRLR   [Node 035] The node is alive.
19:57:39.133 CNTRLR « [Node 035] ping successful
19:57:39.133 CNTRLR » [Node 035] querying node info...
19:57:39.136 SERIAL » 0x0104006023b8                                                       (6 bytes)
19:57:39.136 DRIVER » [Node 035] [REQ] [RequestNodeInfo]
                        payload: 0x23
19:57:39.138 SERIAL « [ACK]                                                                   (0x06)
19:57:39.144 SERIAL « 0x01040160019b                                                       (6 bytes)
19:57:39.144 SERIAL » [ACK]                                                                   (0x06)
19:57:39.145 DRIVER « [RES] [RequestNodeInfo]
                        payload: 0x01
19:57:39.563 SERIAL « 0x011900498423130411015e26705b858e595586725a87739f6c7a88            (27 bytes)
19:57:39.564 SERIAL » [ACK]                                                                   (0x06)
19:57:39.565 DRIVER « [Node 035] [REQ] [ApplicationUpdateRequest]
                        payload: 0x8423130411015e26705b858e595586725a87739f6c7a
19:57:39.568 CNTRLR « [Node 035] node info received
                      supported CCs:
                      · Z-Wave Plus Info
                      · Multilevel Switch
                      · Configuration
                      · Central Scene
                      · Association
                      · Multi Channel Association
                      · Association Group Information
                      · Transport Service
                      · Version
                      · Manufacturer Specific
                      · Device Reset Locally
                      · Indicator
                      · Powerlevel
                      · Security 2
[... 1609 lines cut ...]
                        was sent: true
19:58:06.905 SERIAL « 0x010700130f000022c6                                                 (9 bytes)
19:58:06.906 SERIAL » [ACK]                                                                   (0x06)
19:58:06.907 DRIVER « [REQ] [SendData]
                        callback id:     15
                        transmit status: OK
19:58:06.910 CNTRLR   [Node 035] [+] [Central Scene] interviewComplete: true [Endpoint 0] [internal]
19:58:06.927 CNTRLR   [Node 035] Interview stage completed: CommandClasses
19:58:06.928 CNTRLR   [Node 035] Interview stage completed: OverwriteConfig
19:58:06.929 CNTRLR » [Node 035] requesting node neighbors...
19:58:06.934 SERIAL » 0x01070080230000005b                                                 (9 bytes)
19:58:06.934 DRIVER » [Node 035] [REQ] [GetRoutingInfo]
                        remove non-repeaters: false
                        remove bad links:     false
19:58:06.936 SERIAL « [ACK]                                                                   (0x06)
19:58:06.939 SERIAL « 0x01200180100a92da3891000000000000000000000000000000000000000000000 (34 bytes)
                      0a5
[... cut ...]               (0x06)
19:58:06.945 DRIVER « [RES] [GetRoutingInfo]
                        node ids: 5, 10, 12, 18, 21, 24, 26, 28, 29, 31, 32, 36, 37, 38, 41, 45, 48
19:58:06.948 CNTRLR « [Node 035]   node neighbors received: 5, 10, 12, 18, 21, 24, 26, 28, 29, 31, 3
                      2, 36, 37, 38, 41, 45, 48
19:58:06.949 CNTRLR   [Node 035] Interview stage completed: Neighbors
19:58:06.949 CNTRLR   [Node 035] Interview completed
19:58:06.998 CNTRLR   [Node 035] The node is ready to be used`

mattstocum, 2021-02-17:
[reinterview.txt](https://github.com/zwave-js/node-zwave-js/files/5998274/reinterview.txt)

mattstocum, 2021-02-17:
I just added the controller to the other two groups. I'll check if that changes things later today.

AlCalzone, 2021-02-17:
```
19:58:03.048 DRIVER « [Node 035] [REQ] [ApplicationCommand]
                      └─[AssociationGroupInfoCCCommandListReport]
                          group id: 1
                          commands:
                          · Device Reset Locally: 0x01
                          · Multilevel Switch: 0x03
                          · Basic: 0x03
                          · Central Scene: 0x03
                          · Indicator: 0x03
```
At least the device claims that it sends unsolicited Multilevel Switch commands to the lifeline. Let's see what the Zooz support has to say.

mattstocum, 2021-02-17:
Adding the controller to groups 2 and 3 did get things working properly as a temporary workaround. I’ll update further when I hear back from Zooz.

blhoward2, 2021-03-04:
Any update from Zooz support?

mattstocum, 2021-03-09:
> Any update from Zooz support?

They want to RMA the switch, I haven't had time yet unfortunately.

AlCalzone, 2021-03-16:
Okay I'm going to close this issue for now. Feel free to reopen if it persists with the RMA'ed switch.

closed 2021-03-16 by AlCalzone