zwave-js/zwave-js #1806: ~20 minute delay when completing lock interview on server start

rccoleman, 2021-02-17:
**Describe the bug**

When I restart the server, my 3 Kwikset 914 locks have a ~20min unexpected delay after retrieving usercodes and before transitioning from RestartFromCache to Complete.  These locks have all been added securely and allowed to complete their interview, but this delay is observed every time when the zwjs server restarts.

The logs below are with "INFO" enabled, but the pattern is also present with "DEBUG" and I've included the log with debug enabled below.
[zwavejs2mqtt-debug.zip](https://github.com/zwave-js/node-zwave-js/files/5997818/zwavejs2mqtt-debug.zip)

Scan starts here:

```
2021-02-17 16:59:40.106 INFO MQTT: Connecting to mqtt://192.168.1.159:1883
2021-02-17 16:59:40.156 INFO ZWAVE: Connecting to /dev/ttyACM0
2021-02-17 16:59:40.178 INFO APP: Listening on port 8091 host 0.0.0.0 protocol HTTP
2021-02-17 16:59:40.225 INFO MQTT: MQTT client connected
2021-02-17 16:59:45.691 INFO ZWAVE: Zwave driver is ready
2021-02-17 16:59:45.692 INFO ZWAVE: Controller status: Driver ready
2021-02-17 16:59:45.706 INFO ZWAVE: Scanning network with homeid: 0xe92faa0a
```

The scan proceeds, setting most line-powered nodes "Complete", and retrieves all 30 usercodes from all 3 locks.  At ~17:06:xx, metadata for the 112-0-xx parameters is updated for all 3 locks (sample of last communication):

```
2021-02-17 17:06:30.155 INFO ZWAVE: Node 64: metadata updated: 112-0-31-4
2021-02-17 17:06:30.156 INFO ZWAVE: Node 64: metadata updated: 112-0-31-8
2021-02-17 17:06:32.127 INFO ZWAVE: Node 44: value updated: 112-0-1 1 => 1
2021-02-17 17:06:33.603 INFO ZWAVE: Node 45: value updated: 112-0-1 1 => 1
```

At this point, all 3 locks are marked RestartFromCache, and remain that way for exactly *20 minutes*.  During this time, there is no activity to any of the three locks, but other nodes report in and work properly.  After 20mins, I finally see some activity with the locks that goes like this:

```
2021-02-17 17:26:24.938 INFO ZWAVE: Node 44: value updated: 112-0-33 960049460 => 960049460
2021-02-17 17:26:26.450 INFO ZWAVE: Node 45: value updated: 112-0-33 960049460 => 960049460
2021-02-17 17:26:28.394 INFO ZWAVE: Node 64: value updated: 112-0-33 960049460 => 960049460
2021-02-17 17:26:30.384 INFO ZWAVE: Node 44: value updated: 112-0-34 808464947 => 808464947
2021-02-17 17:26:31.839 INFO ZWAVE: Node 45: value updated: 112-0-34 808464947 => 808464947
2021-02-17 17:26:32.424 INFO ZWAVE: Node 64: value updated: 112-0-34 808464947 => 808464947
2021-02-17 17:26:33.083 INFO ZWAVE: Node 44: value updated: 112-0-31-1 1 => 1
2021-02-17 17:26:33.086 INFO ZWAVE: Node 44: value updated: 112-0-31-2 0 => 0
2021-02-17 17:26:33.088 INFO ZWAVE: Node 44: value updated: 112-0-31-4 1 => 1
2021-02-17 17:26:33.092 INFO ZWAVE: Node 44: value updated: 112-0-31-8 0 => 0
2021-02-17 17:26:33.325 INFO ZWAVE: Node 45: value updated: 112-0-31-1 1 => 1
2021-02-17 17:26:33.328 INFO ZWAVE: Node 45: value updated: 112-0-31-2 0 => 0
2021-02-17 17:26:33.330 INFO ZWAVE: Node 45: value updated: 112-0-31-4 1 => 1
2021-02-17 17:26:33.332 INFO ZWAVE: Node 45: value updated: 112-0-31-8 0 => 0
2021-02-17 17:26:33.756 INFO ZWAVE: Node 42: value updated: 50-0-value-66049 0 => 0
2021-02-17 17:26:33.758 INFO ZWAVE: Node 42: value updated: 50-0-deltaTime-66049 0 => 0
2021-02-17 17:26:33.933 INFO ZWAVE: Node 64: value updated: 112-0-31-1 1 => 1
2021-02-17 17:26:33.936 INFO ZWAVE: Node 64: value updated: 112-0-31-2 0 => 0
2021-02-17 17:26:33.938 INFO ZWAVE: Node 64: value updated: 112-0-31-4 1 => 1
2021-02-17 17:26:33.940 INFO ZWAVE: Node 64: value updated: 112-0-31-8 0 => 0
2021-02-17 17:26:36.311 INFO ZWAVE: Node 44: interview completed, all values are updated
2021-02-17 17:26:36.324 INFO ZWAVE: Node 45: interview completed, all values are updated
2021-02-17 17:26:36.919 INFO ZWAVE: Node 64: interview completed, all values are updated
```

When this happens, the locks are finally all marked Complete.  Other nodes work fine during this 20 minute delay, both battery and line-powered.  When these locks are finally marked complete, they work fine remotely and when manually actuated.  The locks are marked "Alive" during this whole time, as viewed from the ZWJS2MQTT web ui.

I started to notice this behavior when the change to ignore sleeping nodes when marking the scan complete, but it's possible that it's been this way for a while.

**Device information**

Which device(s) is/are affected (make/model)?  Kwikset 914 locks
What are the node IDs? 44, 45, 64

**Last Known Working Configuration**
- [ ] New device
- [x] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [x] Manual Docker build
    - `node-zwave-js` branch: master
    - `zwavejs2mqtt` branch: master
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Steps to reproduce the behavior:
1. Restart the zwjs2mqtt server and monitor the zwjs2mqtt web ui and log
2. Observe 20 minute delay in communication with a Kwikset 914 lock between reading the usercodes and finally marking the device "Complete"

**Additional context**
Add any other context about the problem here.

**Logfile:**

Here's a log with "INFO" enabled: [zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5997645/zwavejs2mqtt.log)

Here's a log with "DEBUG" enabled with the same behavior (search for "Node 45" from the bottom until you see the 20 minute gap): [zwavejs2mqtt-debug.zip](https://github.com/zwave-js/node-zwave-js/files/5997830/zwavejs2mqtt-debug.zip)

AlCalzone, 2021-02-17:
Are they battery powered? If so, are you on 6.3.1?

rccoleman, 2021-02-17:
Yes, they're battery-powered Kwikset 914 locks.  I pulled from master of both zwjs and zwjs2mqtt this morning and rebuilt my container, but the behavior has been consistent for a while now.  This is the version info from zwjs2mqtt:

Home ID 3912215050
Home Hex 0xe92faa0a
App Version 1.1.1+2ea9944.2ea9944
Zwavejs Version 6.4.0-ad7995f8

AlCalzone, 2021-02-17:
Okay, but that is expected (at least for now). Those are battery-powered nodes that have been previously interviewed completely, so the driver knows everything about them it needs to know. They stay in status `RestartFromCache` until they wake up again and their values get refreshed.

It doesn't sound like there is another problem, or am I missing something?

rccoleman, 2021-02-17:
I came across this when I was debugging another seemingly-related issue that I should have mentioned.  During that 20min period of time before the locks reflect "Complete", the Z-Wave network is very slow to respond.  It will often take several seconds, maybe even 10s, for devices to respond, and the delays are unpredictable in length.  Once the locks are marked Complete, device response across the network is instantaneous.  The behavior repeats reliably as soon as I restart the server.  I see the long delay in device response even from the zwjs2mqtt control panel, independent of HA.

My apologies for not getting more quickly to the actual problem - I hate it when folks jump to their own conclusions on the HA forum :).

Here's a log snippet of controlling a GE/Jasco plug-in switch during that time period:

```
2021-02-17 20:14:08.358 INFO ZWAVE: Node 64: metadata updated: 112-0-31-8
2021-02-17 20:14:08.957 INFO ZWAVE: Node 44: value updated: 112-0-1 1 => 1
2021-02-17 20:14:09.204 INFO ZWAVE: Node 45: value updated: 112-0-1 1 => 1
2021-02-17 20:14:09.788 INFO ZWAVE: Node 64: value updated: 112-0-1 1 => 1
2021-02-17 20:14:22.044 INFO ZWAVE: Node 19: value updated: 113-0-Home Security-Motion sensor status 8 => 0
2021-02-17 20:14:22.103 INFO ZWAVE: Node 19: metadata updated: 48-0-Motion
2021-02-17 20:14:22.105 INFO ZWAVE: Node 19: value updated: 48-0-Motion true => false
2021-02-17 20:14:22.349 INFO ZWAVE: Node 19: value updated: 113-0-Home Security-Motion sensor status 0 => 8
2021-02-17 20:14:22.410 INFO ZWAVE: Node 19: metadata updated: 48-0-Motion
2021-02-17 20:14:22.413 INFO ZWAVE: Node 19: value updated: 48-0-Motion false => true
2021-02-17 20:14:33.786 INFO ZWAVE: Node 42: value updated: 50-0-value-66049 0 => 0
2021-02-17 20:14:33.788 INFO ZWAVE: Node 42: value updated: 50-0-deltaTime-66049 0 => 0
2021-02-17 20:14:48.369 INFO ZWAVE: Node 19: value updated: 113-0-Home Security-Motion sensor status 8 => 8
2021-02-17 20:14:48.430 INFO ZWAVE: Node 19: metadata updated: 48-0-Motion
2021-02-17 20:14:48.433 INFO ZWAVE: Node 19: value updated: 48-0-Motion true => true
2021-02-17 20:15:03.783 INFO ZWAVE: Node 42: value updated: 50-0-value-66049 0 => 0
2021-02-17 20:15:03.786 INFO ZWAVE: Node 42: value updated: 50-0-deltaTime-66049 0 => 0
2021-02-17 20:15:19.349 INFO ZWAVE: Node 19: value updated: 113-0-Home Security-Motion sensor status 8 => 0
2021-02-17 20:15:19.411 INFO ZWAVE: Node 19: metadata updated: 48-0-Motion
2021-02-17 20:15:19.413 INFO ZWAVE: Node 19: value updated: 48-0-Motion true => false
2021-02-17 20:15:19.669 INFO ZWAVE: Node 19: value updated: 113-0-Home Security-Motion sensor status 0 => 8
2021-02-17 20:15:19.730 INFO ZWAVE: Node 19: metadata updated: 48-0-Motion
2021-02-17 20:15:19.732 INFO ZWAVE: Node 19: value updated: 48-0-Motion false => true
2021-02-17 20:15:33.784 INFO ZWAVE: Node 42: value updated: 50-0-value-66049 0 => 0
2021-02-17 20:15:33.785 INFO ZWAVE: Node 42: value updated: 50-0-deltaTime-66049 0 => 0
2021-02-17 20:15:50.503 INFO ZWAVE: Node 19: value updated: 113-0-Home Security-Motion sensor status 8 => 0
2021-02-17 20:15:50.562 INFO ZWAVE: Node 19: metadata updated: 48-0-Motion
2021-02-17 20:15:50.564 INFO ZWAVE: Node 19: value updated: 48-0-Motion true => false
2021-02-17 20:16:03.786 INFO ZWAVE: Node 42: value updated: 50-0-value-66049 0 => 0
2021-02-17 20:16:03.788 INFO ZWAVE: Node 42: value updated: 50-0-deltaTime-66049 0 => 0
2021-02-17 20:16:33.782 INFO ZWAVE: Node 42: value updated: 50-0-value-66049 0 => 0
2021-02-17 20:16:33.784 INFO ZWAVE: Node 42: value updated: 50-0-deltaTime-66049 0 => 0
2021-02-17 20:16:57.469 INFO ZWAVE: Node 19: value updated: 113-0-Home Security-Motion sensor status 0 => 8
2021-02-17 20:16:57.530 INFO ZWAVE: Node 19: metadata updated: 48-0-Motion
2021-02-17 20:16:57.532 INFO ZWAVE: Node 19: value updated: 48-0-Motion false => true
2021-02-17 20:17:03.783 INFO ZWAVE: Node 42: value updated: 50-0-value-66049 0 => 0
2021-02-17 20:17:03.786 INFO ZWAVE: Node 42: value updated: 50-0-deltaTime-66049 0 => 0
2021-02-17 20:17:14.732 INFO ZWAVE: Node 19: value updated: 113-0-Home Security-Motion sensor status 8 => 8
2021-02-17 20:17:14.792 INFO ZWAVE: Node 19: metadata updated: 48-0-Motion
2021-02-17 20:17:14.794 INFO ZWAVE: Node 19: value updated: 48-0-Motion true => true
2021-02-17 20:17:23.972 INFO ZWAVE: Node 19: value updated: 113-0-Home Security-Motion sensor status 8 => 8
2021-02-17 20:17:24.033 INFO ZWAVE: Node 19: metadata updated: 48-0-Motion
2021-02-17 20:17:24.034 INFO ZWAVE: Node 19: value updated: 48-0-Motion true => true
2021-02-17 20:17:33.279 INFO ZWAVE: Node 19: value updated: 113-0-Home Security-Motion sensor status 8 => 8
2021-02-17 20:17:33.340 INFO ZWAVE: Node 19: metadata updated: 48-0-Motion
2021-02-17 20:17:33.341 INFO ZWAVE: Node 19: value updated: 48-0-Motion true => true
2021-02-17 20:17:33.783 INFO ZWAVE: Node 42: value updated: 50-0-value-66049 0 => 0
2021-02-17 20:17:33.784 INFO ZWAVE: Node 42: value updated: 50-0-deltaTime-66049 0 => 0
2021-02-17 20:17:46.662 INFO ZWAVE: Node 19: value updated: 113-0-Home Security-Motion sensor status 8 => 8
2021-02-17 20:17:46.723 INFO ZWAVE: Node 19: metadata updated: 48-0-Motion
2021-02-17 20:17:46.725 INFO ZWAVE: Node 19: value updated: 48-0-Motion true => true
2021-02-17 20:18:03.784 INFO ZWAVE: Node 42: value updated: 50-0-value-66049 0 => 0
2021-02-17 20:18:03.786 INFO ZWAVE: Node 42: value updated: 50-0-deltaTime-66049 0 => 0
[... cut ...]
2021-02-17 20:25:04.070 INFO ZWAVE: Node 42: value updated: 50-0-deltaTime-66049 0 => 0
2021-02-17 20:25:04.225 INFO ZWAVE: Node 19: value updated: 113-0-Home Security-Motion sensor status 8 => 8
2021-02-17 20:25:04.353 INFO ZWAVE: Node 19: metadata updated: 48-0-Motion
2021-02-17 20:25:04.355 INFO ZWAVE: Node 19: value updated: 48-0-Motion true => true
```

I'm going through it to see if I can see a pattern, but I toggled the light off and one a few times in there, each time with a multi-second delay before it responded.

rccoleman, 2021-02-17:
Another example with debug enabled:

The light was on, so I switched it off from the zwjs2mqtt control panel, waited for it to go off, switched it back on and waited for it to turn on.

Switched light off:

```
2021-02-17 20:59:14.796 DEBUG SOCKET: Event ZWAVE_API emitted to gb0zDhVFCthpqnF-AAAF
2021-02-17 20:59:14.798 INFO ZWAVE: Calling api writeValue with args: [
  {
    nodeId: 58,
    commandClass: 37,
    endpoint: 0,
    property: 'targetValue'
  },
  false,
  [length]: 2
]
2021-02-17 20:59:14.798 INFO ZWAVE: Writing false to 58-37-0-targetValue
```

11s delay

```
2021-02-17 20:59:25.582 INFO ZWAVE: Node 58: value updated: 37-0-currentValue true => false
2021-02-17 20:59:25.584 INFO ZWAVE: Success zwave api call writeValue { success: true, message: 'Success zwave api call', result: undefined }
```

Switched light on:

```
2021-02-17 20:59:33.097 DEBUG SOCKET: Event ZWAVE_API emitted to gb0zDhVFCthpqnF-AAAF
2021-02-17 20:59:33.097 INFO ZWAVE: Calling api writeValue with args: [
  {
    nodeId: 58,
    commandClass: 37,
    endpoint: 0,
    property: 'targetValue'
  },
  true,
  [length]: 2
]
2021-02-17 20:59:33.098 INFO ZWAVE: Writing true to 58-37-0-targetValue
```

4s delay

```
2021-02-17 20:59:37.340 INFO ZWAVE: Node 58: value updated: 37-0-currentValue false => false
2021-02-17 20:59:37.617 INFO ZWAVE: Node 58: value updated: 37-0-currentValue false => false
2021-02-17 20:59:37.918 INFO ZWAVE: Node 58: value updated: 37-0-currentValue false => true
2021-02-17 20:59:37.919 INFO ZWAVE: Success zwave api call writeValue { success: true, message: 'Success zwave api call', result: undefined }
```

AlCalzone, 2021-02-17:
Okay that does seem like the devices are still being queried and respond extremely slow. While waiting for the response, new commands cannot be sent out.

I suggest you keep an eye out for https://github.com/zwave-js/node-zwave-js/pull/1787, where I'll elimitate the query on restart completely.

rccoleman, 2021-02-17:
I'm certainly looking forward to that.  it doesn't seem like there's any querying going on - just reports from my power monitor (Node 42) and various motion sensors (18 & 19), so it doesn't seem busy and those go through just fine  That's why it seems so curious.  Maybe there is an outstanding command in there somewhere.

In any case, eliminating the query will probably help.

AlCalzone, 2021-02-17:
>  It doesn't seem like there's any querying going on

I could say for sure if you made a `zwave-js` log instead of the `zwavejs2mqtt` log :)

scyto, 2021-02-17:
I soften see something similar - typically when i pull a new container image it takes ~26 minutes; restarts caused by UI changes typically only take 8mins or so.  

When it happens i find the locks are being queried and answering through the whole startup interview process, but certain values / command classes time out.

That after *all* the other nodes have been interviewed the locks will complete interview but have secure set to unknown.

That i then have to do a subsequent re-interview of the locks (takes like 2 mins for 4 locks) and then all is good.

However i have not yet tested a full reboot to see if the changes i just made https://github.com/zwave-js/node-zwave-js/issues/1764#issuecomment-780865266 fix this or not.  

My assumption is that for some reason secure messages are not being passed until after everything else completes, unclear if that is a mesh comms  issues or software.  Digging out todays logs where i have the behavior OP describes.

AlCalzone, 2021-02-17:
```
11:26:59.388 DRIVER   Cache file for homeId 0xdef5ff7b found, attempting to restore the network from
                       cache...
11:26:59.411 DRIVER   Restoring the network from cache failed: SyntaxError: Unexpected end of JSON i
                      nput
```
Seems your cache breaks, that's why a new interview is performed. Which node IDs are affected by this?
> That after all the other nodes have been interviewed the locks will complete interview but have secure set to unknown.

scyto, 2021-02-17:
nodes 2,3,4 and 5 - but before you dig in, let me see if i can repro with the new stick firmware and gain and now that my network is healed.  I guess i can delete the cache to invalidate it (i only saw the lock issue when it was 26m+ interviews)

scyto, 2021-02-17:
With the new FW and gain settings earlier a complete rebuild took 11 minutes compared to the previous ~26
(i deleted all files except settings and nodes.json to make it do full query).

Also the locks now work as expected and show secure straight off.

Of course this doesn't mean other folks are having same sort of comms issues.

scyto, 2021-02-17:
@rccoleman when the locks are doing their 20 minute thing have you sat and watched the zwave js log using tail -f?  (doing nothing else and with log set to debug or silly) i found that super useful in correlating what i see in the UI with what scrolls past - it was how i spotted that for my locks some items would time out and some would not.  Just a suggestion.

rccoleman, 2021-02-17:
Yes, I did.  All I saw were power reports from my Aeotec SmartSwitch 6 and motion sensor reports, appearing to be pretty darn quiet.  I'll try to enable zwjs logs to see if I can see any other traffic going on.

rccoleman, 2021-02-18:
I finally got the zwave-js logs and it looks like it's repeatedly timing out when reading some of the parameters from each of the 3 locks.  This is after it successfully read lots of configuration info from them, so it would be surprising to me if it was a communication issue.  Here's a snippet from one of them, a request for parameter #2 from Node 44, with some other activity interspersed.  This is typical of all three locks for most of the parameters queried in this stage, but not all (some succeed).

```
02:33:43.002 DRIVER « [Node 044] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceReport]
                          nonce: 0x051d8a65e3d8eccd
02:33:43.005 SERIAL » 0x011e00132c17988143ac494344e297f2fd017d800536e3d0d95f4c749425e51d  (32 bytes)
02:33:43.006 DRIVER » [Node 044] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      229
                      └─[SecurityCCCommandEncapsulation]
                        │ nonce id: 5
                        └─[ConfigurationCCGet]
                            parameter #: 2
02:33:43.008 SERIAL « [ACK]                                                                   (0x06)
02:33:43.016 SERIAL « 0x0104011301e8                                                       (6 bytes)
02:33:43.016 SERIAL » [ACK]                                                                   (0x06)
02:33:43.016 DRIVER « [RES] [SendData]
                        was sent: true
02:33:43.142 SERIAL « 0x01070013e500000d03                                                 (9 bytes)
02:33:43.142 SERIAL » [ACK]                                                                   (0x06)
02:33:43.143 DRIVER « [REQ] [SendData]
                        callback id:     229
                        transmit status: OK
02:33:49.227 SERIAL « 0x010f00040013097105000000ff0708006a                                (17 bytes)
02:33:49.228 SERIAL » [ACK]                                                                   (0x06)
02:33:49.229 DRIVER « [Node 019] [REQ] [ApplicationCommand]
                      └─[NotificationCCReport]
                          notification type:   Home Security
                          notification status: 255
                          notification state:  Motion detection
02:33:49.230 CNTRLR   [Node 019] [~] [Notification] Home Security[Motion sensor status] [Endpoint 0]
                      : 8 => 8
02:33:49.554 SERIAL « 0x010a00040013043003ff0c26                                          (12 bytes)
02:33:49.555 CNTRLR   [Node 019] [Binary Sensor] Motion: metadata updated               [Endpoint 0]
02:33:49.557 CNTRLR   [Node 019] [~] [Binary Sensor] Motion: true => true               [Endpoint 0]
02:33:49.559 SERIAL » [ACK]                                                                   (0x06)
02:33:49.560 DRIVER « [Node 019] [REQ] [ApplicationCommand]
                      └─[BinarySensorCCReport]
                          type:  Motion
                          value: true
02:33:53.148 SERIAL » 0x010900132d02984025e6d1                                            (11 bytes)
02:33:53.149 DRIVER » [Node 045] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      230
                      └─[SecurityCCNonceGet]
02:33:53.149 CNTRLR   [Node 044] Timed out while waiting for a response from the node
02:33:53.150 CNTRLR « [Node 044] received no value for parameter #2
```

scyto, 2021-02-18:
Sound exactly like I was seeing for last week until this morning. Upgrading controller firmware, increasing gain and doing heal seemed to eliminate the issue. Not saying we have same issue, just sharing,

rccoleman, 2021-02-18:
I pulled the branch from https://github.com/zwave-js/node-zwave-js/pull/1787 and startup is quite speedy.  My three locks come up with blank info and an Unknown status, but all other nodes come up with proper info and show "Complete" almost immediately.  If I hit "Refresh Values" for each of the locks, they also populate and their entities show up in HA.  Once that's done, everything works instantly!

AlCalzone, 2021-02-18:
> I finally got the zwave-js logs and it looks like it's repeatedly timing out when reading some of the parameters from each of the 3 locks. This is after it successfully read lots of configuration info from them, so it would be surprising to me if it was a communication issue.

That could very well be true. Could you coordinate with @blhoward2 to check which config parameters are defined and shouldn't be there?

rccoleman, 2021-02-18:
I'm pretty sure that the problem is that usercode type (owner, guest, worker) parameters that aren't set time out during the query.  All 3 locks have the first type set to "owner" (1), and the rest of the 29 parameters for each lock time out because I haven't set them.  The curious bit is that one of my locks had usercode #2 set when I bought it and I've since blanked it out and set that slot to "available."  In the zwjs logs, it times out on the usercode type for that slot just like the the other locks, but the zwjs2mqtt UI displays that parameter as "Owner" (1).  I suspect that it's just displaying a cached value, but it was unexpected.

So, it seems like these parameters are all valid, but each timeout contributes ~30s of wasted time with associated retries.  Luckily, it queries all of the locks concurrently, but it still takes nearly 20mins of timed out parameter queries before it moves on to the associations (log is above, timestamps are between ~02:33:xx and ~02:51:55).

The only mitigation that I can see is that each slot is marked with "Available", "Enabled", or "Disabled" in the UserCodeCC entry, and entries that are marked "Available" could be assumed to have no associated "type" config param.

Example:

```
02:28:00.779 DRIVER « [Node 064] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[UserCodeCCReport]
                            user id:   3
                            id status: Available
                            user code:
```

```
02:34:40.585 DRIVER » [Node 064] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      239
                      └─[SecurityCCCommandEncapsulation]
                        │ nonce id: 58
                        └─[ConfigurationCCGet]
                            parameter #: 3
```

```
02:34:50.727 CNTRLR   [Node 064] Timed out while waiting for a response from the node
02:34:50.728 CNTRLR « [Node 064] received no value for parameter #3
```

If that's not a universal solution across all locks, maybe it's enough to just skip the initial query as in https://github.com/zwave-js/node-zwave-js/pull/1787.

blhoward2, 2021-02-18:
If you set a user code for those slots, does it then work correctly on those slots while the code is set? If so I'm not sure what do about that...Thoughts @AlCalzone?

rccoleman, 2021-02-18:
I'm testing that now.

blhoward2, 2021-02-18:
Also, what do you mean by set them as Available? Do you not have an option drop-down that only has owner, guest, vacation?

AlCalzone, 2021-02-18:
The combination between configuration and user code CC is something specific to this device and no generic rule I can implement in the driver.

The next major version will stop requerying the codes on each reboot so ideally you only have to endure the delay once.

rccoleman, 2021-02-18:
> Also, what do you mean by set them as Available? Do you not have an option drop-down that only has owner, guest, vacation?

Here's an example of a usercode type parameter that's set and one that isn't:

[image]

And usercodes with "Enabled" and "Available":

[image]

In any case, it sounds like it's not a universal rule.

blhoward2, 2021-02-18:
Ah, gotcha. That's under values not configuration.

rccoleman, 2021-02-18:
Correct, it's a correlation between the UserCodeCC values and the config parameters.

closed 2021-02-18 by rccoleman

scyto, 2021-02-18:
as a note i moved back from current public dev container to one built on zwavejs branch master and zwavejs2mqtt branch rgbw_switch and my slow  28+ minute issue came back and is identical to one described here where it just time outs on user codes

what i don't understand is why it appears to not be present as an issue in the public dev container.