zwave-js/zwave-js #1807: Driver crashes / network restarts after inclusion

maivorbim, 2021-02-17:
After including a new device, the zwavejs driver crashes and the zwave network restarts the interview for all nodes (approx 60 nodes). Tested with Aeotec Gen 5 Z-stick and Aeotec Aerq temperature sensors. Using Zwavejs 6.2.0 running on HA core-2021.2.3, supervisor-2021.02.10.

I have attached the zwavejs and zwavejs2mqtt silly logs. Crash happens at 20:49:56.100 and the error is:

Zwave JS
```
20:49:56.048 CNTRLR   [Node 088] Node 88 did not respond after 3 attempts, it is presumed dead
20:49:56.100 DRIVER   destroying driver instance...
```

Zwavejs2mqtt
```
2021-02-17 20:49:23.096 INFO ZWAVE: Success zwave api call startInclusion { success: true, message: 'Success zwave api call', result: true }
2021-02-17 20:49:53.152 INFO ZWAVE: Controller status: Inclusion stopped
2021-02-17 20:49:57.021 INFO ZWAVE: Connecting to /dev/serial/by-id/usb-0658_0200-if00
2021-02-17 20:49:57.014 INFO MQTT: MQTT is disabled
2021-02-17 20:49:57.036 INFO APP: Listening on port 44920
```

Please let me know if any further details are required to debug this.

[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/5999007/zwavejs2mqtt.log)
[zwavejs_359.log](https://github.com/zwave-js/node-zwave-js/files/5999008/zwavejs_359.log)

AlCalzone, 2021-02-17:
Duplicate of #1742
Please update :)

maivorbim, 2021-02-20:
have to reopen this as the driver still crashes on 6.4.0 after adding a device to the network.

AlCalzone, 2021-02-20:
Please make a zwave-js log, loglevel `debug` or `silly` and attach it here as a file.

maivorbim, 2021-02-20:
zwavejs_358.log is the log leading up to the crash. I excluded node 91 (Aeotec Multisensor 6) and reincluded it as Node 92, when the driver crashed.
Crash happens at: 

`23:34:00.820 DRIVER   destroying driver instance...
`

zwavejs_4017.log is the log after the crash.
zwavejs2mqtt.log is the log before and after the crash, included for completeness.

[zwavejs_358.log](https://github.com/zwave-js/node-zwave-js/files/6016557/zwavejs_358.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6016558/zwavejs2mqtt.log)
[zwavejs_4017.log](https://github.com/zwave-js/node-zwave-js/files/6016559/zwavejs_4017.log)

AlCalzone, 2021-02-21:
So what's weird here is that you include the node into the network but immediately afterwards, all communication fails due to missing acknowledgements from the node. 
This time another call failed I also forgot to guard against an error. The fix is up at #1839

PR #1839 (2021-02-21): fix: catch version query errors during bootstrap too
fixes: #1807
closed 2021-02-21 by AlCalzone