zwave-js/zwave-js #1823: Enitities missing from Shenzhen Neo - PD03Z - PIR Motion Sensor with Temperature Sensor

deku-m, 2021-02-19:
**Describe the bug**

Not all enitities are show in the device info and also cant use the entities i need like the motion sensor.
As it is not available on node 9/17 but on node 13 there is a motion enitity which is the same type as node 17.
Node 9 is older model.

**Device information**

Which device(s) is/are affected (make/model)?  
Shenzhen Neo Electronics Co., Ltd - PD03Z - PIR Motion Sensor with Temperature Sensor
What are the node IDs?
Node ID: 9, 13, 17

**Last Known Working Configuration**
- [ ] New device
- [ ] Previously working device (node-zwave-js)
    - Which library version/docker image/adapter version?
    - Have you made any recent configuration changes to the device? Describe.
- [X] Previously working device (other platform)
    - Which platform? Zwave integrated (deprecated)
    - Have you made any recent configuration changes to the device? Describe.
    Added them to ZWAVE-JS from clean install of Home Assistant.

**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [X] Other: Official Addon-Z-Wave JS 

**Extra info:**
Driver Version: 6.2.0
Server Version: 1.0.0-beta.6
Home ID: 3945136619
Nodes ready: 10 / 15

**To Reproduce**
Steps to reproduce the behavior:
1. Add node to Zwave with inclusion
2. Wait
3. When ready check enitity.
4. See difference..

**Additional context**
See attachments

**Logfile:**

- [zwave_js_dump.zip](https://github.com/zwave-js/node-zwave-js/files/6010483/zwave_js_dump.zip)
- [image]
- [image]

AlCalzone, 2021-02-19:
Please make a zwave-js log, loglevel `debug` or `silly` and attach it here as a file.

deku-m, 2021-02-19:
Sorry for asking but where can i do that in the addon? 
Is that a new variable in docker container or also possible to add in configuration on addon page?

AlCalzone, 2021-02-19:
To be honest, I don't know. If you have access to zwavejs2mqtt you can do it there.

robertsLando, 2021-02-19:
Settings > Zwave > loglevel

marcelveldt, 2021-02-19:
I've seen issues with this particular motion sensor before. In most/all cases it was enough to manually wake the device once to get the missing values pulled in. The value missing in this case is from Notification CC. 

@deku-m A (temporary) workaround is to enable the legacy binary sensor (the one with the black circle with check icon).

@robertsLando you might want to extend the GH issue tracker template and request the user to include a full dump, which is very easy for debugging.

Home Assistant --> settings --> Integrations --> Z-Wave JS --> Configure --> Create dump --> zip the json file and post it here.

AlCalzone, 2021-02-19:
@marcelveldt does that include logs?

marcelveldt, 2021-02-19:
> @marcelveldt does that include logs?

No, but it does include a dump of the full state (all nodes, all values, metadata etc) so it's a good start.
We're also looking into providing options to create the log and/or increase log levels.

AlCalzone, 2021-02-19:
I've had a user who provided a `jsonl` file earlier, but that isn't really readable, since IDE auto-formatting does not work.

marcelveldt, 2021-02-19:
Yeah, we should look at changing that and just make it a regular json object.
Workaround is to just remove the first line...

deku-m, 2021-02-19:
> I've seen issues with this particular motion sensor before. In most/all cases it was enough to manually wake the device once to get the missing values pulled in. The value missing in this case is [...]
I tried to enable it but didnt allowed me to and stayed greyed out.
But i might have a log from docker as i copied it to the host not sure if it has debug info you want.
As the installation got corrupt by something else i need to restore but will try to give more debug info asap.

wauswaus, 2021-02-21:
> I've seen issues with this particular motion sensor before. In most/all cases it was enough to manually wake the device once to get the missing values pulled in. The value missing in this case is [...]

Waking up the device did indeed add an additional 'item' under the zwavejs2mqtt control panel view (under notification a 'Home Security-Motion sensor status] Home Security (property)' ). The value is 0 when there is no motion and 8 with motion.
However, this new property is not added to the Home Assistant entity list for the device.
The trick with the legacy binary sensor did not work for me; it is always true.

Any suggestion to continue for here?

deku-m, 2021-02-22:
For me the issue has been resolved now. Did factory/hard reset on both the zwave stick as also on the zwave nodes for my issue.
I started again from scratch at hass but then with addon zwavejstomqtt which is very friendly. 
So the hard-reset works.

Now they are showing as they should be. Had it one more time but resolved by pressing the button on the nodes for negotiation.
Not sure if others have issues else im gonna close this ticket?

AlCalzone, 2021-02-22:
> Any suggestion to continue for here?

Raise this issue in the HA repo.

@deku-m Thanks, I'll close this.

closed 2021-02-22 by AlCalzone