zwave-js/zwave-js #1833: Aeon Recessed Door Sensor 5 wakeup 0 [bug]

chrish987, 2021-02-19:
**Describe the bug**

The Aeon Recessed Door sensor seems to have wakeup default to 0. 

I noticed this as these nodes (i have 2) never seem to complete the reinterview process after a server restart. Looking at the configuration I am guessing that a configured wakeup of 0 is stopping the node ever waking up.

When discovered these nodes show up as ZW089-B. Can workaround be manually forcing wakeup (click button on node).

I have manually updated value to 3600 will see if this fixes issue on next restart, but seems like the 0 default value is probably a problem.

**Device information**
Device ID: 134-89-514 (0x0086-0x0059-0x0202)
[image]

AlCalzone, 2021-02-19:
The issue template is there for a reason, you should not simply delete half of it. Looking into my glass sphere I see the following:

The nodes should be ready almost immediately after a restart - even if they are in the RestartFromCache stage. If not, you probably need to update to the latest zwave-js version.
We cannot do anything about the default configuration value.
The query after restart will not be done once #1787 is merged.

blhoward2, 2021-02-20:
You need to wake the sensor back up to have it take the value. The default is stored on the device, we're not setting the default each time. Mine work just fine.

chrish987, 2021-02-20:
The odd thing is that the minimum value for wakeup is 240 yet default 0. I thought it might be a wrong setting in the node definition. I have fixed it by manually changing.

I have some other 'quirks' with behaviors on server restart (e.g. nodes staying in ProtocolInfo when I know they have been through the wakeup cycle (seen it in the logs).

I guess most of this is really cosmetic as the nodes 'work' they send values and notifications - so just the display in the node list.

I will wait until #1787 is released and see what new behavior is.

Thanks

blhoward2, 2021-02-20:
What makes you think 240 is the minimum? 0 disables wake ups.

chrish987, 2021-02-20:
this is from the Gen7 docs, Gen5 hard to find for some reason. Anyway - happy for this issue to be closed - will wait for the #1787 release.
[image]

blhoward2, 2021-02-20:
Well, our config files have no impact on the wake up. That default value is coming from the device itself I imagine.

AlCalzone, 2021-02-21:
> That default value is coming from the device

👆🏻 this.

> I have some other 'quirks' with behaviors on server restart (e.g. nodes staying in ProtocolInfo

If the nodes have been interviewed before, they should not be in the ProtocolInfo stage. But since you didn't provide a log like the template asks for, all we can do here is guess.

chrish987, 2021-02-21:
My log is 95M, I am guessing you do not want all that. Before next restart I will delete the log and send through more relevant information - or will you need things from an earlier log? (i.e. when the device was first included?)

My log file also seems to not have dates, just times, lines look like this:
_03:13:28.018 DRIVER   starting driver...
03:13:28.023 DRIVER   opening serial port /dev/ttyAMA0_
Is this something in my config, or is there a way to enable dates? (i am running the zwavejs2mqtt docker image)

AlCalzone, 2021-02-21:
It compresses fairly well but having a fresh log without old cruft would be better, I agree

closed 2021-02-27 by zwave-js-assistant[bot]

robertsLando, 2023-03-14:
Hi @chrish987, sorry for bothering you in a issue but I don't have your email, could you reach me out via private email or give me your email? I need to ask you something, thanks 🙏🏼 

My email is: daniel.sorridi@gmail.com