zwave-js/zwave-js #1835: bug: lock interview fails with 'giving up' status in UI says protocolinfo

scyto, 2021-02-20:
**Describe the bug**

A lock that was working now appears stuck at protocolinfo despite repeaated re-interviews.
Log indicates system gave up interviewing.
1. if the system gave up interviewing 'protocolinfo' as a state seems the wrong state (is this a zwavejs2mqtt bug?)
2. i cannot fathom from the logs why the lock is being this way (i moved the controller hanging of log - that didn't fix so i doubt a classic comms issues.

**Device information**

Node 04 - Schlage FE599NX

**Last Known Working Configuration**
- [ ] New device
- [X] Previously working device (node-zwave-js)
    - I did move from public docker to `varet/zj2m:rgbw_color_armv7` built by @billiaz today
    - No changes to the device other than programming a user code earlier in the day.

- [ ] Previously working device (other platform)
    - Which platform?
    - Have you made any recent configuration changes to the device? Describe.
    
**Installation information**
How did you install `node-zwave-js`?
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] ioBroker.zwave2 adapter
- [ ] Pkg
- [X] Manual Docker build
    - `node-zwave-js` branch: unknown 
    - `zwavejs2mqtt` branch: unknown 
    - @billiaz what branch did you build `varet/zj2m:rgbw_color_armv7` from?
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**
Steps to reproduce the behavior:
RE-interview node 4 - look at logs and UI.

**Additional context**
Add any other context about the problem here.
[image]

App Version
1.2.1
Zwavejs Version
6.4.0
Zwavejs-server Version
1.0.0-beta.7

**Logfile:**
```
20:28:04.491 CNTRLR   [Node 004] Beginning interview - last completed stage: None
20:28:04.491 CNTRLR   [Node 004] new node, doing a full interview...
20:28:04.491 CNTRLR » [Node 004] querying protocol info...
20:28:04.495 SERIAL » 0x0104004104be                                                       (6 bytes)
20:28:04.496 DRIVER » [REQ] [GetNodeProtocolInfo]
                        payload: 0x04
20:28:04.501 SERIAL « [ACK]                                                                   (0x06)
20:28:04.502 SERIAL « 0x010a014152dc00044003007c                                          (12 bytes)
20:28:04.504 SERIAL » [ACK]                                                                   (0x06)
20:28:04.505 DRIVER « [RES] [GetNodeProtocolInfo]
                        payload: 0x52dc0004400300
20:28:04.510 CNTRLR « [Node 004] received response for protocol info:
                      basic device class:    Routing Slave
                      generic device class:  Entry Control
                      specific device class: Secure Keypad Door Lock
                      is a listening device: false
                      is frequent listening: true
                      is a routing device:   true
                      is a secure device:    unknown
                      is a beaming device:   true
                      maximum baud rate:     40000 kbps
                      version:               3
20:28:04.511 CNTRLR   [Node 004] Interview stage completed: ProtocolInfo
20:28:04.511 CNTRLR » [Node 004] querying node info...
20:28:04.515 SERIAL » 0x01040060049f                                                       (6 bytes)
20:28:04.516 DRIVER » [Node 004] [REQ] [RequestNodeInfo]
                        payload: 0x04
20:28:04.521 SERIAL « [ACK]                                                                   (0x06)
20:28:04.522 SERIAL « 0x01040160019b                                                       (6 bytes)
20:28:04.523 SERIAL » [ACK]                                                                   (0x06)
20:28:04.524 DRIVER « [RES] [RequestNodeInfo]
                        payload: 0x01
20:28:36.876 SERIAL « 0x0106004981000031                                                   (8 bytes)
20:28:36.883 SERIAL » [ACK]                                                                   (0x06)
20:28:36.890 DRIVER « [REQ] [ApplicationUpdateRequest]
                        payload: 0x810000
20:28:36.917 CNTRLR   [Node 004] Querying the node info failed
20:28:36.918 CNTRLR   [Node 004] Interview attempt 1/5 failed, retrying in 5000 ms...
20:28:41.930 CNTRLR   [Node 004] Beginning interview - last completed stage: ProtocolInfo
20:28:41.932 CNTRLR » [Node 004] querying node info...
20:28:41.945 SERIAL » 0x01040060049f                                                       (6 bytes)
20:28:41.947 DRIVER » [Node 004] [REQ] [RequestNodeInfo]
                        payload: 0x04
20:28:41.958 SERIAL « [ACK]                                                                   (0x06)
20:28:41.962 SERIAL « 0x01040160019b                                                       (6 bytes)
20:28:41.962 SERIAL » [ACK]                                                                   (0x06)
20:28:41.965 DRIVER « [RES] [RequestNodeInfo]
                        payload: 0x01
20:29:16.881 SERIAL « 0x0106004981000031                                                   (8 bytes)
20:29:16.882 SERIAL » [ACK]                                                                   (0x06)
20:29:16.884 DRIVER « [REQ] [ApplicationUpdateRequest]
                        payload: 0x810000
20:29:16.889 CNTRLR   [Node 004] Querying the node info failed
20:29:16.890 CNTRLR   [Node 004] Interview attempt 2/5 failed, retrying in 10000 ms...
20:29:26.897 CNTRLR   [Node 004] Beginning interview - last completed stage: ProtocolInfo
20:29:26.902 CNTRLR » [Node 004] querying node info...
20:29:26.911 SERIAL » 0x01040060049f                                                       (6 bytes)
20:29:26.912 DRIVER » [Node 004] [REQ] [RequestNodeInfo]
                        payload: 0x04
20:29:26.918 SERIAL « [ACK]                                                                   (0x06)
20:29:26.920 SERIAL « 0x01040160019b                                                       (6 bytes)
20:29:26.921 SERIAL » [ACK]                                                                   (0x06)
20:29:26.923 DRIVER « [RES] [RequestNodeInfo]
                        payload: 0x01
20:29:52.085 SERIAL « 0x0106004981000031                                                   (8 bytes)
20:29:52.087 SERIAL » [ACK]                                                                   (0x06)
20:29:52.089 DRIVER « [REQ] [ApplicationUpdateRequest]
                        payload: 0x810000
20:29:52.099 CNTRLR   [Node 004] Querying the node info failed
20:29:52.100 CNTRLR   [Node 004] Interview attempt 3/5 failed, retrying in 15000 ms...
20:30:07.117 CNTRLR   [Node 004] Beginning interview - last completed stage: ProtocolInfo
20:30:07.118 CNTRLR » [Node 004] querying node info...
20:30:07.126 SERIAL » 0x01040060049f                                                       (6 bytes)
20:30:07.126 DRIVER » [Node 004] [REQ] [RequestNodeInfo]
                        payload: 0x04
20:30:07.133 SERIAL « [ACK]                                                                   (0x06)
20:30:07.139 SERIAL « 0x01040160019b                                                       (6 bytes)
20:30:07.140 SERIAL » [ACK]                                                                   (0x06)
20:30:07.142 DRIVER « [RES] [RequestNodeInfo]
                        payload: 0x01
20:30:37.493 SERIAL « 0x0106004981000031                                                   (8 bytes)
20:30:37.494 SERIAL » [ACK]                                                                   (0x06)
20:30:37.496 DRIVER « [REQ] [ApplicationUpdateRequest]
                        payload: 0x810000
20:30:37.503 CNTRLR   [Node 004] Querying the node info failed
20:30:37.504 CNTRLR   [Node 004] Interview attempt 4/5 failed, retrying in 20000 ms...
20:30:57.510 CNTRLR   [Node 004] Beginning interview - last completed stage: ProtocolInfo
20:30:57.511 CNTRLR » [Node 004] querying node info...
20:30:57.524 SERIAL » 0x01040060049f                                                       (6 bytes)
20:30:57.525 DRIVER » [Node 004] [REQ] [RequestNodeInfo]
                        payload: 0x04
20:30:57.532 SERIAL « [ACK]                                                                   (0x06)
20:30:57.534 SERIAL « 0x01040160019b                                                       (6 bytes)
20:30:57.535 SERIAL » [ACK]                                                                   (0x06)
20:30:57.537 DRIVER « [RES] [RequestNodeInfo]
                        payload: 0x01
20:31:31.147 SERIAL « 0x0106004981000031                                                   (8 bytes)
20:31:31.148 SERIAL » [ACK]                                                                   (0x06)
20:31:31.150 DRIVER « [REQ] [ApplicationUpdateRequest]
                        payload: 0x810000
20:31:31.161 CNTRLR   [Node 004] Querying the node info failed
20:31:31.162 CNTRLR   [Node 004] Failed all interview attempts, giving up.```

scyto, 2021-02-20:
hmm just came back to the house and realize the lock/unlock buttons on the inside of the door are dead along with led.  Oddly the outside buttons and lock code works perfectly.  I will leave this overnight, if no one wants any logs i will pull the batteries and reboot the bugger.

AlCalzone, 2021-02-20:
1.) if it worked correctly, why does the driver attempt a complete interview? Did your cache break again?

2.) The node info request does not get answered. Without it we don't know what the device supports and cannot continue the interview.

scyto, 2021-02-20:
1 yes my cache breaks very frequently.
2 i am going to pull the batteries, the lights on the inside not working / physical internal buttons not working is very suspcious

scyto, 2021-02-20:
yes lock had actually crashed
reboot of lock was needed

closed 2021-02-20 by scyto